whad-team/whad-client #43: Index out of range during "wble-central --interface ubertooth0; wble-central>scan"

grymoire, 2024-08-18:
I installed via pip. While doing wble-central scan this error returned.
`

  File "/home/pi/Src/env/lib/python3.11/site-packages/whad/ble/profile/advdata.py", line 192, in from_bytes
    limited_disc = ((ad_record[0] & 0x01) != 0)
                     ~~~~~~~~~^^^
IndexError: index out of range
Segmentation fault
`
Full output attached 
[whad.txt](https://github.com/user-attachments/files/16651458/whad.txt)

virtualabs, 2024-08-19:
It looks like a faulty advertisement record caused this error (flags record with no data). This failed because of the wrong check we made before parsing the record data.

virtualabs, 2024-08-19:
Allright, I've fixed the bad check and added some unit tests to avoid any regression. You can test if this fix works by cloning the main branch and giving it a try, or you can just wait for a new release (coming soon).