whad-team/whad-client #44: Add support for LOGITacker tunneling of data over the Unifying dongle raw hid interface

RoganDawes, 2024-08-20:
MaMe82 figured out how to tunnel arbitrary binary data over the Unifying dongle's raw hid endpoint, via an nRF52840 dongle. It would be neat to support it in whad-client too. See https://github.com/RoganDawes/LOGITacker.

virtualabs, 2024-08-20:
Are you talking about LOGITacker's "experimental covert channel for air-gap bridging" feature ?

RoganDawes, 2024-08-20:
Yes. LOGITacker is able to type out a payload (quite a large one!) that identifies and opens the unifying dongle on a vulnerable Windows machine, and then establishes a cmd.exe shell tunneled over the raw hid endpoint (22 bytes per message, I believe) that can be interacted with via the nRF52840 dongle.

While this is a handy implementation, and has no platform dependencies, it would be nice to be able to easily experiment with the typed payload for other platforms, perhaps, or have the tunneled connection terminate on the host rather than on the dongle for automated interaction with the program on the other end.