whad-team/whad-client #55: BLE Link Layer Proxy fails on LL_CHANNEL_MAP_IND

doggkruse, 2024-09-15:
It appears following changes in channel maps is not currently supported.  This limits the usefulness with any sophisticated BLE central such as a smart phone
[ChannelMapProxyFailure.pcapng.gz](https://github.com/user-attachments/files/17007246/ChannelMapProxyFailure.pcapng.gz)

virtualabs, 2024-09-15:
I know for sure that the current version of ButteRFly does not correctly handle channel map updates (and other control procedures by the way) as I experienced this kind of issues myself, especially when dealing with BLE 5 smartphones using the second channel selection algorithm. WHAD's BLE stack does not manage these specific control procedures and they are left to the hardware as they are critical to keep a connection active (and also because precise timing is required).

What devices did you use to setup your link-layer proxy ? Was ButteRFly involved ?

doggkruse, 2024-09-15:
> I know for sure that the current version of ButteRFly does not correctly handle channel map updates (and other control procedures by the way) as I experienced this kind of issues myself, especially [...]

NRF52840 running butterfly.  Are there other supported devices that handle channel map updates and other control PDUs? I

I may look at adding support for Butterfly but just getting into this project

virtualabs, 2024-09-15:
For now butterfly has some known limitations due to its age (it is basically a port of @RCayre's original butterfly firmware used in Mirage with some improvements to make it compatible with WHAD), HCI adapters handle this with no issue at all but the problem is you won't be able to mess with these PDUs are they'll be handled internally by the adapter's firmware (WHAD uses an adaptation layer upon the exposed HCI interface so there's no way to get access to them).

If you just need a link-layer proxy and don't care about any channel map update (or other BLE 5.x new control procedures by the way), you can do it with two Bluetooth HCI adapters and that'll do the job. If you plan to have some fun with a device by messing with channel map updates when in a mitm position, that'll definitely be harder to achieve in the current state of this project.

We put a lot of efforts into this framework but yes, there is still a lot of work to be done in our firmwares too :sweat_smile: ...