whad-team/whad-client #57: ble_peripheral_basic fails to communicate with butterfly

sacca97, 2024-09-16:
`➜  ble git:(main) ✗ python ble_peripheral_basic.py 0
<DeviceReady >
<InfoQueryResp capabilities=[50331734, 67108942, 100663374, 117440590, 16777358], device_id=b'~\x14\x85\x18\xad=\xfc\xbd\xa6\xec(<\xf7\xbd \xee', fw_author=b'Romain Cayre', fw_url=b'https://github.com/whad-team/butterfly', fw_version_major=1, fw_version_minor=0, fw_version_rev=1, max_speed=115200, proto_min_ver=2, type=1>
<DomainInfoQueryResp domain=50331648, supported_commands=134214387>
<DomainInfoQueryResp domain=67108864, supported_commands=4063>
<DomainInfoQueryResp domain=100663296, supported_commands=491>
<DomainInfoQueryResp domain=117440512, supported_commands=2027>
<DomainInfoQueryResp domain=16777216, supported_commands=2031492>
Traceback (most recent call last):
  File "/home/sacca/Projects/whad-client/whad/device/__init__.py", line 1144, in wait_for_message
    msg = self.__msg_queue.get(block=True, timeout=timeout)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.12/queue.py", line 179, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/sacca/Projects/whad-client/examples/ble/ble_peripheral_basic.py", line 28, in <module>
    periph = Peripheral(WhadDevice.create('uart:/dev/ttyACM0'), profile=my_profile)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sacca/Projects/whad-client/whad/ble/connector/peripheral.py", line 111, in __init__
    self.enable_peripheral_mode(adv_data, scan_data)
  File "/home/sacca/Projects/whad-client/whad/ble/connector/__init__.py", line 564, in enable_peripheral_mode
    resp = self.send_command(msg, message_filter(CommandResult))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sacca/Projects/whad-client/whad/device/__init__.py", line 423, in send_command
    return self.__device.send_command(message, filter)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sacca/Projects/whad-client/whad/device/__init__.py", line 1235, in send_command
    raise timedout
  File "/home/sacca/Projects/whad-client/whad/device/__init__.py", line 1229, in send_command
    result = self.wait_for_message(self.__timeout, command=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sacca/Projects/whad-client/whad/device/__init__.py", line 1159, in wait_for_message
    raise WhadDeviceTimeout('WHAD device did not answer to a command')
whad.exceptions.WhadDeviceTimeout: WhadDeviceTimeout(WHAD device did not answer to a command)
➜  ble git:(main) ✗`

virtualabs, 2024-09-16:
Allright, I finally figured out what is happening here.

First, our BLE `Peripheral` class puts the WHAD interface in BLE peripheral mode, providing some advertising records (and scan response records). The problem is that when no advertising data is passed to the Peripheral class `__init__()` method, the WHAD interface is asked to advertise a device with no advertising records... And this makes ButteRFly crash, the firmware that runs on a nRF52840 dongle expecting at least a single record. 

I've modified our `Peripheral` class in order to add a default flags advertising record if none is provided, thus avoiding ButteRFly to crash, but also patched ButteRFly to include some additional checks and return an error if this situation happens in the future. Modifications brought to ButteRFly will be pushed very soon. 

For now, if you want to try our `ble_peripheral_basic.py` example without switching to our `dev` branch, you can simply change the way the peripheral is created:

``` python
    # Create the connector & the profile
    my_profile = MyPeripheral()
    periph = Peripheral(WhadDevice.create(sys.argv[1]), profile=my_profile, adv_data=AdvDataFieldList(
        AdvCompleteLocalName(b'TestMe!'),
        AdvFlagsField()
    ))
```

And everything should work with ButteRFly :crossed_fingers:.

Or if you prefer giving our `dev` branch a try, you may benefit of some recent fixes including the one I've just made. It could be nice to check if my fix works as expected on your side too, so that I can close this issue :smile:.

sacca97, 2024-09-17:
All good. Problem fixed in `dev` branch

virtualabs, 2024-09-17:
Perfect, closing this issue.