whad-team/whad-client #59: LL Connection failure with MAC OS + Broadcom BCM_4387

sacca97, 2024-09-17:
Using the example peripheral device. This is what the Wireshark capture looks like from WHAD:

[image]

Different behavior with external dongle (e.g., realtek) or with BCM_4378.

virtualabs, 2024-09-17:
I know this behavior, it happened to me once. The BLE stack is supposed to send a single LL_VERSION_IND PDU but for some reason it reacts each time it receives an LL_VERSION_PDU by ... sending another  LL_VERSION_PDU. Fun part is that the peripheral answers each PDU while the specs definitely states it must only answer the first request. 

I have no clue why this is happening, for now. And yes, the BCM stack is also weird :rofl:.

sacca97, 2024-09-17:
`    def is_version_sent(self, conn_handle):
        if conn_handle in self.connections:
            self.connections[conn_handle]["version_sent"]
        return False`
        
I think i see the problem of the double response by the peripheral btw @virtualabs . Gonna check out properly tomorrow when I have the devices at hand.

sacca97, 2024-09-18:
Yup that is indeed the problem as method always return false. However it does not change the final outcome

[image]

sacca97, 2024-09-19:
Additionally the Arrival Time is obv wrong, and it could be that one causing sync problems, but I don't know how to fix it, yet