whad-team/whad-client #74: Example ble_existing_connection_sniffer.py does not work

MKesenheimer, 2024-10-16:
The example `ble/ble_existing_connection_sniffer.py` throws an error if executed:

```
Traceback (most recent call last):
  File "/home/mkesenheimer/tools/whad/whad-client/examples/ble/ble_existing_connection_sniffer.py", line 20, in <module>
    sniffer.configure(
  File "/home/mkesenheimer/tools/whad/venv/lib/python3.12/site-packages/whad/ble/connector/sniffer.py", line 186, in configure
    self._enable_sniffing()
  File "/home/mkesenheimer/tools/whad/venv/lib/python3.12/site-packages/whad/ble/connector/sniffer.py", line 124, in _enable_sniffing
    self.sniff_active_connection(access_address, crc_init, channel_map, hop_interval, hop_increment)
  File "/home/mkesenheimer/tools/whad/venv/lib/python3.12/site-packages/whad/ble/connector/__init__.py", line 426, in sniff_active_connection
    msg = self.hub.ble.create_sniff_active_conn(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mkesenheimer/tools/whad/venv/lib/python3.12/site-packages/whad/hub/ble/__init__.py", line 339, in create_sniff_active_conn
    sniff_connreq.channel_map = channel_map.value
                                ^^^^^^^^^^^^^^^^^
AttributeError: 'bytes' object has no attribute 'value'

```

`channel_map` is defined in `ble/ble_existing_connection_sniffer.py` like so:
```
channel_map=bytes.fromhex("ffefffff1f")
```

What is the correct value for `channel_map`?

MKesenheimer, 2024-10-16:
Even replacing the above line with
```
channel_map=0xffefffff1f
```
does not work, as then the following error occurs:
```
  File "/home/mkesenheimer/tools/whad/venv/lib/python3.12/site-packages/whad/hub/ble/__init__.py", line 339, in create_sniff_active_conn
    sniff_connreq.channel_map = channel_map.value

AttributeError: 'int' object has no attribute 'value'
```

Which python type implements the `value` attribute?

MKesenheimer, 2024-10-16:
The next error occurs, if the optional arguments `crc_init` and `channel_map` are removed from `SynchronizedConnection`:

```
Traceback (most recent call last):
  File "/home/mkesenheimer/tools/whad/whad-client/examples/ble/ble_existing_connection_sniffer.py", line 30, in <module>
    for i in sniffer.sniff():
             ^^^^^^^^^^^^^^^
  File "/home/mkesenheimer/tools/whad/venv/lib/python3.12/site-packages/whad/ble/connector/sniffer.py", line 279, in sniff
    if message.hop_increment > 0:
       ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'hop_increment'
```

virtualabs, 2024-10-22:
It seems our example was a bit outdated and also that some part of the BLE sniffer has not been tested enough. I've managed to reproduce this behavior, fixed these errors and I created a PR related to this issue.

You can check if these fixes solve your problems by testing the `bugfix/issue-74` branch.

MKesenheimer, 2024-10-22:
Thank you for addressing these issues. However, I don't think that fully fixed the problems.
I still get problems if I run the examples.

With a butterfly device:
```
$ python ble_existing_connection_sniffer.py uart0
Traceback (most recent call last):
  File "/home/mkesenheimer/tools/ble/whad/whad-client/examples/ble/ble_existing_connection_sniffer.py", line 31, in <module>
    for i in sniffer.sniff():
             ^^^^^^^^^^^^^^^
  File "/home/mkesenheimer/tools/ble/whad/venv/lib/python3.12/site-packages/whad/ble/connector/sniffer.py", line 279, in sniff
    if message.hop_increment > 0:
       ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'hop_increment'
```

With a Bluetooth dongle:
```
$ python ble_existing_connection_sniffer.py hci0
Traceback (most recent call last):
  File "/home/mkesenheimer/tools/ble/whad/whad-client/examples/ble/ble_existing_connection_sniffer.py", line 19, in <module>
    sniffer = Sniffer(dev)
              ^^^^^^^^^^^^
  File "/home/mkesenheimer/tools/ble/whad/venv/lib/python3.12/site-packages/whad/ble/connector/sniffer.py", line 36, in __init__
    raise UnsupportedCapability("Sniff")
whad.exceptions.UnsupportedCapability: UnsupportedCapability(Sniff)
```

MKesenheimer, 2024-10-22:
Sorry, I got something wrong with my virtual environment. The commit fixed the issues for me. Thanks!

closed 2024-10-22 by MKesenheimer