whad-team/whad-client #82: AttributeError: 'Central' object has no attribute 'auto'

sacca97, 2024-10-25:
Cannot use raw packets capabilities.

virtualabs, 2024-10-31:
Thanks for reporting this issue, can you provide at least a minimal working example in order to reproduce it ? It looks like some old code has not been updated (this `auto` attribute was used some time ago and has been reworked to provide a synchronous packet processing feature).

virtualabs, 2024-11-05:
Allright, I see what the issue is. It seems our documentation is not up-to-date with the latest features we implemented in WHAD: this `auto()` method originally implemented only in BLE connectors has been moved into the `WhadDeviceConnector` class in order to generalize this behaviour to all connectors (and domains).

Therefore, instead of calling `auto()`, you must use the `enable_synchronous()` method as shown below:

```python
# Old code: central.auto(False)
# Replaced with the following to disable automatic PDU processing (synchronous mode):
central.enable_synchronous(True)

# To re-enable automatic PDU processing:
central.enable_synchronous(False)
```

I've updated the documentation in branch `bugfix/issue-82` to reflect these changes in the API. If this solution works for you, please let me know.

sacca97, 2024-11-05:
Hi, I guess that works thanks. 
However I gave up into trying raw packets sending since I encountered the same butterfly-related issues (discussed with Romain a while ago).
Thank you again, I will close this.

closed 2024-11-05 by sacca97