whad-team/whad-client #85: Exception with Seeedstudio e5-mini

MKesenheimer, 2024-10-30:
I get an exception if I run the `wsniff` tool with a LoRa-e5-Mini from Seeedstudio:

```
$ wsniff --interface uart0 -w phy
[!] An unexpected exception occured:
Traceback (most recent call last):
  File "/home/kali/tools/ble/whad/venv/lib/python3.12/site-packages/whad/cli/app.py", line 722, in run_app
    application.run()
  File "/home/kali/tools/ble/whad/venv/lib/python3.12/site-packages/whad/tools/wsniff.py", line 162, in run
    sniffer.configuration = configuration
    ^^^^^^^^^^^^^^^^^^^^^
  File "/home/kali/tools/ble/whad/venv/lib/python3.12/site-packages/whad/phy/connector/sniffer.py", line 84, in configuration
    self._enable_sniffing()
  File "/home/kali/tools/ble/whad/venv/lib/python3.12/site-packages/whad/phy/connector/sniffer.py", line 35, in _enable_sniffing
    self.set_frequency(self.__configuration.frequency)
  File "/home/kali/tools/ble/whad/venv/lib/python3.12/site-packages/whad/phy/connector/__init__.py", line 342, in set_frequency
    raise UnsupportedFrequency(frequency)
whad.phy.exceptions.UnsupportedFrequency: (UnsupportedFrequency(...), 2402000000)
```

Before running wsniff, I installed the latest WHAD compatible firmware (STM32WLxx WHAD-compatible firmware version 1.0.0) from https://github.com/whad-team/stm32wlxx-firmware on the device.

Maybe it would be a solution to only give a warning instead of a hard fail due to the exception. Or is something else wrong?

virtualabs, 2024-10-31:
Thanks for this issue, I managed to reproduce this behaviour. And yes, that's unbelievable we forgot to handle this exception in `wsniff` :sweat_smile:. 

There is nothing wrong here except `wsniff` not catching this exception and displaying a nice error message. And indeed, the STM32WL55 only supports sub-GHz frequencies.

We'll fix this pretty soon.

virtualabs, 2024-10-31:
I've fixed this bug in the `bugfix/issue-85` branch, you can check if it works OK on your side too by cloning the repository and testing this specific branch. On my side everything is now correctly handled with a dedicated error message and a gentle program exit instead of an hard fault.

MKesenheimer, 2024-10-31:
Perfect! Looks better now, thank you. 

Btw, I can now sniff LoRa with commands similar to:
```
wsniff --interface uart0 -w phy --lora -f 868100000
```

Great work!

closed 2024-10-31 by MKesenheimer

virtualabs, 2024-11-02:
Perfect, this fix is now merged in the main branch and will be available in the next release. Thank you again for reporting this issue :smile: