whad-team/whad-client #87: Wrong configuration is sent to Seeedstudio LoRa-e5-Mini

MKesenheimer, 2024-10-31:
I think I have another issue to share.
If I set up the device Seeedstudio LoRa-e5-Mini with the following command, a wrong message (presumably the configuration) is sent to the device:

```
$ wsniff --log info --interface uart0 -w phy --lora -f 869525000 -d 1070 -sf 11 -cr 45 -bw 250000 -em
...
INFO:whad.device:sending message <SetLoRaMod bandwidth=250000, cr=0, enable_crc=False, explicit_mode=True, invert_iq=False, preamble_length=12, sf=4> to device <uart0>
```

Notice, that the spreading-factor and the coding rate differ. The datarate is missing.

Also, I do not receive anything, despite having a LoRa transmitter nearby, configured with the above settings.

I am currently on branch `bugfix/issue-85` on commit `1e0c470c920ac64c838af9a91f2064a5413cc39b` with `whad-client`. I installed the 1.0.0 firmware from [stm32wlxx-firmware](https://github.com/whad-team/stm32wlxx-firmware) on the LoRa-e5-Mini.

virtualabs, 2024-11-02:
The message you see is the protobuf configuration message sent to the device, but its `cr` and `sf` fields are values belonging to our protocol definition files (see https://github.com/whad-team/whad-protocol/blob/main/whad/protocol/phy/phy.proto for details). These values correspond to a coding rate of 4/5 and a spreading factor of 11, as requested in your `wsniff` command. Datarate has no meaning when it comes to LoRa demodulation, since it is derived from LoRa's coding rate, spreading factor and bandwidth. You may have forgotten to enable CRC with `-crc`, generally required by the receiver to correctly capture LoRa frames. 

Please note we recently made some changes in our LoRa connector and STM32WL55 compatible firmware as we used to badly manage LoRa's synchronization word due to a specifity of this STM32WL55 SoC (sync word is encoded on two bytes rather than one), and I've just noticed we forgot to make a proper release that includes this modification :man_facepalming: . This is now fixed with the release of version 1.0.1 of our STM32WLxx compatible firmware, see https://github.com/whad-team/stm32wlxx-firmware/releases/tag/v1.0.1.

MKesenheimer, 2024-11-02:
Ah, I see. Thanks for the clarification. I still don't receive anything with `wsniff` from a Meshtastic device. How is your test setup? Maybe I can start from there. In particular, how do you generate LoRa signals that you can sniff with `wsniff`?

virtualabs, 2024-11-03:
If you want to sniff Meshtastic frames, you must also specify the correct synchronization word (0x2B) using the `--sync-word` (or `-w` option) like this:

```
$ wsniff -i uart0 phy --lora -f 869525000  -sf 11 -cr 45 -bw 250000 -em -crc -w "\x2b"
```

I've also developed some tiny tools to interact with meshtastic devices, but it is more a PoC than something ready to use: https://github.com/virtualabs/defcontastic. And be careful, this version uses the old synchronization word (`b424` instead of `2b`).

MKesenheimer, 2024-11-03:
Thank you. I will  give it a try.

MKesenheimer, 2024-11-08:
I did a bit more debugging. Here are some issues I encountered:

- First of all, the `-w` option is listed as `Enable wireshark monitoring` and not as the sync word. Maybe this option is double occupied.
- The `--sync-word` option is missing in the command line help.
- The sync word has to be in the format `2b` and not `\x2b`.

Furthermore, if I use
```
wsniff --log info --interface uart0 -w phy --lora -f 869525000 -sf 11 -cr 45 -bw 250000 -em -crc --sync-word "2b"
```
the program works.
Instead, if I type 
```
wsniff --log info --interface uart0 -w --lora -f 869525000 -sf 11 -cr 45 -bw 250000 -em -crc --sync-word "2b" phy
```
with `phy` at the end as the help menu suggests, I get the following error message:
```
[!] argument domain: invalid choice: '869525000' (choose from 'ble', 'dot15d4', 'esb', 'phy', 'rf4ce', 'unifying', 'zigbee')
[!] unrecognized arguments: --log info --interface uart0 -w --lora -f 869525000 -sf 11 -cr 45 -bw 250000 -em -crc --sync-word 2b phy
[!] You need to specify a domain.
```

Why does it matter were `phy` is placed in the command line argument list?

Also, I am still unable to receive any signals from my Meshtastic device. I use the LoRa presets for my region (European Union 868MHz).

virtualabs, 2024-11-08:
I took some time to try sniffing Meshtastic packets again with a real Meshtastic device and the latest firmware version, and it appears that the LoRa preamble length MUST be set to 8 symbols to correctly receive data on channels using the LONG_FAST preset. The problem is `wsniff` does not provide an option to set the LoRa preamble length :man_facepalming: .

I made a slight modification to our sniffer to allow the user to specify this preamble length parameter and I managed to sniff valid Meshtastic packets with the following command-line:

```
$ wsniff -i uart0 phy --lora -f 869525000  -sf 11 -cr 45 -bw 250000 -em -crc -w "2b" --preamble-length 8
```

I am going to push this modification to the main branch, expect an update soon.

virtualabs, 2024-11-08:
Allright, `main` branch does include this modification and will let you set the LoRa preamble length as shown above. You would be able to sniff Meshtastic packets now :smile:.

MKesenheimer, 2024-11-09:
Finally it works! Thank you so much for your support. 

Maybe it would be helpful for others to include the above command as an example in the documentation.

Keep up the good work!

virtualabs, 2024-11-09:
Thank you again for reporting this issue ! I've added some documentation about this new option in `wsniff` page, but we do have some plans for meshtastic support in a near future so I did not specifically add a line regarding raw Meshtastic packet sniffing.

closed 2024-11-09 by virtualabs