whad-team/whad-client #107: Cannot Sniff zigbee or 802.15.14

gertrudem, 2024-12-20:
On my Ubuntu machine, I installed the whad-client using the second installation method (git cloning), and then running _setup.py install_. I've also flashed a nrf52840 using nrfutil and the latest butterfly firmware like in the docs.

I only verified my dongle's firmware using _whadup_ and seeing it on uart0.
[image]

However, when running the sample code to sniff zigbee or ieee 802.15.4 I get some notices about moved files, and the nothing, even though I know there are several packets going through the air on the channel I'm on.

[image]
[image]

virtualabs, 2024-12-22:
The warnings you get from the `cryptography` module should not alter the way sniffing works (TripleDES is not used in 802.15.4).

I tried to reproduce this on my Debian but everything works pretty fine here :thinking: . I took the same example code (but on channel 11 because my ZigBee network is on this channel) and managed to capture packets as expected.

Can you post the output of the following command when your nRF52 dongle is plugged in your machine: `wup uart0` ? I am not sure if my dongle uses the same version as yours, so it will help me figure out if this issue is caused by the nRF52 firmware, whad-client or the operating system.

gertrudem, 2024-12-24:
I get the exact same result when running the example code on my kali machine. I installedwhad-client using ```sudo python setup.py install```, and it gave me this at the end of the installation:

```
Installed /usr/local/lib/python3.12/dist-packages/whad-1.2.6-py3.12.egg
Processing dependencies for whad==1.2.6
Searching for sphinxcontrib-mermaid==0.9.2
Reading https://pypi.org/simple/sphinxcontrib-mermaid/
Download error on https://pypi.org/simple/sphinxcontrib-mermaid/: [Errno -3] Temporary failure in name resolution -- Some packages may not be found!
Couldn't find index page for 'sphinxcontrib-mermaid' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.org/simple/
Download error on https://pypi.org/simple/: [Errno -3] Temporary failure in name resolution -- Some packages may not be found!
No local packages or working download links found for sphinxcontrib-mermaid==0.9.2
error: Could not find suitable distribution for Requirement.parse('sphinxcontrib-mermaid==0.9.2')
```

[Here](https://www.digikey.ca/en/products/detail/nordic-semiconductor-asa/NRF52840-DONGLE/9491124?gclsrc=aw.ds&&utm_adgroup=&utm_source=google&utm_medium=cpc&utm_campaign=PMax%20Product_Low%20ROAS%20Categories&utm_term=&productid=9491124&utm_content=&utm_id=go_cmp-20291741422_adg-_ad-__dev-c_ext-_prd-9491124_sig-EAIaIQobChMI44LQ79DAigMV_BqtBh2ZqxisEAYYASABEgLh9vD_BwE&gad_source=1&gclid=EAIaIQobChMI44LQ79DAigMV_BqtBh2ZqxisEAYYASABEgLh9vD_BwE&gclsrc=aw.ds) is the model of my nrfs52840, and this is the output of ```wup uart0```:

```
/usr/lib/python3/dist-packages/scapy/layers/ipsec.py:512: CryptographyDeprecationWarning: TripleDES has been moved to cryptography.hazmat.decrepit.ciphers.algorithms.TripleDES and will be removed from this module in 48.0.0.
  cipher=algorithms.TripleDES,
/usr/lib/python3/dist-packages/scapy/layers/ipsec.py:516: CryptographyDeprecationWarning: TripleDES has been moved to cryptography.hazmat.decrepit.ciphers.algorithms.TripleDES and will be removed from this module in 48.0.0.
  cipher=algorithms.TripleDES,
[i] Connecting to device ...
[i] Device details

 Device ID: 3e:67:3c:c2:96:c2:91:c3:a7:70:c3:86:34:c3:bc:c2:9d:3b:c2:a2:3d:c2:bb:c3:ab
 Firmware info: 
 - Author : Romain Cayre
 - URL : https://github.com/whad-team/butterfly
 - Version : 1.0.1

[i] Discovering domains ...
[i] Domains discovered.

 This device supports Bluetooth LE:
 - can sniff data
 - can inject packets
 - can hijack communication
 - can simulate a role in a communication

 List of supported commands:
  - SetBdAddress: can set BD address
  - SniffAdv: can sniff advertising PDUs
  - ReactiveJam: can reactively jam PDU on a single channel
  - SniffConnReq: can sniff a new connection
  - SniffAccessAddress: can detect active connections
  - SniffActiveConn: can sniff an active connection
  - ScanMode: can scan devices
  - CentralMode: can act as a Central device
  - ConnectTo: can initiate a BLE connection
  - SendRawPDU: can send a raw PDU
  - SendPDU: can send a PDU
  - Disconnect: can terminate an active connection (in Central mode)
  - PeripheralMode: can act as a peripheral
  - Start: can start depending on the current mode
  - Stop: can stop depending on the current mode
  - SetEncryption: can enable encryption during a connection
  - HijackMaster: can hijack the Master role in an active connection
  - HijackSlave: can hijack the Slave role in an active connection
  - HijackBoth: can hijack the Master and the Slave role in an active connection
  - PrepareSequence: can prepare a sequence of packets and associate a trigger
  - TriggerSequence: can manually trigger the transmission of a sequence of packets
  - DeleteSequence: can delete a prepared sequence of packets

 This device supports 802.15.4:
 - can sniff data
 - can inject packets
 - can jam communications
 - can simulate a role in a communication
[... 43 lines cut ...]
  - Stop: can stop depending on the current mode
  - SniffPairing: can sniff a pairing process

 This device supports Physical Layer:
 - can sniff data
 - can inject packets
 - can jam communications
 - can not read/write raw packet

 List of supported commands:
  - SetGFSKModulation: can use Gaussian Frequency Shift Keying modulation scheme
  - GetSupportedFrequencies: can return a list of supported frequency ranges
  - SetFrequency: can configure a given frequency
  - SetDataRate: can configure the datarate
  - SetEndianness: can configure the endianness
  - SetTXPower: can configure the transmission power level
  - SetPacketSize: can configure the packet size
  - SetSyncWord: can configure the synchronization word
  - Sniff: can receive arbitrary packets
  - Send: can transmit arbitrary packets
  - Jam: can jam a physical medium
  - Monitor: can monitor a physical medium
  - Start: can start depending on the current mode
  - Stop: can stop depending on the current mode
```

gertrudem, 2024-12-24:
Also, regardless of which butterfly firmware I flash, 1.0.0 or 1.0.1, ```wup uart0``` always displays a version of 1.0.1, I'm sure that I'm doing it correctly, by first putting the nrf52840 into flash mode and waiting for the red light, and then using the command with nrfutil : ```nrfutil dfu usb-serial -pkg butterfly-fwupgrade.zip -p /dev/ttyACM0 -b 115200```until the progress bar reaches 100% without errors

virtualabs, 2024-12-26:
```
Installed /usr/local/lib/python3.12/dist-packages/whad-1.2.6-py3.12.egg
Processing dependencies for whad==1.2.6
Searching for sphinxcontrib-mermaid==0.9.2
Reading https://pypi.org/simple/sphinxcontrib-mermaid/
Download error on https://pypi.org/simple/sphinxcontrib-mermaid/: [Errno -3] Temporary failure in name resolution -- Some packages may not be found!
Couldn't find index page for 'sphinxcontrib-mermaid' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.org/simple/
Download error on https://pypi.org/simple/: [Errno -3] Temporary failure in name resolution -- Some packages may not be found!
No local packages or working download links found for sphinxcontrib-mermaid==0.9.2
error: Could not find suitable distribution for Requirement.parse('sphinxcontrib-mermaid==0.9.2')
```

`Temporary failure in name resolution` looks like some kind of DNS-related error to me, it may explain while the setup failed.

Besides this your dongle seems to work as expected, maybe you are simply listening on the wrong channel. WHAD provides a `wzb-enddevice` command-line tool that implements a scanner, it can help checking the channel your ZigBee network uses:

[image]

gertrudem, 2025-01-02:
`wzb-device ` works, and I see the networks that I want to scan, however when I run the sample code on those channels for the corresponding protocols, they pick up nothing even though I know there is enough traffic.

gertrudem, 2025-01-02:
Would the `Temporary failure in name resolution` for the package that needs to be installed effect the [sample code](https://whad.readthedocs.io/en/latest/zigbee/started.html) in any way?

gertrudem, 2025-01-02:
One additional issue I have with sniffing is that when I use the general `wsniff` command, I get different results from the manual command line vs a python subprocess.

Entering something like: `whsniff -i uart0 -o output.pcapng zigbee -c 11`
into the command line manually will sniff into a pcapng as intended, and can be opened in wireshark to be examined.

But when I execute code like:

```
    command = "wsniff -i uart0 -o output.pcapng zigbee -c 11"
    command = command2.split(" ")

    process = subprocess.Popen(
        command,
        stdout=subprocess.DEVNULL,
        stderr=subprocess.PIPE,
        text=True
   )
    return process
```

it creates output.pcapng but with no contents, no matter the sniffing length. Opening the file as a .txt is just an empty text file.

Additionally, in the subprocess, if I replace `command` with `whsniff -i uart0 zigbee -c 11 | wdump output.pcapng`, it doesn't even create any file.

I am stopping the process both ways by getting the PID and killing it after a timer using `time.sleep()`: 
`os.system(f"sudo kill -SIGTERM {str(process.pid)}")`

virtualabs, 2025-01-07:
I managed to reproduce this issue with `wsniff` and yes, there is a problem when stdout is redirected to `/dev/null` (or to a subprocess pipe as well). 

This problem is caused by WHAD's CLI base class that detects if stdout is piped into another program or file and considers that it should forward packets back and forth when stdout is redirected. This mechanism blocks the sniffing process until the chained tool is started and connects to the first one (`wsniff`) through a Unix socket. In your case, redirecting stdout to `/dev/null` triggers this behaviour with `wsniff` and no packet is logged in the pcap file. 

I'm working on a fix to detect this specific case (stdout redirected to `/dev/null`) in order to avoid this behaviour and save every packet in the target pcap file.

Additionally, I managed to reproduce your issue with `wsniff` piped with `wdump`, and it is caused by subprocess. You need to set the `shell` option to `True` and provide the command line as a string rather than an array, as shown below:

```python
import os
import subprocess
import signal

# Launch wsniff piped with wdump, no output on stdout
command = "wsniff -i uart0 zigbee -c 11 | wdump /tmp/zigbee.pcap"
process = subprocess.Popen(
    command,
    shell=True
)

# Wait for user to press a key
input("Press any key to stop")

# Terminate our process
try:
    os.killpg(os.getpgid(process.pid), signal.SIGINT)
except KeyboardInterrupt:
    pass
```

For now, this cannot work with stdout redirected (as explained above) but a fix is on its way.

gertrudem, 2025-01-07:
Thanks a lot, this really helped out! I toyed around and figured this works best for sniffing into a pcap:

```
import os
import subprocess
import signal
import time

command = "wsniff -i uart0 -o zigbee.pcapng zigbee -c 11"

process = subprocess.Popen(
    command,
    shell = True
)

timerLength = 10
time.sleep(timerLength)

os.killpg(os.getpgid(process.pid), signal.SIGINT)

```

Although, the first post in this chain was about the zigbee/dot15d4 sniffing example code not working. I have tried trying the example code after uninstalling and reinstalling different ways, including the most recent main (not release) version up to 10 am, Jan 7th, 2025. But this works as a substitute. Since you cannot reproduce this error, I think it should be up to you to close this issue. Once again, thanks a ton for your help!

virtualabs, 2025-01-08:
I focused on some other issues you had that led me to other bugs and completely forgot your main issue :sweat_smile:. 

I think the example code misses a `sniffer.start()` instruction to effectively configure the hardware and tell it to sniff packets. The following code works fine on my side:

```python
from whad.device import WhadDevice
from whad.dot15d4 import Sniffer

# Create a compatible device instance
device = WhadDevice.create("uart0")

# Use a sniffer connector
sniffer = Sniffer(device)

# Set channel
sniffer.channel = 11

# Start sniffing
sniffer.start()

# Listen for packets
for packet in sniffer.sniff():
    packet.show()
```

If it works for you too, that simply means our example code is wrong and should be fixed in the doc (and I'm pretty confident that it really is the case).

gertrudem, 2025-01-08:
Ok, adding the `.start()` just made it work. Thank you so much because I won't be having to use subprocesses to automate scanning. 👍

closed 2025-01-08 by gertrudem