whad-team/whad-client #112: winject zigbee/802.15.4 issue

gertrudem, 2025-01-02:
every time I run `winject -i uart0 dot15d4 -c 25 0308e9ffffffff07588e`, I get this as a return 
`[!] Invalid syntax: 0308e9ffffffff07588e`.

The same thing happens when I replace "dot15d4" with "zigbee" and/or choose a different channel number. I know for sure that uart0 is valid.

gertrudem, 2025-01-03:
I got it to run without errors now, but when I look on wireshark the packets I injected, every time the FCS is bad and is 0000.

When I do something like: `winject -i uart0 dot15d4 -c 25 030829ffffffff07588e`, which is a Beacon request packet with a valid FCS. The packet appears to be `030829ffffffff070000` when seen in wireshark, where the last 2 bytes that make up the FCS are turned to 00:00. Thus it's registered with a bad FCS

virtualabs, 2025-01-07:
Thank you for reporting this issue !

Can you give more details about your setup:
* what hardware are you using to inject and monitor packets ?
* can you describe precisely how you perform this packet injection (steps followed, command lines and the resulting output) ?

Wrong FCS can be caused by a hardware device that is not able to capture a packet's FCS value, or possibly a bug in the framework.

virtualabs, 2025-01-07:
I took some time to try to reproduce this issue and experienced the same errors with `winject`. This tool needs some improvements to work flawlessly and I'm working on a series of fixes that will solve most of the issues you had with it.

However, I managed to inject the exact raw Dot15d4 you mentioned (`0308e9ffffffff07588e`) using a nRF52840 while capturing packets with another nRF52840 dongle and did not see any error in the captured PCAP file. I used the following command to inject this packet:

```
$ winject -i uart0 -r dot15d4 -c 16 0308e9ffffffff07588e
```

And the following command to monitor the packets sent over the air:

```
$ wsniff -i uart1 dot15d4 -c 16 | wdump /tmp/zigbee-sniff.pcap
```

I got all the FCS fields OK for every packet captured. The bad FCS you've seen may be a consequence of an error that appears both during injection and reception, especially if you are using a different hardware or sniffer to check the packets sent by `winject`.

virtualabs, 2025-01-08:
After some further research, we indeed have an issue with 802.15.4 FCS in ButteRFly. A fix has been developed and will be merged pretty soon, WHAD client will be also fixed to match the FCS byte order (little-endian) and will display a warning when used with ButteRFly version prior to the 1.1.0 (in development).

gertrudem, 2025-01-08:
Nice, I was also wondering how to use the injection code for IEEE 802.15.4:

```
from whad.device import WhadDevice
from whad.dot15d4 import Sniffer

# Create a compatible device instance
device = WhadDevice.create("uart0")

# Use a default Dot15d4 connector
connector = Dot15d4(device)

# Send 802.15.4 packet
connector.send(b"Hello World !")
```

As I get this error:

```
Traceback (most recent call last):
  File "/home/rfslab/Desktop/test2.py", line 5, in <module>
    device = WhadDevice.create("uart0")
             ^^^^^^^^^^
NameError: name 'WhadDevice' is not defined
```

virtualabs, 2025-01-09:
That's weird, `WhadDevice` is correctly imported in your python script. Looks like an error related to Python and not WHAD directly.

gertrudem, 2025-01-10:
I looked a bit into the code and here's what I found:

in `whad-client/whad/dot15d4/connector/sniffer.py` like the example code, there is no Dot15d4 class, so you import Dot15d4 instead from `whad-client/whad/dot15d4/connector/__init__.py` because you get the previous `NameError`:

```
from whad.device import WhadDevice
from whad.dot15d4 import Dot15d4
``` 

However, after that I get this error:

```
Traceback (most recent call last):
  File "/home/rfslab/Desktop/injecter.py", line 16, in <module>
    connector.send(b'Hello World')
  File "/home/rfslab/miniconda3/lib/python3.12/site-packages/whad/dot15d4/connector/__init__.py", line 220, in send
    if Dot15d4FCS not in pdu:
       ^^^^^^^^^^^^^^^^^^^^^
TypeError: a bytes-like object is required, not 'Packet_metaclass'
```
Probably beacause `send()` in `whad-client/whad/dot15d4/connector/__init__.py` expects a `scapy.layers.dot15d4.Dot15d4, scapy.layers.dot15d4.Dot15d4FCS` input instead of something like `b'Hello World'.`

So I tried to craft a scapy packet as an input parameter:
**### BRPacket.py:**

```
from scapy.all import *
from scapy.layers.dot15d4 import *
from scapy.layers.zigbee import *
def beacon_request(seqnum=0):

    dot15d4 = scapy.layers.dot15d4.Dot15d4()
    dot15d4.fcf_frametype = 3
    dot15d4.fcf_security = 0
    dot15d4.fcf_pending = 0
    dot15d4.fcf_ackreq = 0
    dot15d4.fcf_panidcompress = 0
    dot15d4.fcf_reserved_1 = 0
    dot15d4.fcf_destaddrmode = 2
    dot15d4.fcf_framever = 0
    dot15d4.fcf_srcaddrmode = 0
    dot15d4.seqnum = seqnum

    dot15d4_fcs = Dot15d4FCS()
    dot15d4_fcs.fcf_frametype= 3
    dot15d4_fcs.fcf_destaddrmode= 2
    dot15d4_fcs.seqnum = seqnum

    return (dot15d4, dot15d4_fcs)
```

But when I try to run an adapted version of the sample code:
```
from whad.device import WhadDevice
from whad.dot15d4 import Dot15d4

import BRPacket

device = WhadDevice.create("uart1")

connector = Dot15d4(device)
channel = 19

packet = BRPacket.beacon_request(223)

connector.send(packet, channel)
```

I get another error that I'm not sure how to handle:
```
Traceback (most recent call last):
  File "/home/rfslab/Desktop/injecter.py", line 16, in <module>
    connector.send(packet, channel)
  File "/home/rfslab/miniconda3/lib/python3.12/site-packages/whad/dot15d4/connector/__init__.py", line 221, in send
    packet = Dot15d4FCS(raw(pdu) + Dot15d4FCS().compute_fcs(raw(pdu)))
                        ^^^^^^^^
  File "/home/rfslab/miniconda3/lib/python3.12/site-packages/scapy/compat.py", line 294, in raw
    return bytes(x)
           ^^^^^^^^
TypeError: 'Dot15d4' object cannot be interpreted as an integer
```

virtualabs, 2025-01-12:
I see, this example had worked at some point it looks like we reworked the 802.15.4 API and did not update it.
Also, the documentation of `whad.dot15d4.Dot15d4.send()` method seems confusing, as it only expects a packet of type `scapy.layers.dot15d4.Dot15d4` or `scapy.layers.dot15d4.Dot15d4FCS` and not a tuple. 

I've tested sending a 802.15.4 frame with the following code and it worked pretty well:

```python
from scapy.all import *
from scapy.layers.dot15d4 import *
from scapy.layers.zigbee import *
from whad.device import WhadDevice
from whad.dot15d4 import Sniffer, Dot15d4

def beacon_request(seqnum=0):
    pkt = scapy.layers.dot15d4.Dot15d4()
    pkt.fcf_frametype = 3
    pkt.fcf_security = 0
    pkt.fcf_pending = 0
    pkt.fcf_ackreq = 0
    pkt.fcf_panidcompress = 0
    pkt.fcf_reserved_1 = 0
    pkt.fcf_destaddrmode = 2
    pkt.fcf_framever = 0
    pkt.fcf_srcaddrmode = 0
    pkt.seqnum = seqnum

    return pkt

# Create a compatible device instance
device = WhadDevice.create("uart0")

# Use a default Dot15d4 connector
connector = Dot15d4(device)

# Send 802.15.4 packet
packet = beacon_request(223)

connector.send(packet, channel=19)
``` 

Note that I used the fixed version of WHAD as well as a fixed version ButteRFly, but since it is a Python script you should not experience any error with FCS. 

For the record, sending raw packets as byte buffers must be done by calling `send_mac()` and not `send()`. `send_mac()` accepts a buffer of bytes and you can also specify if you want the FCS to be automatically added or not. It is not correctly documented but I will do a code refactoring of the `whad.dot15d4` module soon (like I did with `whad.ble` or `whad.device` in 2024).

gertrudem, 2025-01-13:
Nice! This fixed the my issue. Thanks so much for the explination.

gertrudem, 2025-01-13:
Could you give me an example of how send_mac() works with a scapy packet? I can the the contents of a scapy packet in byte form: `raw(packet)`, or `packet.build()`. However, neither of these work when passed as pdu:
```
Traceback (most recent call last):
  File "/home/Desktop/injector.py", line 19, in <module>
    connector.send_mac(raw(packet), channel)
  File "/home/miniconda3/lib/python3.12/site-packages/whad/dot15d4/connector/__init__.py", line 248, in send_mac
    packet = pdu / raw(b'\x00\x00')
             ~~~~^~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for /: 'bytes' and 'bytes'
```

virtualabs, 2025-01-14:
The current version of `send_mac()` I've been working for while solving this issue did correctly managed bytes buffer but I took some time to rework other parts of WHAD in order to allow raw 802.15.4 frames without FCS, even if the hardware does expect an FCS value (basically by moving the last 2 bytes of payload into this FCS field :smile: ). It is a bit tricky but now I can send raw frames with `send_mac()`. It needs more testing but I guess it would be helpful to handle protocols that rely on 802.15.4 MAC with no FCS support.

gertrudem, 2025-01-14:
I've tried this branch and it seems to be working well with the new firmware. Although I still get a message telling me to upgrade to the latest firmware even though I flashed mine with `dfu.zip`. Thanks, I can send raw bytes from scapy packets now!

closed 2025-01-14 by gertrudem