whad-team/whad-client #118: How to stop sniffing if there are no packets

gertrudem, 2025-01-14:
in this modified example code:

```
from whad.device import WhadDevice
from whad.zigbee import Sniffer

import time

# Create our whad device object
device = WhadDevice.create("uart0")

# Create our ZigBee sniffer instance
sniffer = Sniffer(device)

sniffer.channel = 19

sniffer.start()

start = time.time()

for packet in sniffer.sniff():
    packet.show()
    if time.time() - start >= 10:
        break

sniffer.stop()
```

Is there a way to stop sniffing 10 seconds after starting? Because the for loop only iterates if a packet is seen, so if no packets are ever seen, the for loop will continue to wait indefinitely for a packet, but not iterate. I know I could just use the `wsniff` command as a subprocess, and the kill it to get some data, but I just wanted to see if this was possible. 

I've played around with some multiprocessesing, but nothing seems to work. The device and sniffer object have to be declared in the same function or space, and sniffer.stop() has to be called after every scan if it needs to be used again reliably, which can only happen after the for loop finishes running, which can only happen after the timer.

Right now I have a horrible solution by sending a custom packet to my sniffer that it recognizes as a signal to exit the loop. But I just want to find a way to exit the sniffing loop even if no packets are seen.

virtualabs, 2025-01-15:
You're right, most of the `Sniffer` connectors do not accept a timeout in parameter and that sucks. I'm working on it :+1:.

virtualabs, 2025-01-15:
I've reworked our `Sniffer` connectors in order to support a `timeout` parameter for all of them, and pushed this modification into the `bugfix/issue-118` branch. You should give it a try with the following code:

```python
from whad.device import WhadDevice
from whad.zigbee import Sniffer

import time

# Create our whad device object
device = WhadDevice.create("uart0")

# Create our ZigBee sniffer instance
sniffer = Sniffer(device)

sniffer.channel = 19

sniffer.start()

start = time.time()

for packet in sniffer.sniff(timeout=10.0):
    packet.show()

sniffer.stop()
```

Timeout is now handled by the `sniff()` method and will be enforced even if no packet has been received.

gertrudem, 2025-01-16:
Great! This is way better than the solution I had implemented. It's been working find so far on what I'm working on.

One last thing. The zigbee channel scanner ran by using `wzb-enddevice` -> `scan`, how can I automate that in a shell script if you happen to know. Using a shell script to enter `wzb-enddevice -i uart0` is easy, but I can't find a way to enter `scan` into the subsequent zigbee cli. Would you know a solution?

virtualabs, 2025-01-17:
For now `wzb-enddevice` does not provide any command that can be launched like `wzb-enddevice -i uart0 scan` but this could be a nice improvement for scripting. 

I'm merging this fix into `main` and will close the issue once done. Feel free to create a new issue to ask for this scan command feature in `wzb-enddevice` in order for us to track it.

virtualabs, 2025-01-17:
Merged branch `bugfix/issue-118` (PR #125) into `main` branch, closing this issue.

closed 2025-01-17 by virtualabs