whad-team/whad-client #128: Missing aarch64 dependencies

XenoKovah, 2025-01-22:
I was just trying to install WHAD (from git-checked-out copy) on an ARM Ubuntu 24.04 VM (on ARM-based Mac) for our BT class, and I ran across the following errors regarding missing dependencies. I don't know how pip install works, but I'm assuming these can be solved with an update to the configuration somehow to make it just work like it does on x86?

```
pip install .
<snip>
      writing /tmp/pip-install-z81opjck/evdev_0bf6d5075cb04d13ac3912d023c4dfbd/evdev/ecodes.c (using /usr/include/linux/input.h /usr/include/linux/input-event-codes.h /usr/include/linux/uinput.h)
      building 'evdev._input' extension
      creating build/temp.linux-aarch64-cpython-312/evdev
      aarch64-linux-gnu-gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O2 -Wall -fPIC -I/home/user/BT1001/whad-client/venv/include -I/usr/include/python3.12 -c evdev/input.c -o build/temp.linux-aarch64-cpython-312/evdev/input.o -std=c99 -Wno-error=declaration-after-statement
      error: command 'aarch64-linux-gnu-gcc' failed: No such file or directory
      [end of output]
```

Solved with `sudo apt-get install gcc-aarch64-linux-gnu -y`

```
pip install .
<snip>
      aarch64-linux-gnu-gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O2 -Wall -fPIC -I/home/user/BT1001/whad-client/venv/include -I/usr/include/python3.12 -c evdev/input.c -o build/temp.linux-aarch64-cpython-312/evdev/input.o -std=c99 -Wno-error=declaration-after-statement
      evdev/input.c:10:10: fatal error: Python.h: No such file or directory
         10 | #include <Python.h>
            |          ^~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
```

Solved with `sudo apt install python3-dev -y`

```
pip install .
<snip>
     aarch64-linux-gnu-gcc -fno-strict-overflow -Wsign-compare -DNDEBUG -g -O2 -Wall -fPIC -I/home/user/BT1001/whad-client/venv/include -I/usr/include/python3.12 -c evdev/input.c -o build/temp.linux-aarch64-cpython-312/evdev/input.o -std=c99 -Wno-error=declaration-after-statement
      aarch64-linux-gnu-gcc: fatal error: cannot execute ‘as’: execvp: No such file or directory
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]
```

Solved with `sudo apt install binutils -y`

After that "pip install ." completed without error.

virtualabs, 2025-01-23:
These missing dependencies are missing from the documentation, we need to update the install procedure to include these requirements. Thanks for reporting this ! 😄

virtualabs, 2025-01-23:
Added some specific instructions on required packages and how to install them on a Debian/Ubuntu system (x64/aarch64) in PR #132.

virtualabs, 2025-01-24:
Up-to-date (synced with `main` branch) is available here: https://whad.readthedocs.io/en/latest/install.html. And yes, I may have fixed this too quickly, I'll add a system prerequisites section with all these details.

virtualabs, 2025-01-24:
Moved system dependencies into a dedicated section before pip and github clone instructions (see #135). Also added a note telling readers that they can file an issue if some specific instructions are required for other systems.

XenoKovah, 2025-01-27:
New documentation LGTM!

closed 2025-01-27 by XenoKovah