whad-team/whad-client #129: Unknown issues when connecting WHAD to WHAD (wble-central to wble-spawn)

XenoKovah, 2025-01-22:
I was trying to get WHAD up and running, but it wasn't working on my ARM system. I was suspicious that was possibly some underlying issue related to ticket #128. So I fell back to testing it on an Ubuntu 24.04 x86 VM. But it still had problems. So tried doing a very simple test that is hopefully easy to replicate (run wble-spawn to take a peripheral and then connect to it and look at the pcap) but even that was having issues, so I don't know what if anything I'm doing wrong.

## Target Virtual Peripheral

I used LightBlue on an iPhone to create a default Blood Pressure virtual peripheral.

## Terminal 1:

I successfully capture the virtual blood pressure profile as virtual-BP.json.

```
wble-central -i hci1 -b 66:F9:3B:4D:34:46 -r profile virtual-BP.json
Searching for target device 66:F9:3B:4D:34:46 ...
Connecting to target device ...
Enumerating services and characteristics ...
<snip>
Writing profile JSON data to virtual-BP.json ...
```

I set up `wble-spawn` to act like the virtual-BP.json, and pipe it to `wdump` (because it won't let me *not* pipe it to anything, even though it seems like it should allow that if I just want to capture the pcap out of band with Sniffle?). After this it is advertising the virtual-BP on BDADDR 11:11:11:11:11:11 (confirmed with Sniffle.)

```
(venv) root@BTVM:/home/user/BT1001/whad-client# hciconfig 
hci1:	Type: Primary  Bus: USB
	BD Address: 22:22:22:22:22:22  ACL MTU: 310:10  SCO MTU: 64:8
	UP RUNNING 
	RX bytes:696 acl:0 sco:0 events:49 errors:0
	TX bytes:3168 acl:0 sco:0 commands:49 errors:0

hci0:	Type: Primary  Bus: USB
	BD Address: 11:11:11:11:11:11  ACL MTU: 310:10  SCO MTU: 64:8
	UP RUNNING 
	RX bytes:1742 acl:0 sco:0 events:98 errors:0
	TX bytes:5976 acl:1 sco:0 commands:108 errors:0
(venv) root@BTVM:/home/user/BT1001/whad-client# wble-spawn -i hci0 --log debug --log-file /tmp/wble-spawn.log -p virtual-BP.json | wdump --log debug --log-file /tmp/wdump.log --force bla.pcap
/!\ ble-spawn stopped (CTL-C)                                
0 packets have been dumped into bla.pcap
```

## Terminal 2:

On the second terminal I want to connect and perform the profiling using a second HCI interface.

```
(venv) root@BTVM:/home/user/BT1001/whad-client# wble-central -i hci1 --log debug --log-file /tmp/wble-central.log -b 11:11:11:11:11:11 profile virtual-BP-WHAD.json
Searching for target device 11:11:11:11:11:11 ...
Connecting to target device ...
[!] BLE peripheral 11:11:11:11:11:11 cannot be found.
```

This seems to have timed out after 30 seconds or so. This is the core failure, but I don't know if it's `wble-central` or `wble-spawn`'s fault.

## Terminal 3: Sniffle pcap

I ran Sniffle in a 3rd terminal, listening for BDADDR 11:11:11:11:11:11 specifically. It saw the connection attempt, and it seemed to think it succeeded, because it *does* see *empty* packets from the peripheral to central. Just never any non-empty ones :-/. This also then perhaps explains why `wdump` says that 0 packets were dumped.

[image]

[image]

## Logs

Attached are the .log files from the above commands + Sniffle pcap. Thanks

[wdump.log](https://github.com/user-attachments/files/18510511/wdump.log)
[wble-spawn.log](https://github.com/user-attachments/files/18510509/wble-spawn.log)
[wble-central.log](https://github.com/user-attachments/files/18510510/wble-central.log)

pcap:
[whad-test.tar.gz](https://github.com/user-attachments/files/18510544/whad-test.tar.gz)

virtualabs, 2025-01-23:
`wble-spawn` creates a BLE device but does not embed any BLE stack, so piping it to `wdump` works but you may only see some packets coming the connected Central device, related to services and characteristics discovery but nothing more (especially if you are using HCI adapters).
If you want to emulate a peripheral with a working BLE stack, you can use `wble-periph` (see https://whad.readthedocs.io/en/latest/cli/ble/ble-periph.html#importing-an-existing-profile).

When using `wble-central` you need to specify if you are trying to connect to a device that uses a public BD address (default) or a random BD address (requires the `--random` option), this may explain why `wble-central` cannot find the target device. If this is a public BD address, there is clearly a problem with WHAD. 

Of course, there may still be errors in WHAD and maybe the problem you are facing is due to some kind of error, so feel free to give more details on your attempts and these errors here 😃.

XenoKovah, 2025-01-23:
Got it. I thought `wble-spawn` was a full-featured peripheral meant to emulate the given profile. I'll try again with `wble-periph` and then move back to my other tests. Thanks

virtualabs, 2025-01-23:
`wble-spawn` is intended to be used for setting MitM when combined with `wble-connect`, but I think we can do something to make it work with `wble-periph` the same way we can pipe `wble-connect` with `wble-central` to query a device.

XenoKovah, 2025-01-23:
Misc comment re:

> When using wble-central you need to specify if you are trying to connect to a device that uses a public BD address (default) or a random BD address (requires the --random option), this may explain [...]

First, IIRC I had tried both with and without `--random` and the results were the same. But I think that the core issue was probably what you identified about `wble-spawn` not being supposed to be used this way.

But second, this also made me realize I didn't actually know how to determine in Linux whether a hciX adapter was using a public or random address, I had just _assumed_ that since I had forced a hardcoded spoofed address that it would be public. I don't see any obvious arguments to hciconfig to determine the answer, and I didn't want to have to read through the HCI log to see if I saw a "LE Set Random Address" command during running WHAD. So it turns out that this can be seen with `bluetoothctl show <BDADDR>`, and for both my CSR dongles with all 1s and all 2s they're public addresses.

XenoKovah, 2025-01-23:
OK, a couple comments/questions about substituting `wble-periph` for `wble-spawn` for doing just a basic test.

1) When I just literally swapped the one for the other in my commands, I was surprised that `wble-periph` seems to default to interactive mode now, even when used with -p (which I thought would make it just run immediately with the information within the profile, rather than dropping into interactive mode). The [documentation](https://whad.readthedocs.io/en/stable/cli/ble/ble-periph.html) made it look like one would have to explicitly pass an `interactive` CLI argument for that to be the case. Did the default behavior change? (And therefore the docs need updating?)

2) Once in interactive mode, I figured I'd just do `start` and it'd start behaving like a second "Blood Pressure" instance, just advertising on 11:11:11:11:11:11. I was therefore surprised when I did a scan and found it advertising as "WhadDev" (since it had correctly advertised as "Blood Pressure" with `wble-spawn`. I double checked the JSON profile, and confirmed it had the correct advertising bytes contained therein. (426C6F6F64205072657373757265) So I think this means `wble-periph` is not parsing and extracting short/complete name fields from an input profile?

3) I also briefly (incorrectly) thought that perhaps it was getting the advertised name from the GATT Device Name (0x2a00) characteristic value. (I realized this was not the case once I saw the `shortname` command which made it clear name `name` and `shortname` corresponded to their AdvData counterparts.) But the side effect is that it made me realize that actually it doesn't look like that is being collected when I run `whad-central profile`, which seems like it's possible a bug? (Or was the thinking that you'd never need to capture the actual values because you'd always just forward requests on to the original peripheral e.g. like when doing MITM?) Below is an example of the commands to grab the profile, but you will see in the output that there's no device name collected. However I then use `gatttool` to confirm it can read the device name from the virtual peripheral fine.

```
(venv) user@BTVM:~/BT1001/whad-client$ wble-central -i hci1 scan on
 RSSI Lvl  Type  BD Address        Extra info
[-065 dBm] [RND] 51:8c:42:09:7d:b3 name:"Blood Pressure"
<snip>
Scan terminated by user
(venv) user@BTVM:~/BT1001/whad-client$ wble-central -i hci1 -b 51:8c:42:09:7d:b3 -r profile virtual-BP.json
Searching for target device 51:8c:42:09:7d:b3 ...
Connecting to target device ...
Enumerating services and characteristics ...
Service Generic Access (0x1800)  (handle 1 to 5)
  Device Name (0x2A00) R, handle 2, value handle: 3
  Appearance (0x2A01) R, handle 4, value handle: 5

Service Generic Attribute (0x1801)  (handle 6 to 9)
  Service Changed (0x2A05) I, handle 7, value handle: 8
    Descriptor type Client Characteristic Configuration (0x2902), handle: 9

Service Device Information (0x180A)  (handle 10 to 14)
  Manufacturer Name String (0x2A29) R, handle 11, value handle: 12
  Model Number String (0x2A24) R, handle 13, value handle: 14

Service d0611e78-bbb4-4591-a5f8-487910ae4366  (handle 15 to 19)
  8667556c-9a37-4c91-84ed-54ee27d90049 WN, handle 16, value handle: 17
    Descriptor type Client Characteristic Configuration (0x2902), handle: 19

Service 9fa480e0-4967-4542-9390-d343dc5d04ae  (handle 20 to 24)
  af0badb1-5b99-43cd-917a-a77bc549e3cc WN, handle 21, value handle: 22
    Descriptor type Client Characteristic Configuration (0x2902), handle: 24

Service Battery (0x180F)  (handle 25 to 28)
  Battery Level (0x2A19) RN, handle 26, value handle: 27
    Descriptor type Client Characteristic Configuration (0x2902), handle: 28

Service Current Time (0x1805)  (handle 29 to 34)
  Current Time (0x2A2B) RN, handle 30, value handle: 31
    Descriptor type Client Characteristic Configuration (0x2902), handle: 32
  Local Time Information (0x2A0F) R, handle 33, value handle: 34

Service 7905f431-b5ce-4e99-a40f-4b1e122d00d0  (handle 35 to 44)
  69d1d8f3-45e1-49a8-9821-9bbdfdaad9d9 W, handle 36, value handle: 37
  9fbf120d-6301-42d9-8c58-25e699a21dbd N, handle 39, value handle: 40
    Descriptor type Client Characteristic Configuration (0x2902), handle: 41
  22eac6e9-24d6-4bb5-be44-b36ace7c7bfb N, handle 42, value handle: 43
    Descriptor type Client Characteristic Configuration (0x2902), handle: 44

Service 89d3502b-0f36-433a-8ef4-c502ad55f8dc  (handle 45 to 56)
  9b3c81d8-57b1-4a8a-b8df-0e56f7ca51c2 WN, handle 46, value handle: 47
    Descriptor type Client Characteristic Configuration (0x2902), handle: 49
  2f7cabce-808d-411f-9a0c-bb92ba96c102 WN, handle 50, value handle: 51
    Descriptor type Client Characteristic Configuration (0x2902), handle: 53
  c6b2f38c-23ab-46d8-a6ab-a3a870bbd5d7 RW, handle 54, value handle: 55

Service Blood Pressure (0x1810)  (handle 57 to 65)
  Blood Pressure Feature (0x2A49) R, handle 58, value handle: 59
  Intermediate Cuff Pressure (0x2A36) N, handle 60, value handle: 61
    Descriptor type Client Characteristic Configuration (0x2902), handle: 62
  Blood Pressure Measurement (0x2A35) I, handle 63, value handle: 64
    Descriptor type Client Characteristic Configuration (0x2902), handle: 65

Writing profile JSON data to virtual-BP.json ...
(venv) user@BTVM:~/BT1001/whad-client$ gatttool -i hci1 -b 51:8c:42:09:7d:b3 -t random --characteristics
[... cut ...]
handle = 0x003a, char properties = 0x02, char value handle = 0x003b, uuid = 00002a49-0000-1000-8000-00805f9b34fb
handle = 0x003c, char properties = 0x10, char value handle = 0x003d, uuid = 00002a36-0000-1000-8000-00805f9b34fb
handle = 0x003f, char properties = 0x20, char value handle = 0x0040, uuid = 00002a35-0000-1000-8000-00805f9b34fb
(venv) user@BTVM:~/BT1001/whad-client$ gatttool -i hci1 -b 51:8c:42:09:7d:b3 -t random --char-read -a 3
Characteristic value/descriptor: 69 50 68 6f 6e 65 
```

that 0x2a00 value just translates to "iPhone" (since it's a virtual peripheral running on an iPhone).

virtualabs, 2025-01-24:
1. You're right. I fixed the documentation while working on issue #133, adding the missing `--profile` option in `wble-periph` and removing the `interactive` command as it is no more required.
2. I checked it and yes, this is definitely missing. We may infer a device name based on its profile advertising and scan response data (and we have Python code for that, so that's not really an issue). I suppose I'm used to the `name` command and used it when I was testing the tool 😅 
3. Should we read the device name from the corresponding Device name characteristic instead of relying on the advertised short or complete name ?

Besides this, would it be a good idea to save the advertised device name in metadata ? I don't know how you handle it in BTIDES, I'll have to have a look at the schema.

XenoKovah, 2025-01-24:
Re 3: My feeling is that you *shouldn't* use an advertised name as the Device Name (2a00) property, as I've seen them differ plenty of times. My recommendation is primarily that `whad-central profile` should be reading any readable characteristics and storing that data as part of the profile JSON file. That way the virtual peripheral can appear like a real peripheral if anyone runs GATT enumeration against it (i.e. with another instance of `whad-central profile` or equivalents like `gatttool`, Blue2thprinting, etc.). I do capture information to that depth (and beyond) in BTIDES. When I get a chance I'll provide a 1:1 comparison of the WHAD JSON view of a GATT hierarchy vs. the same thing in BTIDES.

virtualabs, 2025-01-24:
I've started working on some kind of helper class able to access/modify some key AD records (complete local name, shortened local name, etc.) and modified `wble-periph` to use this helper. It can now retrieve and modify the device name from the advertisement data if you import an existing profile.

There's still a lot of work to get something clean and somehow a bit clever (currently it manages to insert a new AD record in the advertising data or scan response data depending on its size, but does not find the best order to fit everything in 2*31 bytes), but I think it will do the job.

And you're right, `wble-central` must read all characteristic values and store them into the saved JSON.

virtualabs, 2025-01-25:
Took some time to upgrade the advertising data manager of `wble-periph`. Now it uses the provided advertising data if you give it a specific profile with `-p` until you modify them. If advertising data is modified, it arranges the record to optimize space in a transparent way so you don't have to bother what goes in the advertisement or in the scan response. It is designed to let the user focus on the AD records rather than finding how to make them fit. Still some testing to do, but seems to work pretty neat.

virtualabs, 2025-01-26:
Aaaaand I've just modified the GATT stack layer to automatically read values of readable characteristics if requested when performing a discovery procedure and enabled it for GATT profile export. Also added a *data* property in saved characteristics information to save each value. Tried it with `wble-periph` and everything is loaded fine 😄. Again, it surely needs more testing.

I've pushed all these fixes to branch `bugfix/issue-129`:
- `wble-periph` now loads the device name from the provided GATT JSON file (if specified with `-p`)
- `wble-periph` advertising data management has been completely reworked to optimize how AD records are sorted in advertisement and scan response data
- `wble-central` saves every characteristic's value in JSON file (only for readable characteristic of course)
- `wble-periph` uses the saved characteristics's values if available in the JSON file by default

I know there is still some modifications to bring to `wble-periph` but feel free to give a try to the improved version I'm working on ! And of course, I would be more than happy to consider any additional remark or suggestion 😄

XenoKovah, 2025-01-27:
I gave the `bugfix/issue-129` branch a try, but it errors out when it receives an ATT_ERROR_RSP of type "Insufficient Authentication".

```
wble-central -i hci0 --log debug --log-file ./profiling.log -b 5c:84:a3:ba:ec:35 -r profile Virtual-BP.json
Searching for target device 5c:84:a3:ba:ec:35 ...
Connecting to target device ...
Enumerating services and characteristics ...
[!] ATT error: insufficient authentication
```

For iPhones you're basically always guaranteed to get that error when you try to read from some characteristics which might otherwise be readable on other devices (like 2a19 Battery Level), if you're not paired with the device. I think the best response to that in the context of `wble-central profile` is probably to just ignore such errors.

On that front I'm also attaching an two examples of what a BTIDES profile for that device looks like when I do GATT enumeration on the same device. The GATT enumeration is done with a Sniffle-based tool that I haven't released yet since I'm still testing it, but its basic purpose is trying to work towards optimizing getting the GATT and LL 2thprint information as fast as possible. (Hence my question [here](https://github.com/nccgroup/Sniffle/issues/108) in that I want to also upgrade to 2M PHY when possible.)

[VirtualBP-full-expanded.btides.json](https://github.com/user-attachments/files/18559406/VirtualBP-full-expanded.btides.json)

The `VirtualBP-full-expanded.btides.json` is basically all the information I can currently extract from the PCAP via [PCAP_to_BTIDES](https://github.com/darkmentorllc/Blue2thprinting/blob/BTIDES/Analysis/PCAP_to_BTIDES.py) (and with the `--verbose-BTIDES` option given for increased human-readability).

[VirtualBP-profile-expanded.btides.json](https://github.com/user-attachments/files/18559407/VirtualBP-profile-expanded.btides.json)

The `VirtualBP-profile-expanded.btides.json` is just that file cut down to remove everything except the GATT + Advertisement information for clarity, which basically equates to the data seen from `wble-central profile`. You may notice I captured a lot more "Read Not Permitted" errors. That's because I try to read from every handle, not just characteristic values which have readable property. (I'm looking for things which don't behave according to their stated properties. That's probably not something you should do in the WHAD profiling context.)

[Virtual-BP-WHAD-profie-expanded.json](https://github.com/user-attachments/files/18559389/Virtual-BP-WHAD-profie-expanded.json)

And then I attached a copy of `Virtual-BP-WHAD-profie-expanded.json` which corresponds to the profile data collected from an earlier version of WHAD before the `insufficient authentication` abort. You can see we're structuring things _roughly_ the same wrt GATT data. But for advertisement data BTIDES pulls them out to individual per-AdvData types.

XenoKovah, 2025-01-27:
I gave the `bugfix/issue-129` branch a try, but it errors out when it receives an ATT_ERROR_RSP of type "Insufficient Authentication".

```
wble-central -i hci0 -b 5c:84:a3:ba:ec:35 -r profile Virtual-BP.json
Searching for target device 5c:84:a3:ba:ec:35 ...
Connecting to target device ...
Enumerating services and characteristics ...
[!] ATT error: insufficient authentication
```

For iPhones you're basically always guaranteed to get that error when you try to read from some characteristics which might otherwise be readable on other devices (like 2a19 Battery Level), if you're not paired with the device. I think the best response to that in the context of `wble-central profile` is probably to just ignore such errors.

On that front I'm also attaching an two examples of what a BTIDES profile for that device looks like when I do GATT enumeration on the same device. The GATT enumeration is done with a Sniffle-based tool that I haven't released yet since I'm still testing it, but its basic purpose is trying to work towards optimizing getting the GATT and LL 2thprint information as fast as possible. (Hence my question [here](https://github.com/nccgroup/Sniffle/issues/108) in that I want to also upgrade to 2M PHY when possible.)

[VirtualBP-full-expanded.btides.json](https://github.com/user-attachments/files/18559406/VirtualBP-full-expanded.btides.json)

The `VirtualBP-full-expanded.btides.json` is basically all the information I can currently extract from the PCAP via [PCAP_to_BTIDES](https://github.com/darkmentorllc/Blue2thprinting/blob/BTIDES/Analysis/PCAP_to_BTIDES.py) (and with the `--verbose-BTIDES` option given for increased human-readability).

[VirtualBP-profile-expanded.btides.json](https://github.com/user-attachments/files/18559407/VirtualBP-profile-expanded.btides.json)

The `VirtualBP-profile-expanded.btides.json` is just that file cut down to remove everything except the GATT + Advertisement information for clarity, which basically equates to the data seen from `wble-central profile`. You may notice I captured a lot more "Read Not Permitted" errors. That's because I try to read from every handle, not just characteristic values which have readable property. (I'm looking for things which don't behave according to their stated properties. That's probably not something you should do in the WHAD profiling context.)

[Virtual-BP-WHAD-profie-expanded.json](https://github.com/user-attachments/files/18559389/Virtual-BP-WHAD-profie-expanded.json)

And then I attached a copy of `Virtual-BP-WHAD-profie-expanded.json` which corresponds to the profile data collected from an earlier version of WHAD before the `insufficient authentication` abort. You can see we're structuring things _roughly_ the same wrt GATT data. But for advertisement data BTIDES pulls them out to individual per-AdvData types.

virtualabs, 2025-01-27:
Oh, never tried it against a device that requires pairing because I don't have one here 😭. I'll try to borrow an iPhone from a friend in order to reproduce this crash and fix it. Have you tried on a "normal" device that does not have authenticated characteristics, just to see if it works fine on your side ?

XenoKovah, 2025-01-27:
I was thinking it was just a perfunctory test, but actually it turns out for a random whitelabel [lightbulb](https://amzn.to/4jAibJu) I have (which advertises as "ZenggeMesh" and where incidentally I remember adding to CLUES that the `00010203-0405-0607-0809-0a0b0c0d1910` is the TeLink (proprietary) mesh service :)), it looks like its succeeding in the `main` branch, but 100% failing in `checkout bugfix/issue-129`

```
(venv) root@BTVM:/home/user/BT1001/whad-client# wble-central -i hci0 -b <snip> profile lightbulb-ZenggeMesh.json
Searching for target device <snip> ...
Connecting to target device ...
Enumerating services and characteristics ...
Service Generic Access (0x1800)  (handle 1 to 6)
  Device Name (0x2A00) R, handle 2, value handle: 3
  Appearance (0x2A01) R, handle 5, value handle: 6

Service Device Information (0x180A)  (handle 7 to 15)
  Firmware Revision String (0x2A26) R, handle 8, value handle: 9
  Manufacturer Name String (0x2A29) R, handle 10, value handle: 11
  Model Number String (0x2A24) R, handle 12, value handle: 13
  Hardware Revision String (0x2A27) R, handle 14, value handle: 15

Service 00010203-0405-0607-0809-0a0b0c0d1910  (handle 16 to 28)
  00010203-0405-0607-0809-0a0b0c0d1911 RWN, handle 17, value handle: 18
  00010203-0405-0607-0809-0a0b0c0d1912 RWW, handle 20, value handle: 21
  00010203-0405-0607-0809-0a0b0c0d1913 RW, handle 23, value handle: 24
  00010203-0405-0607-0809-0a0b0c0d1914 RW, handle 26, value handle: 27

Writing profile JSON data to lightbulb-ZenggeMesh.json ...
(venv) root@BTVM:/home/user/BT1001/whad-client# git checkout bugfix/issue-129
branch 'bugfix/issue-129' set up to track 'origin/bugfix/issue-129'.
Switched to a new branch 'bugfix/issue-129'
(venv) root@BTVM:/home/user/BT1001/whad-client# wble-central -i hci0 -b <snip> profile lightbulb-ZenggeMesh-bugfix-issue-129.json
Searching for target device <snip> ...
Connecting to target device ...
Enumerating services and characteristics ...
[!] GATT Timeout while discovering services and characteristics. Aborted.
(venv) root@BTVM:/home/user/BT1001/whad-client# wble-central -i hci0 -b <snip> profile lightbulb-ZenggeMesh-bugfix-issue-129.json
Searching for target device <snip> ...
Connecting to target device ...
Enumerating services and characteristics ...
[!] GATT Timeout while discovering services and characteristics. Aborted.
(venv) root@BTVM:/home/user/BT1001/whad-client# wble-central -i hci0 -b <snip> profile lightbulb-ZenggeMesh-bugfix-issue-129.json
Searching for target device <snip> ...
Connecting to target device ...
Enumerating services and characteristics ...
[!] GATT Timeout while discovering services and characteristics. Aborted.
```

pcap is not illuminating except that the lightbulb doesn't seem to be replying:

[image]

Attached log (with bdaddr removed since it's public)

[wble-central.log](https://github.com/user-attachments/files/18561823/wble-central.log)

XenoKovah, 2025-01-27:
I feel like there might be something particular about this lightbulb though... it seems like my GATT enumerator is also having issues connecting without it timing out immediately due to non-reply from the peripheral. Whereas for both mine & WHAD, it seems that they're successful with higher probability right after the bulb is rebooted...

XenoKovah, 2025-01-27:
OK I checked against a Bluetooth bathroom scale and profile w/ reads worked fine there.

XenoKovah, 2025-02-02:
Just as a reminder to myself for when I'm checking back on tickets: this issue is done from the perspective of WHAD-to-WHAD connectivity, insofar as once I used the correct invocation it worked fine. 

But the ticket is still open just until it can be tested against a device which sends back an ATT "insufficient authentication" error (e.g. as are common on Apple devices), to ensure that's handled gracefully and that profiling continues, ignoring whichever characteristic yielded that error.