whad-team/whad-client #130: Missing available arguments in wdump documentation

XenoKovah, 2025-01-23:
I just re-opened my VM to see I had a [wdump documentation](https://whad.readthedocs.io/en/stable/cli/generic/wdump.html) tab opened, and realized it doesn't mention it supporting --log or --log-file which I used in #129. I found those via --help which lists the following options which aren't listed on the documentation page:

```
  -h, --help            show this help message and exit
  --interface INTERFACE, -i INTERFACE
                        specifies the WHAD interface to use
  --no-color            disable colors in output
  --log {error,warn,info,debug}
                        set the logging level for this application
  --log-file LOGFILE_PATH
                        write the log output into LOGFILE_PATH
  --version             show program's version number and exit
```

virtualabs, 2025-01-23:
Good catch, and since it is supported in a generic way we should add a little something about debugging CLI tools in the documentation (instead of editing every CLI tool doc). Or is it better to update every CLI tool doc to include these options ? I don't know what's best.

XenoKovah, 2025-01-23:
I would say I have a slight preference to see all CLI args on each tool's page. Basically if you were just to put a link to "How to debug tools." on each page which supported --log and --log-file, my eyes would automatically skip over it and be drawn to the argument listing, so I'd miss it to begin with. The only reason I even saw the existence of these `--log-` and `--log-file` arguments was from looking at other open tickets when trying to figure out what wasn't working for me. (And I see now that they're not actually listed on other pages as well. It's just that I happened to have the wdump page still up this morning.)

You could perhaps save yourself some work if there's a way to include some sort of predefined "ARGS_SUPPORTED_BY_ALL_TOOLS" tag which then gets interpolated out to all of the arguments which are meant to be supported by all tools. (And then if some tool ever is supposed to support it but it got missed, then someone will eventually try to use it and report it doesn't work.)

virtualabs, 2025-01-23:
I agree, I'll take the time required to update the documentation pages of the tools that do support these options.

virtualabs, 2025-01-23:
I've merged #131 into `main` that adds missing debug options to tools documentation pages (for those that support these options).

closed 2025-01-23 by virtualabs