whad-team/whad-client #133: wble-periph documentation missing -p/--profile at top

XenoKovah, 2025-01-23:
While the https://whad.readthedocs.io/en/stable/cli/ble/ble-periph.html#command-line-options documentation mentions -p lower in the examples, it doesn't mention it at the top next to the other CLI args.

virtualabs, 2025-01-24:
Fixed the documentation and merged it into `main`, taking into account the comment you made in #129 about the `interactive` command that is no longer required.

virtualabs, 2025-01-24:
Oops, I should not close this issue without your acknowledgement.

XenoKovah, 2025-01-27:
LGTM 👍

Just for my own awareness, is it that the documentation page with "latest" in the URL (https://whad.readthedocs.io/en/latest/cli/ble/ble-periph.html) will correspond to the main branch and the one with "stable" (https://whad.readthedocs.io/en/stable/cli/ble/ble-periph.html) will correspond to the last tagged release? (Because I was looking at the stable URL given above and not seeing the change but then realized the URL given in the other ticket used "latest" instead.)

virtualabs, 2025-01-27:
Yes, "latest" version in documentation points to the main branch while "stable" version points to the documentation of the more recent release.

closed 2025-01-27 by virtualabs