whad-team/whad-client #134: Feature request: incorporate CLUES_data.json for UUID names

XenoKovah, 2025-01-23:
While I was looking at #129 output of all the various UUIDs on my iPhone that was serving as a test virtual peripheral, it made me wonder "How many of those do we not know what they are yet?" and "I wonder if some of those UUIDs are automatically randomized or changing between releases, or whether we know what they all are at this point and they'll stay known forever?"

So I went and searched for the first few and found they're easy ones like Continuity (d0611e78-bbb4-4591-a5f8-487910ae4366) and Nearby (9fa480e0-4967-4542-9390-d343dc5d04ae) which are already in [CLUES_data.json](https://github.com/darkmentorllc/CLUES_Schema/blob/master/CLUES_data.json). So the ask here is to pull in CLUES_data as a sub-module and then anywhere you're printing out UUID names based on the BT SIG assigned numbers, also check UUIDs against CLUES_data and print the "UUID_name" field as well (or probably f"{company} {UUID_name}" to be a little more descriptive). CLUES integration will be a lot easier lift than BTIDES ;)

virtualabs, 2025-01-24:
Yes, that is definitely a good idea and will be an opportunity for me to have a look at the CLUES database and the other databases you are working for. I think that we can put the company and UUID name  along with the GATT service UUID and keep only the UUID name for characteristics belonging to each service. I'll try to add this soon (and update the install instructions as well for people who wants to install it from our github repo 😉 ).

virtualabs, 2025-03-11:
I've just noticed that I included this feature into PR #156, still waiting for approval. You can give it a try by checking out branch `bugfix/issue-129-clues` and using `wble-central` to discover a device's services and characteristics. It will use CLUES data to resolve unknown services and characteristics UUIDs.

XenoKovah, 2025-03-14:
This doesn't seem to be working? 

```
$ git log -1
commit 5f47634d8142c0c47839864425a2b648423e82c3 (HEAD -> bugfix/issue-129-clues, origin/bugfix/issue-129-clues)
Merge: a0f2f04 d275e57
Author: Damien Cauquil <dcauquil@quarkslab.com>
Date:   Mon Feb 24 16:26:12 2025 +0100

    Merge branch 'bugfix/issue-129-clues' of github.com:whad-team/whad-client into bugfix/issue-129-clues
$ wble-central -i hci0 --bdaddr 63:5b:4c:d1:06:58 -r profile foo.json
Searching for target device 63:5b:4c:d1:06:58 ...
Connecting to target device ...
Enumerating services and characteristics ...
Service Generic Access (0x1800)  (handle 1 to 5)
  Device Name (0x2A00) R, handle 2, value handle: 3
    Value: 486f6d65506f64
  Appearance (0x2A01) R, handle 4, value handle: 5
    Value: 8002

Service Generic Attribute (0x1801)  (handle 6 to 9)
  Service Changed (0x2A05) I, handle 7, value handle: 8
    Descriptor type Client Characteristic Configuration (0x2902), handle: 9
    Value: 

Service Device Information (0x180A)  (handle 10 to 14)
  Manufacturer Name String (0x2A29) R, handle 11, value handle: 12
    Value: 4170706c6520496e632e
  Model Number String (0x2A24) R, handle 13, value handle: 14
    Value: 417564696f4163636573736f7279352c31

Service 9fa480e0-4967-4542-9390-d343dc5d04ae  (handle 15 to 19)
  af0badb1-5b99-43cd-917a-a77bc549e3cc WN, handle 16, value handle: 17
    Descriptor type Client Characteristic Configuration (0x2902), handle: 19
    Value: 

Writing profile JSON data to foo.json ...
```
`9fa480e0-4967-4542-9390-d343dc5d04ae` and `af0badb1-5b99-43cd-917a-a77bc549e3cc` are included in CLUES, but I don't see any sort of indication in that output, or in the JSON file.

virtualabs, 2025-03-16:
Oh wait, it has been implemented in the class that wraps a peripheral (when using a central to connect to it) but not in our GATT profile class 🤦. Could  you just test with `wble-central` in interactive mode (`wble-central -i hciX`) and make sure the `profile` command shows extra information taken from your CLUES's database ?

virtualabs, 2025-04-11:
@RCayre also reported this feature as non-working, and you were right. I made a small change in the code to improve its quality and killed the whole feature because of a wrong check. I'm working on a fix.

virtualabs, 2025-04-12:
Fixed and merged into `main` branch with PR #167.

virtualabs, 2025-04-16:
WHAD package is missing the CLUES_data.json file, thus throwing an exception.

virtualabs, 2025-04-16:
Project setup files have been fixed to correctly include the missing files. Changes have been merged into the `main` branch. You could give a try and confirm that it solves the issues you had (or not).