whad-team/whad-client #137: HCI adapters can be "backwards" from a WHAD perspective

XenoKovah, 2025-01-27:
(I'm on the latest `main` branch commit 9b91b568b001952dbeaa84409d6d639c89c4b826)

I was executing `wble-connect -i hci0 -r -b 58:09:ea:38:c4:34 | wble-spawn -i hci1 -p Virtual-BP.json` (with the older copy of `Virtual-BP.json`)

I wasn't seeing advertisements from the fake peripheral at address `11:11:11:11:11:11` in Sniffle (which is where I would expect them given that I was passing -i hci1 to `wble-spawn`). Then I randomly decided to sniff `22:22:22:22:22:22`, and sure enough I found the fake advertisements there! It turns out that apparently the `whadup` perspective on which HCI adapter is which, is backwards on this VM right now, with hci0 -> hci1 and hci1 -> hci0.

```
(venv) root@BTVM:/home/user/BT1001/whad-client# hciconfig 
hci0:	Type: Primary  Bus: USB
	BD Address: 22:22:22:22:22:22  ACL MTU: 310:10  SCO MTU: 64:8
	UP RUNNING 
	RX bytes:4720 acl:0 sco:0 events:229 errors:0
	TX bytes:13342 acl:0 sco:0 commands:277 errors:0

hci1:	Type: Primary  Bus: USB
	BD Address: 11:11:11:11:11:11  ACL MTU: 310:10  SCO MTU: 64:8
	UP RUNNING 
	RX bytes:84688 acl:0 sco:0 events:583 errors:0
	TX bytes:32549 acl:58 sco:0 commands:725 errors:0
(venv) root@BTVM:/home/user/BT1001/whad-client# whadup
[i] Available devices
- uart0
  Type: UartDevice
  Index: 0
  Identifier: /dev/ttyUSB0

- uart1
  Type: UartDevice
  Index: 1
  Identifier: /dev/ttyUSB1

- hci0
  Type: HCIDevice
  Index: 0
  Identifier: hci1

- hci1
  Type: HCIDevice
  Index: 1
  Identifier: hci0
```

XenoKovah, 2025-01-27:
OK, I was able to reproduce this as follows:

1) Turn **on** the adapter you want to be hci0
2) Turn **on** the adapter you want to be hci1
3) Turn **off** hci0. Now from a `whadup` perspective "hci0" will be "hci1"

```
(venv) root@BTVM:/home/user/BT1001/whad-client# whadup
[i] Available devices
- hci0
  Type: HCIDevice
  Index: 0
  Identifier: hci1
```

4) Turn **on** the adapter you just turned off (i.e. hci1). Now it will be mapped to hci0 from WHAD's perspective.

```
[i] Available devices
- hci0
  Type: HCIDevice
  Index: 0
  Identifier: hci1

- hci1
  Type: HCIDevice
  Index: 1
  Identifier: hci0
```

virtualabs, 2025-01-29:
I've just pushed a fix for this in the `main` branch, now HCI identifiers match those assigned by the system.

XenoKovah, 2025-01-29:
Confirmed it's working on the latest

closed 2025-01-29 by XenoKovah