whad-team/whad-client #138: Feature request: privacy-preserving logs (by default)

XenoKovah, 2025-01-28:
I think perhaps a lot of the time you probably don't care about what exactly BDADDRs are used in the advertisements going by, but rather just need to know whether they're public or random and whether they're the specific BDADDR target which was given on the CLI.

Thanks to WiGLE.net, it's unfortunately very possible for BDADDRs presented in log files to be mapped back to a person's house.

So the request is that log files replace BDADDRs with just stable-but-random IDs (e.g. as a simple idea, pick a nonce when the tool starts, and hash the BDADDR + nonce and use the hash in the log instead of the original BDADDR? Of course that could then make correlation between logs more difficult, so it needs thinking about by you, since you know better than I how you use your logs. (Maybe a rotating nonce on the filesystem that all tools share?))

Currently I try to manually delete lines out of logs if I see a BDADDR, but of course I could be missing some, and then consequently any logs I upload are just sitting on the internet forever giving away my home address. So I'd feel more comfortable uploading logs (and I'm sure others would too) if I didn't have to worry about physically-proximate BDADDRs being included.

You could of course then add a CLI argument to not mask BDADDRs if it was really necessary in some case.

virtualabs, 2025-01-29:
This is an excellent idea ! We need to think about it in order to come up with an obfuscation technique, so I guess it may require some time. But again, this is definitely required.

virtualabs, 2025-02-07:
I've started working on a possible solution but something stroke me: debug messages exchanged between any hardware device and the python client may also contain sensitive data like bluetooth addresses, and they are logged if logging level is set to `debug`.  That means we also need to obfuscate these messages to reflect the obfuscated addresses, and that's another story (not impossible though, but it implies to rework the whole code rather than just the part related to BLE ...). 

And of course replacing addresses in the log messages before they are being displayed/saved is not an option. This problem is indeed harder than I thought 🤔.

XenoKovah, 2025-02-07:
Would perhaps just post-processing log files to randomize BDADDRs be a reasonable approach?

virtualabs, 2025-02-07:
Well, it will work for sure but not 100% reliable (especially for binary-encoded addresses like `"\x66\x55\x44\x33\x22\x11"` that may appear when logging WHAD messages). I'll try to PoC something and see if it does the job.