whad-team/whad-client #139: MITM command that works on physical hardware doesn't work on Ubuntu 22.04 x86 or Ubuntu 24.04 ARM VMs

XenoKovah, 2025-01-28:
So now I'm getting back to testing the core issue I've been having with WHAD: it doesn't seem to work for me in VMs. This is a deal breaker in terms of our ability to use it in the class. Because for in-person or OST2 classes we can't be fiddling around with everyone's random laptop configuration, so we need to be able to provide a single known-good Ubuntu Linux VMware VM for x86 and ARM (for the Mac students.)

The following command was confirmed to be working fine on bare-metal Ubuntu 22.04 with the latest `main` branch of whad-client.
`wble-connect --log debug --log-file ./wble-connect-main-HW.log -i hci0 11:22:33:44:55:66 | wble-spawn --log debug --log-file ./wble-spawn-main-HW.log -i hci1 -p VirtualBP.json`

(Where once again I was testing with 2x CSR dongles with BDADDRs spoofed to 11:11:11:11:11:11 and 22:22:22:22:22:22.)

The fact that it was working was confirmed via two Sniffle captures which showed the same traffic appearing at both sides. When I executed the same command on my ARM Ubuntu 24.04 VM, it didn't work. For good measure, per ticket #52, I also tried reversing the order of `wble-spawn` and `wble-connect`, but this made no difference.

At that point I wanted to eliminate the variable that perhaps it was due to Ubuntu 24.04. (In general when setting up the class VM I had found that some tools which worked in the past on Ubuntu 20.04 and 22.04 didn't work on 24.04.) So I started upgrading the bare-metal Linux system to Ubuntu 24.04. After upgrading to 24.04 I confirmed the command still works. This suggested that the issue wasn't Ubuntu itself.

While waiting for the bare-metal system upgrade I also tested the other way, by opening an existing Ubuntu 22.04 x86 VM, and confirming the command didn't work there either (despite working on Ubuntu 22.04 bare metal.) 

(Note: I was literally moving the same USB BT dongles between the bare-metal machine and VMs.)

This has led me to the conclusion that there may be something about how HCI adapters are handled by VMware which is preventing WHAD from working within the VMs.

Can you confirm if you can run the above MITM command on any of the 4 combinations of x86/ARM Ubuntu 22.04/24.04? If so then I can try to replicate the same (I can also test on Windows or Linux host OSes instead of macOS; but VMware, rather than VirtualBox is a requirement, as that's what we've standardized on for OST2, to avoid extra QA work.)

virtualabs, 2025-01-29:
This is weird. We gave a WHAD workshop at various cons and were using an Lubuntu-based virtual machine (OVA available here: https://drive.google.com/file/d/1JJK7QjhWLELgWdbSySIyrmI9UX-0Rkj8/view?usp=drive_link) and it was working fine AFAIK. I'll try with an Ubuntu 22.04 in VMware as soon as possible.

XenoKovah, 2025-01-29:
Thanks, I'll give the VM a try. What's the password?

XenoKovah, 2025-01-29:
I was having trouble connecting to the `peripheral_ots` from the Lubuntu VM, which was working fine from the Ubuntu VM. So I went to change it to advertise on only channel 37 (in case it was Sniffle missing the connection, though I was pretty sure it wasn't, because when that happens Sniffle stops seeing advertisements). But then I remembered why I didn't like peripheral_ots/central_otc: because on the Nordic dongle you have to hack around to make the central work with only the single button available to it (the code assumes 3 or 4 buttons IIRC), and that code is too unwieldy to easily provide.

So instead I used `samples/bluetooth/peripheral_hr` and `samples/bluetooth/central_hr` (because I like that `peripheral_hr` shows a blinking light while it's waiting for connection, and therefore I can tell if WHAD connected or not even without looking at the pcap.)

Note: I made the following change in `samples/bluetooth/peripheral_hr/src/main.c` -> `main()` to make it only advertise on channel 37 so make Sniffle capture more reliable since it only then needs to listen on channel 37:

```
        struct bt_le_adv_param custom_adv_param = {
                // Force UVP to only advertise on channel 37, thus increasing reliability of Sniffle catching connections,
                // by allowing Sniffle to only sniff on channel 37 (now the only place connections can be sent to the UVP)
                .options = BT_LE_ADV_OPT_CONNECTABLE | BT_LE_ADV_OPT_SCANNABLE | BT_LE_ADV_OPT_DISABLE_CHAN_38 | BT_LE_ADV_OPT_DISABLE_CHAN_39 ,
                .interval_min = BT_GAP_ADV_FAST_INT_MIN_1, // These can be changed to "SLOW" instead of "FAST"
                .interval_max = BT_GAP_ADV_FAST_INT_MAX_1,
                .peer = NULL,
        };

<snip>

//      err = bt_le_adv_start(BT_LE_ADV_CONN_ONE_TIME, ad, ARRAY_SIZE(ad), sd, ARRAY_SIZE(sd));
        err = bt_le_adv_start(&custom_adv_param, ad, ARRAY_SIZE(ad), sd, ARRAY_SIZE(sd));

```

Interestingly it generated the same random BDADDR as the `peripheral_ots`: D6:61:69:9E:72:75 ("Anyone who considers arithmetical methods of producing random digits is, of course, in a state of sin." - John Von Neumann ;) I guess they're not using Nordic's [TRNG](https://docs.nordicsemi.com/bundle/nwp_057/page/WP/nwp_057/security_services/trng.html))

ANYWAY, even with a known-good and confirmed-working `peripheral_hr`/`central_hr` combo, the Lubuntu image is still failing with `wble-connect` never even attempting a CONNECT_IND as far as Sniffle can tell. (Despite this working from the Ubuntu image.) This is also despite the fact that the `wble-central profile` command, or interactive mode, connect to it just fine.

```
(venv) whad@whad-vm:~/whad-client$ git log -1
commit 01a4e92470f2f65ecb8f57dac69a4720c7699bf0 (HEAD -> main, origin/main, origin/HEAD)
Author: Damien Cauquil <dcauquil@quarkslab.com>
Date:   Wed Jan 29 09:55:07 2025 +0100

    Removed an unecessary commented line.
(venv) whad@whad-vm:~/whad-client$ pip3 install .
Processing /home/whad/whad-client
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
<snip>
Building wheels for collected packages: whad
  Building wheel for whad (pyproject.toml) ... done
  Created wheel for whad: filename=whad-1.2.6-py3-none-any.whl size=800810 sha256=b14936c4623ad4a755ca4ee3c5a0ffe0385156197c689d0118cebe431bd76f10
  Stored in directory: /home/whad/.cache/pip/wheels/98/3a/9c/a7762305a6cccd26d76eae73c2f6bab93e95158d54f6788a38
Successfully built whad
Installing collected packages: whad
  Attempting uninstall: whad
    Found existing installation: whad 1.2.6
    Uninstalling whad-1.2.6:
      Successfully uninstalled whad-1.2.6
Successfully installed whad-1.2.6
(venv) whad@whad-vm:~/whad-client$ wble-connect -i hci0 d6:61:69:9e:72:75 | wdump -f Zephyr.pcap
[!] Cannot connect to d6:61:69:9e:72:75
```

I'm not sure it's necessarily worth debugging the Lubuntu VM results though. (But I can provide logs if you think it's worth it.) I'm more interested in seeing your results within an Ubuntu VMware VM (since I'm guessing you used VirtualBox w/ this OVA?)

A nagging question is whether the workshop exercised behaviors are sufficiently representative of the MITM issue I'm seeing. I can see that [https://github.com/whad-team/whad-workshop](https://github.com/whad-team/whad-workshop) was checked out on there, but the contents of that repository lead me to believe the primary thing that was tested was e.g.  scanning and GATT enumeration. Did you actually test this sort of `wble-connect | wble-spawn` MITM behavior with workshop participants?

XenoKovah, 2025-01-29:
Oh and you said "I'll try with an Ubuntu 22.04 in VMware as soon as possible." Please try with 24.04 not 22.04 (as I'm going to be updating all the OST2 classes to use 24.04 & Win11 this coming year.)

virtualabs, 2025-01-29:
Oh I see what's happening here: is your device BD address random ? `wble-central` keeps track of the address type and connects fine in interactive mode because it specifies in a transparent way (not noticeable by the user) the target device's address type. 

You can specify with `wble-connect` that your target device uses a random address with the  `--random` (or `-r`) option. We also had this issue during a workshop, and I think we might also improve `wble-connect` to tell the user that we found a device with a similar address that is random instead of public. 

FTR, we used both ButteRFly dongles and HCI dongles during our workshop, but we got less errors/issues using HCI dongles.

XenoKovah, 2025-01-29:
OK, yes, that's totally what the `wble-connect` issue was. And it's annoying because I _know_ I looked at the `--help` to remind myself if this was the command that took `-r`...and I apparently my eyes completely missed it as the last line of the help output :-/. I've clearly got too many VMs & physical machines set up, so now I'm copying commands from the wrong ones and reusing them in the wrong context. (So I edited the original msg to have the `-r` for the next time I go to copy and paste or re-type (when copy-paste isn't working in a VM))

So the net result though is that Lubuntu behaves the same as Ubuntu: `wble-connect` works, `wble-spawn` works, but when a Zephyr central connects to the WHAD peripheral, it succeeds but there's no MITM traffic forwarding occurring.

pcap from connection to Zephyr Peripheral:

[image]

pcap from connection to WHAD Peripheral:

[image]

XenoKovah, 2025-01-29:
New data point. I also happen to have VirtualBox on my x86 Mac. I upgraded it to the latest (because it was crashing, natch), and I am seeing the same results as VMware :-/. But I'm logged in from within that VM now just so I can get the actual commands I used documented *just in case I'm making a simple mistake again*...

```
(venv) whad@whad-vm:~/whad-client$ wble-connect --log debug --log-file ./wble-connect-main-lubuntu-vbox.log -i hci0 -r d6:61:69:9e:72:75 | wble-spawn --log debug --log-file ./wble-spawn-main-lubuntu-vbox.log -i hci1 -p ZephyrHR.json
^C/!\ ble-spawn stopped (CTL-C)
/!\ wble-connect stopped (CTL-C)
(venv) whad@whad-vm:~/whad-client$ git log -1
commit 01a4e92470f2f65ecb8f57dac69a4720c7699bf0 (HEAD -> main, origin/main, origin/HEAD)
Author: Damien Cauquil <dcauquil@quarkslab.com>
Date:   Wed Jan 29 09:55:07 2025 +0100

    Removed an unecessary commented line.
(venv) whad@whad-vm:~/whad-client$ hciconfig 
hci1:   Type: Primary  Bus: USB
        BD Address: 22:22:22:22:22:22  ACL MTU: 310:10  SCO MTU: 64:8
        UP RUNNING 
        RX bytes:2968 acl:0 sco:0 events:159 errors:0
        TX bytes:11341 acl:5 sco:0 commands:183 errors:0

hci0:   Type: Primary  Bus: USB
        BD Address: 11:11:11:11:11:11  ACL MTU: 310:10  SCO MTU: 64:8
        UP RUNNING 
        RX bytes:2997 acl:0 sco:0 events:159 errors:0
        TX bytes:11225 acl:5 sco:0 commands:179 errors:0

(venv) whad@whad-vm:~/whad-client$ whadup
[i] Available devices
- hci0
  Type: HCIDevice
  Index: 0
  Identifier: hci0

- hci1
  Type: HCIDevice
  Index: 1
  Identifier: hci1

- uart0
  Type: UartDevice
  Index: 0
  Identifier: /dev/ttyUSB0

- uart1
  Type: UartDevice
  Index: 1
  Identifier: /dev/ttyUSB1

```

`whad@whad-vm:~/Sniffle/python_cli$ sudo python3 ./sniff_receiver.py -s /dev/ttyUSB0 -c 37 -o W-C-to-Z-P.pcap -m d6:61:69:9e:72:75`

`whad@whad-vm:~/Sniffle/python_cli$ sudo python3 ./sniff_receiver.py -s /dev/ttyUSB1 -o Z-C-to-W-P.pcap -m 22:22:22:22:22:22`

XenoKovah, 2025-01-29:
And for good measure I went back and copied the command into VMware but it still didn't work from there. Logs from that attached,

[wble-spawn-main-lubuntu-vmware-sanitized.log](https://github.com/user-attachments/files/18592106/wble-spawn-main-lubuntu-vmware-sanitized.log)
[wble-connect-main-lubuntu-vmware-sanitized.log](https://github.com/user-attachments/files/18592107/wble-connect-main-lubuntu-vmware-sanitized.log)

XenoKovah, 2025-02-07:
Any luck testing this in VMware and replicating the results? We're a couple weeks from "pencils down" on the BLE class, and I'd like to get WHAD into it, but this is a hard blocker that only you can debug and see if it's possible to work around.

virtualabs, 2025-02-07:
Unfortunately, I'm on vacation starting from tomorrow until the 15th of Feb. and I'm not sure I can bring my BT setup with me. I'll do my best to try to solve this ASAP. Marking this issue as urgent.

virtualabs, 2025-02-11:
I had some free time so I downloaded a Ubuntu x64 22.04 LTS and install it using VMWare player 17.6.0 build-24238078. I've then installed WHAD and played a bit using two USB BT dongles ... and managed to create a proxy with no issue (connect and spawn, targeting a public BD address). I had some issues using the internal Bluetooth adapter (the one from my laptop's motherboard) but USB dongles seem to work fine. Let me know if this setup differs from yours, I'll adjust it to fit your exact configuration.

XenoKovah, 2025-02-12:
I was testing on an ARM and x86 Mac, but I'm assuming you don't have one to test on, so now I'm making fresh VMs on a Linux and Windows host OSes. Which did you use?

Also it's buried in the text up there but "please try with 24.04 not 22.04 (as I'm going to be updating all the OST2 classes to use 24.04 & Win11 this coming year.))".

XenoKovah, 2025-02-12:
Same results on Windows host OS (Windows 10 Pro 22H2 build _19045.5487) + VMware 17.6.2-build24409262

```
(venv) user@BTVM-W:~/BT1501/whad-client$ hciconfig 
hci1:	Type: Primary  Bus: USB
	BD Address: 22:22:22:22:22:22  ACL MTU: 310:10  SCO MTU: 64:8
	UP RUNNING 
	RX bytes:5714 acl:0 sco:0 events:93 errors:0
	TX bytes:5410 acl:0 sco:0 commands:105 errors:0

hci0:	Type: Primary  Bus: USB
	BD Address: 11:11:11:11:11:11  ACL MTU: 310:10  SCO MTU: 64:8
	UP RUNNING 
	RX bytes:4460 acl:0 sco:0 events:95 errors:0
	TX bytes:5905 acl:0 sco:0 commands:107 errors:0

(venv) user@BTVM-W:~/BT1501/whad-client$ wble-spawn --log debug --log-file ./wble-spawn-main-ubuntu24.04-windows-host-vmware.log -i hci1 -p ZephyrPeripheralHR.json | wble-connect --log debug --log-file ./wble-connect-main-ubuntu24.04-windows-host-vmware.log -i hci0 -r cb:93:9e:b4:70:42
```

(I did spawn then connect since you said it was being more reliable for you.)

Confirmed both connections with Sniffle, which continued following the connections and seeing only empty packets until I hit ctrl-c.

```
user@BTVM-W:~/BT1501/Sniffle/python_cli$ sudo python3 sniff_receiver.py -s /dev/ttyUSB0 -c 37 -m cb:93:9e:b4:70:42 -o zephyr-peripheral.pcap
```
[image]

```
user@BTVM-W:~/BT1501/Sniffle/python_cli$ sudo python3 sniff_receiver.py -s /dev/ttyUSB1 -m 22:22:22:22:22:22 -o WHAD-peripheral.pcap
```
[image]

XenoKovah, 2025-02-12:
Result on Linux host OS: it works. (I took literally the same USB hub full of dongles to the Linux machine and plugged it in and ran the same commands as the above Windows test and it worked there. Apologies but the screen resolution is way too small on this Linux host OS...)

[image]

[image]

Current status of Ubuntu 24.04 VM testing:

| | ARM macOS | x86 macOS | x86 Windows | x86 Linux | 
|---|---|---|---|---|
| VMware | Doesn't work | Doesn't work | Doesn't work | Works | 
| Virtual Box | Untested | Works | Untested | Untested | 
| Bare metal | Untested | Untested | Untested | Works |

XenoKovah, 2025-02-12:
And just because I noticed I had an old Ubuntu 22.04 VM sitting on the Windows box, I went ahead and updated it and confirmed Windows host OS (Windows 10 Pro 22H2 build _19045.5487) + VMware 17.6.2-build24409262 + Ubuntu 22.04 *does not* work. Which further contributes to my hypothesis that it's down to the virtualization software somehow breaking the connection between the HCI adapters.

virtualabs, 2025-02-13:
I knew my setup might be wrong and in fact it was. I'll test on my side on a Linux OS running VMWare Player with Ubuntu 24.04 but I guess it'll give the same results as Ubuntu 22.04. I have a Windows machine at home, I'll give it a try too and see if I manage to reproduce this issue. 
It's good to see that you get similar results running VMWare Player on a Linux host though.

If this issue is caused by virtualization software it may be more difficult than expected to figure out why and more importantly to find a working fix that can be shipped in WHAD (except if we can find a trick to bypass a wrong behavior). I'll get my hands on a Windows machine in the week-end and will try to replicate this issue. In the meantime, I'll focus on the logs to draw some hypothesis about what may happen at the HCI level. Could also be interesting to see what `hcidump` sees from the guest PoV.

virtualabs, 2025-02-17:
I've setup an Ubuntu 24.04 VM on a laptop running Windows with VMWare Workstation Player (17.6.2 build-24409262) and managed to reproduce the issue !

As I suspected, there is no Connection Complete event due to the fact the system uses LE extended advertising and the LE extended connection procedure, and this is why `wble-connect` does not consider the BLE connection created and does not forward packets. It seems to me that the issue resides in the way WHAD implements its HCI virtual device, which is basically old and does not take the new HCI LE events and commands into account or at least probe the hardware to check what features are supported. I don't know exactly why this works pretty good on Linux hosts or emulated Ubuntu VMs on Linux hosts and not in a Windows environment running VMWare Workstation Player. I'll do more tests during the week, but I cannot guarantee to have a working fix pretty soon. I'll do my best to figure out a quick fix for now, but it looks like the code interfacing the HCI interface needs a good rework. Lucky for us, it is easy to monitor thanks to BlueZ `hcidump` (which is not aware of the new HCI LE commands and events by the way). But If I'm right about the cause of this issue, it can be solved in WHAD and there is no need to worry about the virtualization software.

I tested `bluetoothctl` in this Ubuntu VM and I managed to correctly connect to a device and even read some characteristics value, so it looks like everything is working fine with this tool.

I'll keep you posted !

XenoKovah, 2025-02-17:
Great to hear you were able to reproduce it, as that's the first step towards figuring it out! You should try `btmon -w <logfile>` rather than `hcidump`. IIRC `hcidump` is deprecated and doesn't even ship by default anymore, whereas `btmon` & `bluetoothctl` do.

virtualabs, 2025-02-18:
`btmon` is perfect, indeed ! It took some time to figure out what was happening, but I guess I found the root cause and managed to get a working configuration.

First, let's talk a bit about this LE MTU exchange request that was never forwarded, because it is a huge mistake I made. When a BLE connection is initiated through HCI, WHAD receives a specific event indicating the connection is established, and then does its job to relay PDUs. Despite what I wrote earlier, this event is usually correctly received by WHAD as well as the LE MTU exchange request, but the latter is 99% of the time lost because of a bad behavior. 

In WHAD, we first create an interface (an instance of `WhadDevice`) that is some kind of communication layer with the real hardware handling the WHAD protocol and then what we attach a *connector* to the interface (basically any class derived from `WhadDeviceConnector`. It takes two steps to get a working *connector* able to do whatever we want with any BLE HCI dongle. And that's the problem. When using pipes, we use a Unix socket and a dedicated interface to make WHAD believe this is a legitimate device (some kind of wrapper class if you prefer).

Let's consider the following command line:

```
$ wble-connect -i hci0 11:22:33:44:55:66 | wble-spawn -i hci1 -p some-profile.json
```

We first execute `wble-connect`, and it initiates a LE connection to the specified target device. Once connected, the device immediately sends a LE MTU exchange request to our emulated central device. `wble-connect` however, outputs a specific URI on its standard output to give all the required informations to `wble-spawn`: what Unix socket to connect to, details about the current BLE connection, etc. `wble-spawn` then connects to the Unix socket, performs a quick device discovery and then starts relaying PDUs. But the first PDU came **before** `wble-connect`'s HCI interface is linked to a *connector*, and every message received during this short time is simply lost. Because there's a background thread handling WHAD messages sent by the interface that simply sends these messages to nowhere because no connector has been assigned to this interface. And this is why this MTU exchange request is always lost. 

Why is this only happening with VMWare ? I have no idea for now. But on a real computer the time spent between interface creation and the connector assignment is very short, and the PDU is received when a *connector* has already been assigned and everything goes smoothly. In VMWare, it always come **before** `wble-spawn` connects, and therefore it is lost.

I made a fix in the `bugfix/issue-139` branch to solve this issue (but wait, there is  more to read). The fix is pretty simple: we do not process messages until a connector is assigned to an interface, and boom no more message lost.

Once this fix applied, I had a few successes: I managed to get a working proxy once, and it never worked anymore. I had a lot of issues with my HCI dongles: connection-related events never showed up, connection initiation HCI messages were missing, a complete mess. Restarting the VM did not solve this. And then I stopped the `bluetooth` service, and everything was back to normal. I guess this service had some interaction with the dongles that conflicts with WHAD. Once disabled, I was able to create a BLE proxy using `wble-connect` and `wble-spawn` the usual way. Even `wble-proxy` worked smoothly !

**So, here is my working configuration running on VMWare Workstation Player on Windows 10 with Ubuntu 24.04:**

* VMWare's USB controller version set to 2.0
* Bluetooth service must be stopped in Ubuntu: `sudo service bluetooth stop` if it is more convenient for your training
* Running the `bugfix/issue-139` branch (that also includes a fix from issue #119)

I don't know if it will work on MacOS, but there's a high probability the root cause is very similar. I thought it was related to BLE 5.x but nope, I was wrong. Still, it would be interesting to update the code dealing with HCI devices when we'll add support for BLE 5 (next in the TODO list)

XenoKovah, 2025-02-18:
Oh wow, it's the return of `sudo service bluetooth stop`! I remember that as the secret sauce to making btlejuice and GATTacker work more reliably ;) (Except IIRC it was only mentioned in the btlejuice documentation, not GATTacker, and I picked it up from the former to improve the reliability of the latter.)

Anyway, to a first approximation it works in VMware Fusion on a Mac!

WHAD-C to Zephyr-P
[image]

Zephyr-C to WHAD-P
[image]

(I had just highlighted some lower level details that differed before remembering that that's to be expected from different centrals/peripherals. What matters is the ATT traffic, and that looks the same!)

I'll do more tests with the real Zephyr UltraVulnerablePeripheral / Android UltraVulnerableCentral that we use in class and let you know if anything else comes up. Thanks for the quick fix!

virtualabs, 2025-02-19:
Let me know if it solves your issues with virtualization on MacOS so I can merge this fix into our `main` branch and plan a new release 😄

virtualabs, 2025-02-19:
For now I've merged my fix into `main` through PR #150, I'll keep this open as a reminder to add a section dedicated to virtualization in the project documentation.

virtualabs, 2025-02-20:
Updated documentation with PR #153, I reproduced your summary table and completed with the information I have about VirtualBox (tested during some workshops we gave in 2024).

virtualabs, 2025-02-20:
Lucky me, I forgot to mention this in the documentation 😅.

virtualabs, 2025-02-25:
Problem is now solved and documentation has been updated. Should I close this one ?

XenoKovah, 2025-02-25:
Yep