whad-team/whad-client #140: Typos/issues on "Compatible devices" page

XenoKovah, 2025-01-28:
I'm starting to fiddle with ButteRFly now (I had tried it once a long time ago w/ MIRAGE but it was broken by then.)

The below pertain to https://whad.readthedocs.io/en/stable/device/compat.html

## Typos:
sone -> some

"4. Make sure a red LED blinks before continuing with next steps. If not retry step 4." should say "step 3" at the end instead of "step 4". (Note: this appears twice, so make sure you replace both if you're doing find and replace, but I'm going to recommend below deleting one of the sections it appears in.)

## Issues:

### Section duplicated incorrectly
This section:
https://whad.readthedocs.io/en/stable/device/compat.html#installing-butterfly-firmware-on-an-nrf52840-mdk-dongle-nrfutil
is a duplicate of this section:
https://whad.readthedocs.io/en/stable/device/compat.html#installing-butterfly-firmware-on-a-nordic-nrf52840-dongle
However I think the former should be entirely deleted. Because it's not applicable to the MDK dongle. Only the section preceding it, https://whad.readthedocs.io/en/stable/device/compat.html#installing-butterfly-firmware-on-an-nrf52840-mdk-dongle-uf2-bootloader applies to the MDK dongle.

### MDK dongle instructions incorrect
Within https://whad.readthedocs.io/en/stable/device/compat.html#installing-butterfly-firmware-on-an-nrf52840-mdk-dongle-uf2-bootloader it includes a step 3 of "Press the tactile switch during 2 seconds and release, a red LED must blink". However, I tried that and it doesn't do anything. So I googled it and found that the correct procedure to make the device appear as a mass storage device at plug-in is to hold the only available button down *while it's unplugged*, and while it's being plugged in, until a light which would otherwise be blue becomes green. The official description (if you want to just copy it) is here: https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/guides/ncs/samples/usb/mass/#testing.

### Nordic dongle instructions slightly incorrect
Within https://whad.readthedocs.io/en/stable/device/compat.html#installing-butterfly-firmware-on-a-nordic-nrf52840-dongle it says "Plug your Nordic nRF52840 dongle in your computer and press the RESET button for 2 seconds". That's not been my experience that it requires 2 seconds. Just a single click/tap will do. 

However, you will probably want to be clear about which button is the reset button, as there are two on the board. The reset button is the one which is harder to see and pressed *parallel* to the PCB:

[image]
(I think I got this pic from here: https://docs.zephyrproject.org/latest/boards/nordic/nrf52840dongle/doc/index.html as we use that dongle + Zephyr RTOS in the class for our "Ultra-Vulnerable Peripheral" ongoing lab.)

### nrfutil is difficult to get on ARM Linux :-/

Re the instructions "Download and install Nordic nrfutil on your computer", unfortunately Nordic currently only ships an x86-based binary for nrfutil. (https://docs.nordicsemi.com/bundle/nrfutil/page/README.html#supported-operating-systems) For ARM Linux, I found myself having to install python2 (which isn't available through apt anymore on Ubuntu 24.04) and using the older python-based nrfutil. I don't know if you want to copy these instructions but here's what I have from our class-VM setup page:

```
sudo apt install -y build-essential checkinstall libncursesw5-dev libssl-dev libsqlite3-dev tk-dev libgdbm-dev libc6-dev libbz2-dev libffi-dev
wget https://www.python.org/ftp/python/2.7.18/Python-2.7.18.tgz
tar -xvf Python-2.7.18.tgz
cd Python-2.7.18
./configure --enable-optimizations
make -j 10
# THERE WILL BE SSL ERRORS ON TEST 142 UP THROUGH Thread-248. IGNORE THEM! (Those are just unit tests failing, and we don't need to use SSL.)
sudo make install
curl https://bootstrap.pypa.io/pip/2.7/get-pip.py -o get-pip.py
sudo python2.7 get-pip.py
pip2.7 install nrfutil
```

virtualabs, 2025-01-29:
It looks like my MDK dongle does not work anymore (no CDC UART device detected), but I took some time to review the documentation and you are right. I've modified the documentation and added a section explaining how to build *nrfutil* on ARM Linux.

Still some modifications to do and I'll push a branch with the modified version.

virtualabs, 2025-01-31:
Updated documentation in `main`, should be available in https://whad.readthedocs.io/en/latest/ soon.

XenoKovah, 2025-02-01:
LGTM

closed 2025-02-01 by XenoKovah