whad-team/whad-client #143: Consistent failure to profile custom Zephyr peripheral

XenoKovah, 2025-01-29:
Since I just remembered you said you didn't have an iPhone to test the LightBlue virtual blood pressure device in, I'll switch to using Zephyr for tests since I saw that #52 mentioned how to set it up (though FWIW I'm using 3.7 not 3.6).

In the class we have an "Ultra-Vulnerable Peripheral" (UVP) which is based on Zephyr and customized to have extra vulnerabilities. Interestingly it seems that I reliably can't profile it with WHAD using the following command (on either Ubuntu 24.04 or the Lubuntu image (originally I thought it was the Lubuntu image's problem, but it's not, it's something about this peripheral))

`wble-central -i hci1 -b ca:fe:13:37:00:01 -p UVP.json -r --log debug --log-file ./profile-failure.log profile`

I'm not really sure what's going on because I can see from the pcap WHAD sending a SCAN_REQ and getting back a SCAN_RSP. But it seems like WHAD's not recognizing that it got back a SCAN_RSP and so not continuing on to CONNECT_IND? I've attached a log from the profile command but I don't know how useful it will be with all the `bd_addr`-containing lines grepped out... (A good argument for #138 perhaps, if this log turns out to be inconclusive.)

```
Timestamp: 13.782900  Length: 27  RSSI: -35  Channel: 37  PHY: 1M  CRC: 0x93450B
Ad Type: ADV_IND
ChSel: 1 TxAdd: 1 RxAdd: 0 Ad Length: 25
AdvA: CA:FE:13:37:00:01 (Static)
0x0000:  60 19 01 00 37 13 fe ca  02 01 06 02 0a 14 05 03  `...7...........
0x0010:  0a 18 0f 18 06 08 55 56  50 30 31                 ......UVP01

Timestamp: 13.783330  Length: 14  RSSI: -31  Channel: 37  PHY: 1M  CRC: 0x2CADC1
Ad Type: SCAN_REQ
ChSel: 0 TxAdd: 0 RxAdd: 1 Ad Length: 12
ScanA: 22:22:22:22:22:22 (Public) AdvA: CA:FE:13:37:00:01 (Static)
0x0000:  83 0c 22 22 22 22 22 22  01 00 37 13 fe ca        ..""""""..7...

Timestamp: 13.783656  Length: 38  RSSI: -35  Channel: 37  PHY: 1M  CRC: 0x7BB7CA
Ad Type: SCAN_RSP
ChSel: 0 TxAdd: 1 RxAdd: 0 Ad Length: 36
AdvA: CA:FE:13:37:00:01 (Static)
0x0000:  44 24 01 00 37 13 fe ca  0a ff 4d 44 44 4d 34 45  D$..7.....MDDM4E
0x0010:  56 41 21 12 24 17 2f 2f  44 61 72 6b 4d 65 6e 74  VA!.$.//DarkMent
0x0020:  6f 72 2e 63 6f 6d                                 or.com

Timestamp: 13.815645  Length: 27  RSSI: -35  Channel: 37  PHY: 1M  CRC: 0x93450B
Ad Type: ADV_IND
ChSel: 1 TxAdd: 1 RxAdd: 0 Ad Length: 25
AdvA: CA:FE:13:37:00:01 (Static)
0x0000:  60 19 01 00 37 13 fe ca  02 01 06 02 0a 14 05 03  `...7...........
0x0010:  0a 18 0f 18 06 08 55 56  50 30 31                 ......UVP01

Timestamp: 13.847962  Length: 27  RSSI: -35  Channel: 37  PHY: 1M  CRC: 0x93450B
Ad Type: ADV_IND
ChSel: 1 TxAdd: 1 RxAdd: 0 Ad Length: 25
AdvA: CA:FE:13:37:00:01 (Static)
0x0000:  60 19 01 00 37 13 fe ca  02 01 06 02 0a 14 05 03  `...7...........
0x0010:  0a 18 0f 18 06 08 55 56  50 30 31                 ......UVP01

Timestamp: 13.880799  Length: 27  RSSI: -35  Channel: 37  PHY: 1M  CRC: 0x93450B
Ad Type: ADV_IND
ChSel: 1 TxAdd: 1 RxAdd: 0 Ad Length: 25
AdvA: CA:FE:13:37:00:01 (Static)
0x0000:  60 19 01 00 37 13 fe ca  02 01 06 02 0a 14 05 03  `...7...........
0x0010:  0a 18 0f 18 06 08 55 56  50 30 31                 ......UVP01
```

[profile-failure-sanitized.log](https://github.com/user-attachments/files/18588288/profile-failure-sanitized.log)

XenoKovah, 2025-02-19:
Update: this is working on the real UltraVulnerablePeripheral, it's just really slow to connect for some unknown reason. It sits on `Searching for target device CA:FE:13:37:00:01` for 30 seconds (out of 38 total) before actually connecting. That seems like a likely bug.

```
(venv) root@BTVM:/home/user/BT1001/whad-client# time wble-central -i hci0 -b CA:FE:13:37:00:01 -r profile UVP2.json
Searching for target device CA:FE:13:37:00:01 ...
Connecting to target device ...
Enumerating services and characteristics ...
Service Generic Attribute (0x1801)  (handle 1 to 8)
  Service Changed (0x2A05) I, handle 2, value handle: 3
    Descriptor type Client Characteristic Configuration (0x2902), handle: 4
  Client Supported Features (0x2B29) RW, handle 5, value handle: 6
  Database Hash (0x2B2A) R, handle 7, value handle: 8

Service Generic Access (0x1800)  (handle 9 to 15)
  Device Name (0x2A00) RW, handle 10, value handle: 11
  Appearance (0x2A01) R, handle 12, value handle: 13
  Peripheral Preferred Connection Parameters (0x2A04) R, handle 14, value handle: 15

Service Battery (0x180F)  (handle 16 to 20)
  Battery Level (0x2A19) RN, handle 17, value handle: 18
    Descriptor type Client Characteristic Configuration (0x2902), handle: 19

Service Device Information (0x180A)  (handle 21 to 25)
  Model Number String (0x2A24) R, handle 22, value handle: 23
  Manufacturer Name String (0x2A29) R, handle 24, value handle: 25

Service 444d0000-4865-6d6c-6f63-6b21f09fa4aa  (handle 26 to 43)
  444d0001-4865-6d6c-6f63-6b21f09fa4aa R, handle 27, value handle: 28
  444d0002-4865-6d6c-6f63-6b21f09fa4aa W, handle 29, value handle: 30
  444d0003-4865-6d6c-6f63-6b21f09fa4aa RI, handle 31, value handle: 32
    Descriptor type Client Characteristic Configuration (0x2902), handle: 33
  444d0004-4865-6d6c-6f63-6b21f09fa4aa RW, handle 34, value handle: 35
  444d0005-4865-6d6c-6f63-6b21f09fa4aa R, handle 36, value handle: 37
  444d0006-4865-6d6c-6f63-6b21f09fa4aa R, handle 38, value handle: 39
  444d0007-4865-6d6c-6f63-6b21f09fa4aa R, handle 40, value handle: 41
  444d0008-4865-6d6c-6f63-6b21f09fa4aa R, handle 42, value handle: 43

Service Object Transfer (0x1825)  (handle 44 to 62)
  OTS Feature (0x2ABD) R, handle 45, value handle: 46
  Object Name (0x2ABE) RW, handle 47, value handle: 48
  Object Type (0x2ABF) R, handle 49, value handle: 50
  Object Size (0x2AC0) R, handle 51, value handle: 52
  Object ID (0x2AC3) R, handle 53, value handle: 54
  Object Properties (0x2AC4) R, handle 55, value handle: 56
  Object Action Control Point (0x2AC5) WI, handle 57, value handle: 58
    Descriptor type Client Characteristic Configuration (0x2902), handle: 59
  Object List Control Point (0x2AC6) WI, handle 60, value handle: 61
    Descriptor type Client Characteristic Configuration (0x2902), handle: 62

Writing profile JSON data to UVP2.json ...

real	0m38.353s
user	0m7.128s
sys	0m0.161s
```

XenoKovah, 2025-02-19:
I will note that this seems to also manifest as an inability to use `wble-proxy` despite the device definitely advertising/scan responding (as shown above).

```
(venv) root@BTVM:/home/user/BT1001/whad-client# wble-proxy -i hci0 -p hci1 --link-layer --output foo.pcap ca:fe:13:37:00:01 
Scanning for target device (timeout: 30 seconds)...
/!\ Target device not found, connection timeout exceeded.
```

virtualabs, 2025-02-19:
Found it, WHAD's advertising URI parsing had an error in it (I assumed the URI scheme was encoded on 2 bytes while it is encoded in a single UTF-8 character). I've pushed a fix in branch `bugfix/issue-143`, I've tested it on my machine and it's way better 😄, LMK if it's also working for you.

XenoKovah, 2025-02-19:
Ah, I see where both of our confusion is coming from: Despite how "U+0017" in the Assigned Numbers document *looks* like it's supposed to be a 2-byte value, the actual values (as defined in the Core Specification Supplement) are UTF-**8** *not* UTF-16. So Wireshark/our code was correct, it should only be a single byte 0x17 for the URI encoding preceding the rest of the characters.

[image]

XenoKovah, 2025-02-19:
Confirmed this is still happening with the merged-to-main version too (not that I'd expect it to differ from the now-removed `bugfix/issue-143` branch)

```
    "devinfo": {
        "adv_data": "020106020a1405030a180f1806085556503031",
        "bd_addr": "ca:fe:13:37:00:01",
        "addr_type": 1,
        "scan_rsp": "0aff4d44444d3445564121122417002f4461726b4d656e746f722e636f6d"
    }
```

virtualabs, 2025-02-19:
Oops, I thought I was done with it but forgot to fix the encoding routine (parsing was correctly fixed though). I'm working on a fix, will merge pretty soon in `main`.

virtualabs, 2025-02-19:
I've just merged PR #151 that fixes this issue into `main`. I've tested it against your device and get the correct values in the JSON (`0x17 0x2b` instead of `0x17 0x00 0x2b`).

XenoKovah, 2025-02-19:
Bingo, JSON looks good to me too now.

```
    "devinfo": {
        "adv_data": "020106020a1405030a180f1806085556503031",
        "bd_addr": "ca:fe:13:37:00:01",
        "addr_type": 1,
        "scan_rsp": "0aff4d44444d34455641211124172f4461726b4d656e746f722e636f6d"
    }
```

And pcap of WHAD peripheral confirms:

[image]

I think you can close this for real now :)

(unfortunately this doesn't seem to be what's causing the Android phone to fail to connect to the fake UVP during MITM, so I'll open a different issue for that.)