whad-team/whad-client #144: No error reported if file passed for -p is missing in MITM command

XenoKovah, 2025-01-29:
I accidentally messed up my reproduction of #139 on the Lubuntu image, by making the `wble-central profile` command wrong, so the net result is it printed the profile but didn't save it to a file. When I then later did the `wble-connect | wble-spawn ... -p ./ZephyrOTS.json` command, I didn't receive any error about the fact that `./ZephyrOTS.json` didn't actually exist. So then I went back to trying just `wble-spawn` and that showed what the problem is.

It seems like I should get an error message if the file is missing such that it couldn't possibly succeed, rather than just no output, which is what I associate with a working WHAD MITM.

virtualabs, 2025-01-31:
I tried to reproduce this issue but I get an error message from `wble-spawn` (well, it is a raw Python exception uncaught by whad):

```
$ wble-connect -i hci1 xx:xx:xx:60:fc:5c | wble-spawn -i hci0 -p doesnotexist.json
[!] An unexpected exception occured:
Traceback (most recent call last):
  File "/home/virtualabs/perso/github/whad-client/whad/cli/app.py", line 784, in run_app
    application.run()
  File "/home/virtualabs/perso/github/whad-client/whad/ble/cli/ble_spawn.py", line 248, in run
    with open(self.args.profile,'rb') as profile:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'doesnotexist.json'
```

XenoKovah, 2025-02-01:
OK, I think I see what was happening now. If `wble-connect` is attempting to connect, but not succeeding, it sits there not outputting anything. But I was interpreting that as a successful connection + spawn, whereas in reality it hadn't finished the connection yet. If I let it sit until the 30s timeout (or however long it is), then it indeed proceeds through to the `wble-spawn`, which does correct handle the nonexistent profile file via an error as shown above and below. So I'll close this as it's not an issue.

```
(venv) user@bare-metal:~/BT1001/whad-client$ wble-connect  -i hci0 ca:fe:13:37:00:01 | wble-spawn  -i hci1 -p doesnotexist.json
[!] Cannot connect to ca:fe:13:37:00:01
[!] An unexpected exception occured:
Traceback (most recent call last):
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/cli/app.py", line 784, in run_app
    application.run()
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/cli/ble_spawn.py", line 248, in run
    with open(self.args.profile,'rb') as profile:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'doesnotexist.json'
```

virtualabs, 2025-02-02:
I cannot leave `wble-spawn` with such bad exception handling. I've added some code to catch IO errors that may occur when `wble-spawn` tries to access a profile JSON file (in PR #146). Now it displays meaningful error messages instead of a raw Python exception. PR has been merged into `main` branch.

XenoKovah, 2025-02-02:
OK, since you re-opened this anyway... one misc thing I was thinking was that perhaps there should be some sort of output from `wble-connect` so that it's clear when it's still attempting a connection? That way it wouldn't have an ambiguous UI where no output could mean both "attempting connection" and "connection succeeded".

virtualabs, 2025-02-02:
I've fixed this and merged PR #147 into `main`. Now `wble-connect` outputs its current status (connecting/connected) when connecting to a device. I've also fixed another crash that happened when no BD address was provided.

XenoKovah, 2025-02-19:
Just to confirm I've seen the updated connect outputting status and it LGTM.