whad-team/whad-client #152: Android phone cannot connect to WHAD fake peripheral

XenoKovah, 2025-02-19:
## Context

I am continuing to test WHAD's MITM capabilities. They worked for me for Zephyr-stock-example to Zephyr-stock-example. But now I want to try my actual use case: Zephyr-custom to Android custom app (which is the more realistic setup we're using in our class.)

Currently it's looking like it's not working. I can see ATT requests seemingly being forwarded from the WHAD fake Peripheral to fake Central, but then the real UVP never replies, as if it didn't get them for whatever reason.

**WHAD-to-UVP:**  
[image]

**Android-to-WHAD-fake-P:**  
[image]

This is in contrast to the traffic between UVP and UVC which looks like this in the absence of the MITM:

[image]

# Replication

## USB HCI dongle setup:
One of the HCI dongles (let's assume it's hci1) needs to have its BDADDR changed to ca:fe:13:37:00:01. Because the UVC will only look for and connect to that BDADDR.

The BDADDR of the 2nd device doesn't matter. Mine's still set to 11:11:11:11:11:11.

## UVP setup:
I am using the UltraVulnerablePeripheral with install instructions previously described [here](https://github.com/whad-team/whad-client/issues/143#issuecomment-2668870283) as the peripheral. 

## UVC setup:
I am using a custom Android "UltraVulnerableCentral" app that Veronica developed. The UVC connects to the UVP and acts as an interface for a basic game that exercises BT functionality.

The UVC can be installed on a phone from Ubuntu 24.04 as follows (**Note: we've only tested *this version* on Android 14** on the cheapo $30 phones we can get from Target/Walmart. Though previously we had tested with Android 11/12 ebay phones.)

```
sudo apt update
sudo apt install -y android-sdk-platform-tools
git clone https://gitlab.com/opensecuritytraining/bt1501_code_for_class_tmp.git
<make sure the Android phone is connected to the Linux VM, and has USB debugging turned on, and that you were prompted on the phone to trust the VM and said OK on the phone>
adb install -t  bt1501_code_for_class_tmp/UVC.apk
adb push bt1501_code_for_class_tmp/UVC-Bootstrap/config.ini /sdcard/Android/media/com.darkmentor.bleclass/
```

## Confirm the baseline expected behavior:

### Power on UVP

The UVP should have an LED blinking while it's disconnected and solid when it's connected.

### Connect with UVC.

Open the application and click the slider in the top right corner which has an ❌ next to it. It should attempt to connect to the UVP, and if it succeeds, you should see a 🔗 displayed, and the UVP should stop blinking.

### Confirm communications

Click into challenge 1-A, select the last option, and hit submit. It should say this is correct, and then you go back to the main screen. **This is the behavior we want to see when the MITM is in effect, but which is not currently working.**

## Steps to MITM UVC/UVP:

You should have the capability to capture two independent pcaps. I'm using Sniffle but you can use WHAD if you don't have 2x Sonoff dongles.

### Terminal 1: pcap1:
Start this first:
```
sudo apt install -y python3-serial
cd ~
git clone https://github.com/nccgroup/Sniffle
cd ~/Sniffle/python_cli
sudo python3 sniff_receiver.py -s /dev/ttyUSB0 -m ca:fe:13:37:00:01  -c 37 -o True-UVP.pcap
```

Power on the UVP and confirm you start seeing packets output by Sniffle.

### Terminal 0: WHAD:

Profile UVP:
```
wble-central -i hci0 -b ca:fe:13:37:00:01 -r profile UVP.json
```

Start the WHAD connection and impersonation:

```
sudo service bluetooth stop
wble-connect --log debug --log-file ./wble-connect-main-U2404-vmware-mac.log -i hci0 -r ca:fe:13:37:00:01 | wble-spawn --log debug --log-file ./wble-spawn-main-U2404-vmware-mac.log -i hci1 -p UVP.json
```

### Terminal 1: pcap1:

You should now see that Sniffle is following the connection between WHAD as a Central and the UVP.

### Terminal 2: pcap 2:

```
cd ~/Sniffle/python_cli
sudo python3 sniff_receiver.py -s /dev/ttyUSB0 -m ca:fe:13:37:00:01  -c 37 -o Fake-UVP-WHAD.pcap
```

You should immediately see the fake advertisements by WHAD impersonating the UVP, every ~1.3s.

### Connect with UVC.

Open the application and click the slider in the top right corner which has an ❌ next to it. It should attempt to connect to the UVP, and if it succeeds, you should see a 🔗 displayed, and the UVP should stop blinking.

### Confirm communications

Click into challenge 1-A, select the last option, and hit submit. **This is where it fails, because it seems like the UVC can't communicate to the real UVP through the MITM and get back an answer. So it times out and disconnects.**

virtualabs, 2025-02-20:
That is a very interesting scenario. I've just discovered that `wble-spawn` chained with `wble-connect` does no longer wait for a connection before connecting to the target device... I've updated `wble-connect` to fix some issues related to pending PDUs but I guess we have a similar problem here (but that's not the reason it does not work as expected with your UVC app).

During my first tests, I also noted `wble-proxy` does not provide an option to tell that we want to proxify a device identified by random or public BD address, so I've made some modifications and managed to get it working. I don't know if it could be a good thing to automatically detect the address type based on the specified BD address. Using this modified `wble-proxy` combined with the use of a CSR4.0 dongle configured with a spoofed BD address, `wble-proxy` does the job and the UVC application successfully connects to it (not checked with Sniffle, but it's all green in the app and stays green).

It seems we have again a packet forwarding issue with the `wble-connect`/`wble-spawn` combo, I'll investigate this later in the week.

XenoKovah, 2025-02-20:
> "I don't know if it could be a good thing to automatically detect the address type based on the specified BD address"

I think that would be ideal. I note that one of the problems with me manually spoofing the BDADDR right now via the copy of the `bdaddr` tool in the GATTacker repo, is that it always sets it to a *public* BDADDR. That means that when the MITM is going on, the original UVP's random-static BDADDR is being substituted with a public BDADDR, which means it's technically detectable that there is spoofing going in. Now the UVC doesn't actually care about the type of the BDADDR, and I'm sure most things don't either, but still *some* things could, and thus more accuracy is better.

virtualabs, 2025-02-26:
I've tested this again after some heavy modifications I've made to `wble-connect`, `wble-spawn` and `wble-proxy`, and it seems to work fine with my setup when using branch `bugfix/issue-159`. There are still some minor issues I guess and I need to do more tests to figure them out, but I managed to create a functional BLE proxy with `wble-proxy` and ` wble-connect -i hci1 -r ca:fe:13:37:00:01 |  wble-spawn -i hci2 -p /tmp/uvp01.json`.

I sometimes get the Android app failing at connecting to the UVP01 device but most of the time it ends up well with the app's main screen showing a green status in its top-right corner (and it stays green, I kept the connection active for more than 10 minutes without any disconnection).

By the way, the app crashes when I switch on the connection when Bluetooth is not enabled.

XenoKovah, 2025-02-26:
Alright, I'll give `bugfix/issue-159` a try and let you know.

XenoKovah, 2025-02-26:
> I sometimes get the Android app failing at connecting to the UVP01 device but most of the time it ends up well with the app's main screen showing a green status in its top-right corner (and it stays green, I kept the connection active for more than 10 minutes without any disconnection).

It sounds like you tested through the "Connect with UVC." step I mentioned above, but did you test the "Confirm communications" step I listed too?

Because right now when I use the connect-and-spawn approach, I can see that it's connecting and proxying the initial GATT enumeration communications fine, however when I click into a quiz question and submit an answer the Android UI just freezes. And I don't see the the packets in the pcap associated with that, which essentially boil down to a 3 byte write to handle 23 of "here's what I think the answer is for this question" and a read from handle 25 w/ 3 byte read response of whether it was correct or not.

[image]

[image]

XenoKovah, 2025-02-26:
Also `wble-proxy` doesn't seem to even be attempting a connect based on looking at pcap sniffing for only ca:fe:13:37:00:01 and never seeing any CONNECT_IND.

```
(venv) root@BTVM:/home/user/BT1001/whad-client# git branch
  bugfix/issue-129
  bugfix/issue-139
* bugfix/issue-159
  main
(venv) root@BTVM:/home/user/BT1001/whad-client# git log -1
commit 31f93ae52f7efeefafb985f8d6628b8245e59def (HEAD -> bugfix/issue-159, origin/bugfix/issue-159)
Author: Damien Cauquil <dcauquil@quarkslab.com>
Date:   Tue Feb 25 17:06:27 2025 +0100

    Improved `wble-connect` and `wble-spawn` to fix gattacker-style MitM.
(venv) root@BTVM:/home/user/BT1001/whad-client# pip3 install .
Processing /home/user/BT1001/whad-client
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: protobuf==3.20.* in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (3.20.3)
Requirement already satisfied: scapy==2.5.0rc3 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (2.5.0rc3)
Requirement already satisfied: elementpath==2.4.0 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (2.4.0)
Requirement already satisfied: pyserial==3.5.0 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (3.5)
Requirement already satisfied: pycryptodomex==3.11.0 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (3.11.0)
Requirement already satisfied: pyusb==1.2.1 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (1.2.1)
Requirement already satisfied: cryptography==36.0.0 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (36.0.0)
Requirement already satisfied: prompt_toolkit==3.0.43 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (3.0.43)
Requirement already satisfied: hexdump==3.3 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (3.3)
Requirement already satisfied: pynput==1.7.7 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (1.7.7)
Requirement already satisfied: Sphinx>=5.3.0 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (7.3.7)
Requirement already satisfied: sphinx-rtd-theme==2.0.0 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (2.0.0)
Requirement already satisfied: sphinxcontrib-mermaid==0.9.2 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (0.9.2)
Requirement already satisfied: requests==2.27.1 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (2.27.1)
Requirement already satisfied: distro==1.6.0 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (1.6.0)
Requirement already satisfied: cffi>=1.12 in ./venv/lib/python3.12/site-packages (from cryptography==36.0.0->whad==1.2.6) (1.17.1)
Requirement already satisfied: wcwidth in ./venv/lib/python3.12/site-packages (from prompt_toolkit==3.0.43->whad==1.2.6) (0.2.13)
Requirement already satisfied: six in ./venv/lib/python3.12/site-packages (from pynput==1.7.7->whad==1.2.6) (1.17.0)
Requirement already satisfied: evdev>=1.3 in ./venv/lib/python3.12/site-packages (from pynput==1.7.7->whad==1.2.6) (1.7.1)
Requirement already satisfied: python-xlib>=0.17 in ./venv/lib/python3.12/site-packages (from pynput==1.7.7->whad==1.2.6) (0.33)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in ./venv/lib/python3.12/site-packages (from requests==2.27.1->whad==1.2.6) (1.26.20)
Requirement already satisfied: certifi>=2017.4.17 in ./venv/lib/python3.12/site-packages (from requests==2.27.1->whad==1.2.6) (2024.12.14)
Requirement already satisfied: charset-normalizer~=2.0.0 in ./venv/lib/python3.12/site-packages (from requests==2.27.1->whad==1.2.6) (2.0.12)
Requirement already satisfied: idna<4,>=2.5 in ./venv/lib/python3.12/site-packages (from requests==2.27.1->whad==1.2.6) (3.10)
Requirement already satisfied: docutils<0.21 in ./venv/lib/python3.12/site-packages (from sphinx-rtd-theme==2.0.0->whad==1.2.6) (0.20.1)
Requirement already satisfied: sphinxcontrib-jquery<5,>=4 in ./venv/lib/python3.12/site-packages (from sphinx-rtd-theme==2.0.0->whad==1.2.6) (4.1)
Requirement already satisfied: sphinxcontrib-applehelp in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (2.0.0)
Requirement already satisfied: sphinxcontrib-devhelp in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (2.0.0)
Requirement already satisfied: sphinxcontrib-jsmath in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (1.0.1)
Requirement already satisfied: sphinxcontrib-htmlhelp>=2.0.0 in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (2.1.0)
Requirement already satisfied: sphinxcontrib-serializinghtml>=1.1.9 in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (2.0.0)
Requirement already satisfied: sphinxcontrib-qthelp in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (2.0.0)
Requirement already satisfied: Jinja2>=3.0 in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (3.1.5)
Requirement already satisfied: Pygments>=2.14 in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (2.19.1)
Requirement already satisfied: snowballstemmer>=2.0 in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (2.2.0)
Requirement already satisfied: babel>=2.9 in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (2.16.0)
[... cut ...]
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/connector/central.py", line 144, in connect
    raise PeripheralNotFound
whad.ble.exceptions.PeripheralNotFound
(venv) root@BTVM:/home/user/BT1001/whad-client# whadup
[i] Available devices
- hci0
  Type: HCIDevice
  Index: 0
  Identifier: hci0

- hci1
  Type: HCIDevice
  Index: 1
  Identifier: hci1
```

```
sudo python3 sniff_receiver.py -s /dev/ttyUSB0 -m ca:fe:13:37:00:01  -c 37 -o True-UVP2.pcap
<ctrl-c>
wireshark True-UVP2.pcap
```

[image]

virtualabs, 2025-02-26:
This `wble-proxy` exception is weird, I'll do more testing. And you're right, I did not test the "Confirm communications" step.

virtualabs, 2025-02-27:
I've made signifcant progress and discovered (and fixed) multiple bugs.

First, we had an issue with WHAD's BLE stack. I noticed that the service with UUID `444d0000-4865-6d6c-6f63-6b21f09fa4aa` was correctly discovered but not detected when I connected to the emulated device with *nRF Connect*. If the Android application cannot see this service, it seems fairly reasonable to think that there is no way for it to interact with characteristic or CCCDs belonging to this service and it could explain why there is no write request sent to the proxy. I dug into the BLE stack to discover that there was a bug when a response to a `ReadByGroupTypeRequest` is prepared: the UUID size in the returned list of attributes was determined by the first attribute to include into the list and any further attribute with a UUID size not matching the one of the first attribute was simply ignored ! This is why the service with UUID `444d0000-4865-6d6c-6f63-6b21f09fa4aa` was never returned during the primary services discovery procedure. I fixed this bug by returning a `ReadByGroupTypeResponse` when such situation occurs, leading the Central to send another `ReadByGroupTypeRequest` starting with the handle of the attribute with a longer (or shorter) UUID size. I tested this fix with `wble-proxy` and the service was correctly discovered.

Then I stumbled upon another issue: when the Android app subscribed for indications on a specific characteristic, the UVP01 device kept disconnecting. Having a look at the debug output from UVP01 helped to figure out what was happening: WHAD wrote 0x01 0x00 into the characteristic's CCCD, while it should have written 0x02 0x00 (indication bit set). After a short investigation, it turned out to be a default value set in a method's parameter that caused this mess and I quickly fixed it. I tried again with `wble-proxy` and the Android application and everything went smoothly this time. The application discovered the service, registered for indications without causing the UVP01 to disconnect, and I could send answers and get a response from the device !

I then tried to use a link-layer proxy and faced a weird issue: when registering for indication, I observed this output sent on UVP01's serial connection:

```
Connected
game_status_ccc_write() value: 0x0002
game_status_ccc_changed() value: 0x0002
game_status_indicate() start
is_indication_set=1, ind_attr=0x2dad8, indicating=0, game_status_sent=0
[19:45:38.008,117] <wrn> bt_att: No ATT channel for MTU 192
[19:45:38.008,117] <wrn> bt_gatt: No buffer available to send indication
game_status_indicate() end
game_status_indicate() start
is_indication_set=1, ind_attr=0x2dad8, indicating=0, game_status_sent=0
[19:45:39.008,575] <wrn> bt_att: No ATT channel for MTU 192
[19:45:39.008,575] <wrn> bt_gatt: No buffer available to send indication
game_status_indicate() end
game_status_indicate() start
is_indication_set=1, ind_attr=0x2dad8, indicating=0, game_status_sent=0
[19:45:40.009,002] <wrn> bt_att: No ATT channel for MTU 192
[19:45:40.009,002] <wrn> bt_gatt: No buffer available to send indication
game_status_indicate() end
game_status_indicate() start
is_indication_set=1, ind_attr=0x2dad8, indicating=0, game_status_sent=0
[19:45:41.009,429] <wrn> bt_att: No ATT channel for MTU 192
[19:45:41.009,429] <wrn> bt_gatt: No buffer available to send indication
game_status_indicate() end
game_status_indicate() start
is_indication_set=1, ind_attr=0x2dad8, indicating=0, game_status_sent=0
[19:45:42.009,857] <wrn> bt_att: No ATT channel for MTU 192
[19:45:42.009,857] <wrn> bt_gatt: No buffer available to send indication
game_status_indicate() end
game_status_indicate() start
is_indication_set=1, ind_attr=0x2dad8, indicating=0, game_status_sent=0
[19:45:43.010,284] <wrn> bt_att: No ATT channel for MTU 192
[19:45:43.010,284] <wrn> bt_gatt: No buffer available to send indication
game_status_indicate() end
game_status_indicate() start
is_indication_set=1, ind_attr=0x2dad8, indicating=0, game_status_sent=0
[19:45:44.010,711] <wrn> bt_att: No ATT channel for MTU 192
[19:45:44.010,711] <wrn> bt_gatt: No buffer available to send indication
game_status_indicate() end
```

I suppose that there is some error related to the MTU exchange procedure but I need more time to fix this. It also means that `wble-proxy` in link-layer mode and chaining `wble-connect` with `wble-spawn` will trigger this behavior (for now) and that it won't work at the moment.

Last but not least, I also modified `wble-proxy` to automatically detect the target device address type and use it automatically when setting up the emulated peripheral, so the advertised peripheral must have a similar address type and should be 100% similar to the original one.

I've pushed these fixes into `bugfix/issue-152` (and not `bugfix/issue-159` as specified in a above comment) and let you try on your setup to confirm it solves the issue when using `wble-proxy`. 

Again, thank you for your patience and your help on this it helps a lot discovering critical bugs and improving WHAD 🙏 !

XenoKovah, 2025-02-27:
Glad to see more bugs getting fixed, and glad to hear it's working for you as that means we're close! Unfortunately its still not working for me, and freezing at the answer submission stage.

FWIW I saw the above errors about "No buffer available to send indication" in the serial output, and got the below error with `wble-proxy -i hci0 -p hci1 ca:fe:13:37:00:01` (i.e. without passing `--link-layer`) which is probably worth looking into.

```
(venv) root@BTVM:/home/user/BT1001/whad-client# wble-proxy -i hci0 -p hci1 ca:fe:13:37:00:01
Scanning for target device (timeout: 30 seconds)...
Found target device ca:fe:13:37:00:01 (random)
Proxy is ready, press a key to stop.
Remote device connected
[1, 9, 16, 21, 26, 44]
[21, 26, 44]
[26, 44]
[44]
[]
>>> Characteristic 2A19 read
00000000: 49                                                I
[!] Subscribed to notification for charac. 444d0003-4865-6d6c-6f63-6b21f09fa4aa
[!] Subscribed to notification for charac. 2A19
<<< [!] Notification for charac. 2A19:
00000000: 47                                                G
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/device/device.py", line 97, in run
    self.__device.process_messages()
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/device/device.py", line 820, in process_messages
    self.dispatch_message(message)
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/device/device.py", line 774, in dispatch_message
    self.on_domain_msg(domain, message)
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/device/device.py", line 1080, in on_domain_msg
    self.__connector.on_packet(packet)
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/connector/base.py", line 751, in on_packet
    self.on_data_pdu(conn_pdu)
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/connector/central.py", line 267, in on_data_pdu
    self.__stack.on_data_pdu(pdu.metadata.connection_handle, pdu)
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/stack/__init__.py", line 77, in on_data_pdu
    self.send('ll', pdu, tag='data', conn_handle=conn_handle)
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/common/stack/layer.py", line 598, in send
    return self.send_from(self.name, destination, data, tag=tag, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/common/stack/layer.py", line 624, in send_from
    handler(data, **kwargs)
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/stack/llm/__init__.py", line 483, in on_data_pdu_recv
    self.on_data_pdu(pdu, conn_handle)
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/stack/llm/__init__.py", line 533, in on_data_pdu
    self.send(l2cap_layer, bytes(pdu.payload), fragment=pdu.LLID == 0x1)
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/common/stack/layer.py", line 598, in send
    return self.send_from(self.name, destination, data, tag=tag, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/common/stack/layer.py", line 624, in send_from
    handler(data, **kwargs)
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/stack/l2cap/__init__.py", line 85, in on_data_received
    self.on_l2cap_packet(L2CAP_Hdr(self.state.fifo[:self.state.expected_len]))
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/stack/l2cap/__init__.py", line 93, in on_l2cap_packet
    self.send('att', packet.getlayer(ATT_Hdr))
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/common/stack/layer.py", line 598, in send
    return self.send_from(self.name, destination, data, tag=tag, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/common/stack/layer.py", line 622, in send_from
    handler(source, data, **kwargs)
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/stack/att/__init__.py", line 107, in on_packet
    self.on_handle_value_notification(att_pkt.getlayer(ATT_Handle_Value_Notification))
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/stack/att/__init__.py", line 445, in on_handle_value_notification
    self.send('gatt', GattHandleValueNotification(
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/common/stack/layer.py", line 598, in send
    return self.send_from(self.name, destination, data, tag=tag, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/common/stack/layer.py", line 624, in send_from
[... 9 lines cut ...]
    ^^^^^^^^^^^^
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/profile/device.py", line 182, in value
    return self.write(val)
           ^^^^^^^^^^^^^^^
[... cut ...]
    raise GattTimeoutException
whad.ble.stack.gatt.exceptions.GattTimeoutException
>>> Characteristic 444d0004-4865-6d6c-6f63-6b21f09fa4aa written
00000000: 01 61 04                                          .a.
^C/!\ wble-proxy stopped (CTL-C)
```

From the pcap of the phone-to-WHAD-fake-UVP I can see the final packet sent was the 3 byte write of the presumed answer to handle 23 (which was also the last thing seen in the log above)
[image]

I'll try a few more times and add more context as I find it.

virtualabs, 2025-02-27:
Well, I don't really understand what this `No buffer available to send indication` means but I only had it when using `wble-connect` combined with `wble-spawn`. AFAIK, this error makes the remote device skip the indication, and this may put the Android application in an unexpected state. I'm digging into it.

And I see that I forgot to remove some debug prints in the code, I'll fix that later.

XenoKovah, 2025-02-27:
I was previously watching the pcaps while performing the action on the phone, but now that I watched the `wble-proxy` output, I can see the error actually occurs after connection, not upon sending the write to the characteristic. And indeed, even with the error on the screen currently, if I do the write, I see it show up in the `wble-proxy` CLI output. (But the pcap says the phone's not getting any reply and ultimately just timing out.) Here's the sanitized log too.

[wble-proxy-sanitized.log](https://github.com/user-attachments/files/19009068/wble-proxy-sanitized.log)

XenoKovah, 2025-02-27:
A misc thing I'll note, which isn't really critical to the functionality of the app, but which is indicative of something else not being handled: the notifications from the copied-and-pasted Zephyr Heartrate example inside the UVP are being seen correctly within the WHAD-fake-Central-to-real-UVP connection:

[image]

However, none of them seem to be being proxied through to the WHAD-fake-UVP-to-real-Android-Central connection:

[image]

XenoKovah, 2025-02-27:
And just to show I'm on the latest and correct branch:

```
commit 33c0d7908dbae0b557cebf687f596c84d63f7fb3 (HEAD -> bugfix/issue-152, origin/bugfix/issue-152)
Author: Damien Cauquil <dcauquil@quarkslab.com>
Date:   Thu Feb 27 07:41:58 2025 +0100

    Fixed multiple bugs related to BLE
    
    - BLE stack FindByGroupTypeRequest did not correctly handle attributes with different UUID sizes,
      resulting in those with different sizes to be omitted in the response sent to the Central
    - Indications were not correctly configured (mismatched as notifications) and this triggered a
      disconnection from a Peripheral as the related characteristic had no "notify" permission.
    - `wble-proxy` now automatically detects the target device address type and create an emulated
      device with same address type.
(venv) root@BTVM:/home/user/BT1001/whad-client# git branch
  bugfix/issue-129
  bugfix/issue-139
* bugfix/issue-152
  bugfix/issue-159
  main
(venv) root@BTVM:/home/user/BT1001/whad-client# pip3 install .
Processing /home/user/BT1001/whad-client
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: protobuf==3.20.* in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (3.20.3)
Requirement already satisfied: scapy==2.5.0rc3 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (2.5.0rc3)
Requirement already satisfied: elementpath==2.4.0 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (2.4.0)
Requirement already satisfied: pyserial==3.5.0 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (3.5)
Requirement already satisfied: pycryptodomex==3.11.0 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (3.11.0)
Requirement already satisfied: pyusb==1.2.1 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (1.2.1)
Requirement already satisfied: cryptography==36.0.0 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (36.0.0)
Requirement already satisfied: prompt_toolkit==3.0.43 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (3.0.43)
Requirement already satisfied: hexdump==3.3 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (3.3)
Requirement already satisfied: pynput==1.7.7 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (1.7.7)
Requirement already satisfied: Sphinx>=5.3.0 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (7.3.7)
Requirement already satisfied: sphinx-rtd-theme==2.0.0 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (2.0.0)
Requirement already satisfied: sphinxcontrib-mermaid==0.9.2 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (0.9.2)
Requirement already satisfied: requests==2.27.1 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (2.27.1)
Requirement already satisfied: distro==1.6.0 in ./venv/lib/python3.12/site-packages (from whad==1.2.6) (1.6.0)
Requirement already satisfied: cffi>=1.12 in ./venv/lib/python3.12/site-packages (from cryptography==36.0.0->whad==1.2.6) (1.17.1)
Requirement already satisfied: wcwidth in ./venv/lib/python3.12/site-packages (from prompt_toolkit==3.0.43->whad==1.2.6) (0.2.13)
Requirement already satisfied: six in ./venv/lib/python3.12/site-packages (from pynput==1.7.7->whad==1.2.6) (1.17.0)
Requirement already satisfied: evdev>=1.3 in ./venv/lib/python3.12/site-packages (from pynput==1.7.7->whad==1.2.6) (1.7.1)
Requirement already satisfied: python-xlib>=0.17 in ./venv/lib/python3.12/site-packages (from pynput==1.7.7->whad==1.2.6) (0.33)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in ./venv/lib/python3.12/site-packages (from requests==2.27.1->whad==1.2.6) (1.26.20)
Requirement already satisfied: certifi>=2017.4.17 in ./venv/lib/python3.12/site-packages (from requests==2.27.1->whad==1.2.6) (2024.12.14)
Requirement already satisfied: charset-normalizer~=2.0.0 in ./venv/lib/python3.12/site-packages (from requests==2.27.1->whad==1.2.6) (2.0.12)
Requirement already satisfied: idna<4,>=2.5 in ./venv/lib/python3.12/site-packages (from requests==2.27.1->whad==1.2.6) (3.10)
Requirement already satisfied: docutils<0.21 in ./venv/lib/python3.12/site-packages (from sphinx-rtd-theme==2.0.0->whad==1.2.6) (0.20.1)
Requirement already satisfied: sphinxcontrib-jquery<5,>=4 in ./venv/lib/python3.12/site-packages (from sphinx-rtd-theme==2.0.0->whad==1.2.6) (4.1)
Requirement already satisfied: sphinxcontrib-applehelp in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (2.0.0)
Requirement already satisfied: sphinxcontrib-devhelp in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (2.0.0)
Requirement already satisfied: sphinxcontrib-jsmath in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (1.0.1)
Requirement already satisfied: sphinxcontrib-htmlhelp>=2.0.0 in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (2.1.0)
Requirement already satisfied: sphinxcontrib-serializinghtml>=1.1.9 in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (2.0.0)
Requirement already satisfied: sphinxcontrib-qthelp in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (2.0.0)
Requirement already satisfied: Jinja2>=3.0 in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (3.1.5)
Requirement already satisfied: Pygments>=2.14 in ./venv/lib/python3.12/site-packages (from Sphinx>=5.3.0->whad==1.2.6) (2.19.1)
[... cut ...]
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/stack/gatt/__init__.py", line 847, in write
    msg = self.wait_for_message(GattWriteResponse)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/stack/gatt/__init__.py", line 179, in wait_for_message
    raise GattTimeoutException
whad.ble.stack.gatt.exceptions.GattTimeoutException
>>> Characteristic 444d0004-4865-6d6c-6f63-6b21f09fa4aa written
00000000: 01 61 04                                          .a.
k
```

virtualabs, 2025-02-27:
I think there is an issue with the connection MTU. If I understand correctly how the game works, once the Android application has subscribed for indications the game is supposed to send an indication with the game state, which is larger than 23 bytes. Therefore, Zephyr cannot create an indication response because the buffer is too small, leading to this error and the indication not being sent in the end.

**So we're back with MTU exchange procedure issues.**

Besides this, I've noted that my USB Bluetooth dongle does not accept MTU over 40 bytes and that seems to be what causes the error on the UVP side, as it still tries to send an indication with a large buffer that is greater than 40 bytes (by default ?).

```
> ACL Dat..RX flags 0x02 dlen 7  #882 252.943033
      ATT: Exchange MTU Request (0x02) len 2
        Client RX MTU: 247
< ACL Dat..TX flags 0x00 dlen 7  #883 252.955614
      ATT: Exchange MTU Response (0x03) len 2
        Server RX MTU: 23
> HCI Event:... (0x13) plen 5  #884 253.063722
        Num handles: 1
        Handle: 16 Address: CA:FE:13:37:00:01 (Static)
        Count: 1
> HCI Event:.. (0x3e) plen 11  #885 253.302715
      LE Data Length Change (0x07)
        Handle: 16 Address: CA:FE:13:37:00:01 (Static)
        Max TX octets: 40
        Max TX time: 432
        Max RX octets: 40
        Max RX time: 432
> HCI Event:... (0x3e) plen 6  #886 254.501730
      LE PHY Update Complete (0x0c)
        Status: Success (0x00)
        Handle: 16 Address: CA:FE:13:37:00:01 (Static)
        TX PHY: LE 2M (0x02)
        RX PHY: LE 2M (0x02)
```

There is also a *LE Data Length Change* event sent by the controller, but I'm not sure this 40-byte limitation cannot be changed. The thing is we surely need to tell the HCI dongle how much data we want it to handle following the *ATT MTU Exchange request*, and we don't handle it for now.

XenoKovah, 2025-02-27:
Interesting if there's a dongle issue too... FWIW you can modify the UVP's limit by editing `UltraVulnerablePeripheral/prj.conf` and setting `CONFIG_BT_CTLR_DATA_LENGTH_MAX=30` or commenting out that line (in which case it appears to default to 27). Though neither made a difference for me, I still get the same error on connect. (I get that WHAD's just rejecting the LL_LENGTH_REQ though, so it shouldn't necessarily make a difference.)

[image]

[image]

XenoKovah, 2025-02-27:
Perhaps it's time to selectively add support for LL_LENGTH_REQ/RSP? :)

Because the point about Zephyr can't send the indication because it's 192 bytes large, doesn't seem like it's necessarily the core issue; because in the default case without the MITM when they successfully negotiate a 40 byte MTU, Zephyr happily fragments it. 

[image]

So it'd be weird to me if it didn't fragment in the can't-negotiate-a-non-default-MTU case, *unless* there's some special logic about how it deals with getting explicitly rejected for an LL_LENGTH_REQ.

XenoKovah, 2025-02-27:
Nice, this version fixed the hang at answer submission! Unfortunately it's still not quite correct. What you should see after you submit question 1-A with answer 4, is that it says correct, but then when you navigate back to the front page, the emoji for 1-A on the front page is also updated to be the trophy (or frown if you get it wrong). The front page UI update is handled by that large indication. However, no front page update suggests the indication is not fully handled correctly. 

And indeed, it looks to me like the full and correct indication is getting back from the true UVP to the WHAD Central (and defragmented)...

[image]

```<<< [!] Notification for charac. 2A19:
00000000: 49                                                I
<<< [!] Indication for charac. 444d0003-4865-6d6c-6f63-6b21f09fa4aa:
00000000: 01 61 02 01 62 00 01 63  00 01 64 00 01 65 00 01  .a..b..c..d..e..
00000010: 66 00 01 67 00 02 61 02  02 62 00 02 63 00 02 64  f..g..a..b..c..d
00000020: 00 02 65 00 02 66 00 02  67 00 03 61 00 03 62 00  ..e..f..g..a..b.
00000030: 03 63 00 03 64 00 03 65  00 03 66 00 03 67 00 04  .c..d..e..f..g..
00000040: 61 00 04 62 00 04 63 00  04 64 00 04 65 00 04 66  a..b..c..d..e..f
00000050: 00 04 67 00 05 61 00 05  62 00 05 63 00 05 64 00  ..g..a..b..c..d.
00000060: 05 65 00 05 66 00 05 67  00 06 61 00 06 62 00 06  .e..f..g..a..b..
00000070: 63 00 06 64 00 06 65 00  06 66 00 06 67 00 07 61  c..d..e..f..g..a
00000080: 00 07 62 00 07 63 00 07  64 00 07 65 00 07 66 00  ..b..c..d..e..f.
00000090: 07 67 00 08 61 00 08 62  00 08 63 00 08 64 00 08  .g..a..b..c..d..
000000A0: 65 00 08 66 00 08 67 00  09 61 00 09 62 00 09 63  e..f..g..a..b..c
000000B0: 00 09 64 00 09 65 00 09  66 00 09 67 00           ..d..e..f..g.
<<< [!] Notification for charac. 2A19:
00000000: 48                                                H
<<< [!] Indication for charac. 444d0003-4865-6d6c-6f63-6b21f09fa4aa:
00000000: 01 61 02 01 62 00 01 63  00 01 64 00 01 65 00 01  .a..b..c..d..e..
00000010: 66 00 01 67 00 02 61 02  02 62 00 02 63 00 02 64  f..g..a..b..c..d
00000020: 00 02 65 00 02 66 00 02  67 00 03 61 00 03 62 00  ..e..f..g..a..b.
00000030: 03 63 00 03 64 00 03 65  00 03 66 00 03 67 00 04  .c..d..e..f..g..
00000040: 61 00 04 62 00 04 63 00  04 64 00 04 65 00 04 66  a..b..c..d..e..f
00000050: 00 04 67 00 05 61 00 05  62 00 05 63 00 05 64 00  ..g..a..b..c..d.
00000060: 05 65 00 05 66 00 05 67  00 06 61 00 06 62 00 06  .e..f..g..a..b..
00000070: 63 00 06 64 00 06 65 00  06 66 00 06 67 00 07 61  c..d..e..f..g..a
00000080: 00 07 62 00 07 63 00 07  64 00 07 65 00 07 66 00  ..b..c..d..e..f.
00000090: 07 67 00 08 61 00 08 62  00 08 63 00 08 64 00 08  .g..a..b..c..d..
000000A0: 65 00 08 66 00 08 67 00  09 61 00 09 62 00 09 63  e..f..g..a..b..c
000000B0: 00 09 64 00 09 65 00 09  66 00 09 67 00           ..d..e..f..g.
```
but then not passed through to the UVC.:

[image]

(Also the Heartrate notifications are passing through to the UVC correctly in this version too. 👍)

Also FWIW AFAIK (since Veronica programmed it) the UVC doesn't actually control (m)any low level stuff. That's all just down to however the Android stack behaves. (I know this because I wanted her to set some things like hardcoded CONNECT_IND parameters to use in quiz questions, but she said that stuff wasn't exposed in Android.)

XenoKovah, 2025-07-13:
I don't really know when this was fully fixed, but this thread seems stale because I can see with the latest main (fecb6f675bc90dd4297bfa3661804ad132451d3f) the front page update via indication seems to work just fine. I think this should be closed at this point since there are other newer tickets handling any remaining issues.