whad-team/whad-client #154: wble-proxy doesn't connect successfully even when manual commands do

XenoKovah, 2025-02-20:
I just wanted to note that it seems that commit e403f7070999721ba1da9707a377a7d78eb1e52c of main,  `wble-proxy -i hci0 -p hci1  ca:fe:13:37:00:01` (with or without `--link-layer`) doesn't seem to ever connect successfully to `ca:fe:13:37:00:01` (based on CLI output and pcap), despite a multi-command approach in #152 working). This seems like it's likely a separate issue than #152, so I'm adding a separate tracker.

XenoKovah, 2025-02-20:
Also I'm assuming this timeout exception is supposed to be caught and handled more gracefully.

```
(venv) root@BTVM:/home/user/BT1001/whad-client# wble-proxy -i hci1 -p hci0  ca:fe:13:37:00:01
Scanning for target device (timeout: 30 seconds)...
[!] An unexpected exception occured:
Traceback (most recent call last):
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/cli/app.py", line 784, in run_app
    application.run()
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/cli/ble_proxy.py", line 229, in run
    self.spawn_proxy()
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/cli/ble_proxy.py", line 288, in spawn_proxy
    proxy.start()
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/tools/proxy.py", line 841, in start
    self.__target = self.__central.connect(self.__target_bd_addr)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/BT1001/whad-client/venv/lib/python3.12/site-packages/whad/ble/connector/central.py", line 144, in connect
    raise PeripheralNotFound
whad.ble.exceptions.PeripheralNotFound
```

virtualabs, 2025-02-20:
That's something I've noticed when investigating issue #152, and I fixed this but did not push the changes yet. I know exactly why this is happening and will issue a proper fix for it soon.

Whiskerc, 2025-02-25:
I'm seeing the same error on my setup using two Makerdiary nRF52840 MDK dongles. 
Command line: `wble-proxy -i uart:/dev/ttyACM0 -p uart:/dev/ttyACM1 --link-layer -s xx:xx:xx:xx:xx:xx` 
This is on a raspberry Pi 5 running raspbian. 

If there is any guidance, or help I can do to test the fix please let me know

virtualabs, 2025-07-14:
Fine, I'm closing this issue.

closed 2025-07-14 by virtualabs