whad-team/whad-client #161: Documentation Error Zigbee, IEEE 802.15.4

gertrudem, 2025-03-07:
In the [read the docs](https://whad.readthedocs.io/en/latest/zigbee/started.html) page for Zigbee, there is no mention of `sniffer.start()` for the sniffing example code.

Furthermore, for the [IEEE 802.15.14](https://whad.readthedocs.io/en/latest/dot15d4/started.html) section, there should not be a timeout parameter for the sniffing for loop, unlike the timeout that exists for Zigbee. This is because the code doesn't support it:

whad-client/whad/dot15d4/connector/__init__.py:
```
def sniff_dot15d4(self, channel:int = 11) -> bool:
        """
        Sniff 802.15.4 packets (on a single channel).
        """
        if not self.can_sniff():
            raise UnsupportedCapability("Sniff")

        # Create a SniffMode message
        msg = self.hub.dot15d4.create_sniff_mode(channel)

        resp = self.send_command(msg, message_filter(CommandResult))
        return isinstance(resp, Success)

```

virtualabs, 2025-03-10:
You are totally right about the code not supporting a timeout parameter in the IEEE 802.15.4 section and this Zigbee example code. Will fix this pretty soon.

gertrudem, 2025-03-13:
Hi, I just tested the timeout on IEEE 802.15.4, and it works very well on the branch. However the code documentation example on readthedocs has not been corrected.

virtualabs, 2025-03-13:
Perfect. Documentation on ReadTheDocs is synced with the repository's main branch, it will be updated once the modifications are merged into it. In the meantime, you can build the doc locally using the following commands and check that the example code has correctly been updated. Of course, you need to activate a Python virtual environment if you have one defined for this project.

```shell
$ cd doc
$ make html
$ firefox build/html/index.html
```

Regarding the last command that uses `firefox`, you can use any web browser you like to browse the generated documentation 😅. 

Let me know if the documentation is correctly fixed or not  (I've checked on my machine and it seems to be ok). Once the changes merged into `main`, they will be reflected into the ReadTheDocs online documentation.

gertrudem, 2025-03-14:
Zigbee looks good on the new doc, I think you're good to push to main

gertrudem, 2025-03-14:
Oh sorry, one last issue I just saw was that in the 802.15.4 sending packets code example for [readthedocs](https://whad.readthedocs.io/en/latest/dot15d4/started.html), I think the wrong thing is imported.

I think it should be `from whad.dot15d4 import Dot15d4` instead of `from whad.dot15d4 import Sniffer`

virtualabs, 2025-03-16:
And you're right, I've updated the `bugfix/issue-161` branch to fix this. Nice catch (again) 😄 !