whad-team/whad-client #169: BLE MiTM exception: TypeError: Packet.__rdiv__() missing 1 required positional argument: 'other'

DuxSec, 2025-04-11:
I am encountering a TypeError when attempting to create a (keyboard) MiTM scenario with the wble-spawn command in the whad-client. The error seems related to the use of the / operator in the scapy library when constructing HCI commands.

I build from the main github repo source.

OS: Debian GNU/Linux 12.6
Python Version: 3.11
Scapy version: 2.5.0rc3
wble-connect 1.2.6
wble-spawn 1.2.6
wshark 1.2.6

Command that leads to the issue
```
(venv) root@debian-gnu-linux-12-6:~/pro# wble-connect --interface hci0 ff:ff:11:ad:ba:bc | wshark | wble-spawn --interface hci1 -p modified.json
```

output
```
Connecting to ff:ff:11:ad:ba:bc ...
Connected to ff:ff:11:ad:ba:bc
[!] An unexpected exception occured:
Traceback (most recent call last):
  File "/root/whad-client/venv/lib/python3.11/site-packages/whad/cli/app.py", line 784, in run_app
    application.run()
  File "/root/whad-client/venv/lib/python3.11/site-packages/whad/ble/cli/ble_spawn.py", line 369, in run
    self.create_input_proxy(adv_data, scan_rsp, int(self.args.conn_handle),
  File "/root/whad-client/venv/lib/python3.11/site-packages/whad/ble/cli/ble_spawn.py", line 407, in create_input_proxy
    peripheral = Peripheral(self.interface, adv_data=adv_data, scan_data=scan_data,
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/whad-client/venv/lib/python3.11/site-packages/whad/ble/connector/peripheral.py", line 207, in __init__
    self.set_bd_address(bd_address, public=public)
  File "/root/whad-client/venv/lib/python3.11/site-packages/whad/ble/connector/base.py", line 516, in set_bd_address
    resp = self.send_command(msg, message_filter(CommandResult))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/whad-client/venv/lib/python3.11/site-packages/whad/device/connector.py", line 346, in send_command
    return self.__device.send_command(message, filter)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/whad-client/venv/lib/python3.11/site-packages/whad/device/device.py", line 691, in send_command
    self.send_message(command, keep)
  File "/root/whad-client/venv/lib/python3.11/site-packages/whad/device/device.py", line 1122, in send_message
    self._on_whad_message(message)
  File "/root/whad-client/venv/lib/python3.11/site-packages/whad/device/device.py", line 1132, in _on_whad_message
    getattr(self, callback_name)(message)
  File "/root/whad-client/venv/lib/python3.11/site-packages/whad/device/virtual/hci/__init__.py", line 880, in _on_whad_ble_set_bd_addr
    if self._set_bd_address(message.bd_address, message.addr_type):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/whad-client/venv/lib/python3.11/site-packages/whad/device/virtual/hci/__init__.py", line 531, in _set_bd_address
    self._write_command(command, wait_response=False)
  File "/root/whad-client/venv/lib/python3.11/site-packages/whad/device/virtual/hci/__init__.py", line 291, in _write_command
    hci_command /= command
  File "/root/whad-client/venv/lib/python3.11/site-packages/scapy/packet.py", line 601, in __div__
    return other.__rdiv__(self)  # type: ignore
           ^^^^^^^^^^^^^^^^^^^^
TypeError: Packet.__rdiv__() missing 1 required positional argument: 'other'
```

virtualabs, 2025-04-12:
What hardware (bluetooth adapter) are you using as *hci1* ? You may use `hciconfig`, `lsusb` and `bt-adapter` to get more information on it.

virtualabs, 2025-04-12:
I think I've figured out the cause of this issue. I've pushed a candidate fix in branch `bugfix/issue-169`, you can checkout this branch and give it a try. Please note that a recent change in the main branch requires a `git submodule init` command to be issued after checkout before installation.

DuxSec, 2025-04-12:
It seems to work!
```
(env) root@debian-gnu-linux-12-6:~/bugfix/whad-client# wble-proxy -i hci0 -p hci2  -w ff:ff:11:ad:ba:bc
Scanning for target device (timeout: 30 seconds)...
Found target device ff:ff:11:ad:ba:bc (public)
Proxy is ready, press a key to stop.
```

virtualabs, 2025-04-12:
Fix merged into main in PR #170.

closed 2025-04-12 by virtualabs