whad-team/whad-client #177: (Request) Flipper Zero Support

alphafox02, 2025-04-21:
First just saying I'm glad I had another look at Mirage - it brought me here again.

Second, just a follow up for an old request.

https://github.com/RCayre/mirage/issues/29

virtualabs, 2025-04-21:
Flipper Zero's Bluetooth controller is integrated into ST Micro's STM32WB55 and uses a proprietary BLE stack maintained by ST. This stack runs on a specific protected core, using protected memory, and is shipped in a signed and encrypted container. It is technically possible to break the protected environment and take control of the protected core, but AFAIK there is no real jailbreaking process for this SoC. Moreover, the BLE controller is completely undocumented and a lot of efforts need to be put into reverse-engineering its core registers from the decrypted proprietary BLE stack, understand what these registers are used for and eventually write a tiny BLE hackable BLE stack to allow raw BLE PDU sniffing and injection.

For now, the only compatibility we can imagine is similar to using a Flipper Zero as a BLE central or peripheral (like any HCI adapter), without any further possibilities. This still can be done easily by relying on the Flipper's default BLE stack, but I can't see any benefit using a Flipper Zero rather than an internal (or USB) Bluetooth adapter. 

Moreover, it would make more sense to me to be able to hack the Flipper's BLE stack to allow BLE PDU sniffing and injection in order to implement some attacks like connection hijacking, reactive jamming, advertisement spamming and why not breaking encryption used in some weak pairings, using a dedicated app on the Flipper.

alphafox02, 2025-04-23:
Thank you for taking time to respond. I still have my same flipper zero. I’ll look closer at this repo and see how it’s meant to be used vs Mirage and if there’s ever anything I can test with the flipper zero, I’d be happy to do so.

Grouty75, 2025-05-09:
Hi I just tried updating flipper and now its crashed and now it's dead 🙃  what can I do???  🤯

Grouty75, 2025-05-09:
??ello?

closed 2025-05-09 by alphafox02