whad-team/whad-client #178: Can't implement a LoRa Meshtastic sniffer in Python

MKesenheimer, 2025-04-23:
I'm trying to implement a LoRa Meshtastic sniffer in Python with WHAD. Basically I want to write my own `wsniff` module specifically to receive Meshtastic packets.

I verified that I can receive packets with the `wsniff` command:

```
wsniff -i uart0 phy --lora -f 869525000  -sf 11 -cr 45 -bw 250000 -em -crc -w "2b" --preamble-length 8
```

I am using the Seeedstudio-e5-mini with the whad compatible firmware.

First, I tried with the LoRa connector.
```
from whad.device import WhadDevice
from whad.phy.connector.lora import LoRa

# Create our device
device = WhadDevice.create("uart0")

# Create our LoRa connector
lora = LoRa(device)

# Configure our LoRa sniffer
lora.frequency = 869525000
lora.sf = 11
lora.cr = 45
lora.bw = 250000
lora.preamble_length = 8
lora.syncword = b"\x2b"
lora.enable_crc = True
lora.enable_explicit_mode = True

# Start sniffing
lora.start()

# Listen for any valid packet received
while True:
    packet = lora.wait_packet()
    if packet is not None:
        packet.show()
```

Second, I configured a sniffer and tried that:
```
from whad.device import WhadDevice
from whad.phy.connector.sniffer import Sniffer
from whad.phy.sniffing import SnifferConfiguration, LoRaConfiguration

# Create our WHAD device
device = WhadDevice.create("uart0")

# Create our sniffing configuration
config = SnifferConfiguration()
lora = LoRaConfiguration()

# Configure demodulation
config.lora = True
config.frequency = 869525000
config.syncword = b"\x2b"
lora.spreading_factor = 11
lora.coding_rate = 45
lora.bandwidth = 250000
lora.preamble_length = 8
lora.enable_crc = True
lora.enable_explicit_mode = True
config.lora_configuration = lora
print(config.lora_configuration)

# Create a sniffer based on our configuration
sniffer = Sniffer(device)
sniffer.configuration = config

# Start sniffing
sniffer.start()

# Listen for any valid packet received
for packet in sniffer.sniff():
    packet.show()
```

I couldn't receive any packets with both scripts. Are the scripts correct or am I doing something wrong?

I also have several remarks:

- the documentation is a bit unspecific how to configure a LoRa sniffer with python
- the code is inconsistent. For example, sometimes it is called `lora.coding_rate`, sometimes `lora.cr` and so on.
- furthermore, the documentation is not correct at different locations, for example at [Getting started](https://whad.readthedocs.io/en/latest/phy/started.html). I had to change the following lines
```
from whad.phy import SnifferConfiguration, Sniffer
```
to
```
from whad.phy.connector.sniffer import Sniffer
from whad.phy.sniffing import SnifferConfiguration
```
- receiving and displaying packets is inconsistent. Sometimes we have to call (note that I had to include `if packet is not None:` opposed to the documentation:
```
while True:
    packet = lora.wait_packet()
    if packet is not None:
        packet.show()
```
sometimes the call to display packets is like this:
```
for packet in sniffer.sniff():
    packet.show()
```
It would be great, if the calls are similar to only remember one syntax.

Thanks for your great work! I appreciate any input from you.

virtualabs, 2025-05-05:
I tried your second attempt at sniffing Meshtastic packet and figured out what's wrong with it. `SnifferConfiguration` defines a `sync_word` property used to configure the demodulator, but in your code you set a `syncword` property (and not `sync_word`), thus causing your sniffer to use the default syncword 0xAA.

Quick fix to get it working:

```python
# Configure demodulation
config.lora = True
config.frequency = 869525000
config.sync_word = b"\x2b"
```

And yes, it points again to code inconsistencies as sometimes it is defined as `syncword` and some others `sync_word` 😅.

virtualabs, 2025-05-06:
Note to self:

- Improve LoRa connector:
  - [ ]  as a transition, implement explicit setters/getters for manipulating LoRa settings and prevent any error (while keeping the old code to remain compatible with previous tools that may have been developed)

  - [ ] add predefined constants for spreading factor, coding rate, and bandwidth to limit the use of unsupported values by users

- Improve code consistency:
  - [ ] Define a unique syntax for LoRa settings (coding rate, spreading factor, bandwith, preamble length, explicit mode, crc and synchronization word), and use it everywhere
  - [ ] Define a *sniffer* interface and use it in all our sniffers. Sniffers are specialized connectors that must expose the same methods

- Improve documentation
  - [ ] add some basic examples with this LoRa connector (configuration, receiving and sending packets)
  - [ ] add deprecation warnings for property setters to force users to rely on explicit setters/getters (rather than directly setting properties of the `LoRa` connector) -- only in documentation for now, we will later issue a warning when properties are directly set instead of using the explicit setters
  - [ ] document the difference between a *sniffer* and a *connector* (promiscuous mode vs. reception mode) and the way they handle packets
  - [ ] add sniffer interface into our Python API documentation
  - [ ] update documentation to reflect the use of our new sniffer interface, especially the standard way to sniff packets

MKesenheimer, 2025-05-07:
Thanks for your comments and the detailed explanation. Both scripts are working now. For completeness and if anyone is interested in the code, I uploaded both scripts (had to rename *.py to *.txt).

[meshtastic-receiver.txt](https://github.com/user-attachments/files/20078661/meshtastic-receiver.txt)
[meshtastic-sniffer.txt](https://github.com/user-attachments/files/20078664/meshtastic-sniffer.txt)

virtualabs, 2025-05-07:
I'm keeping this one open as a reminder.