whad-team/whad-client #181: Bug?: indication callback "characteristic" is an int

XenoKovah, 2025-05-05:
I wrote a simple test to enable notifications and indications on every characteristic that supported either when connecting to the UVP, based on the example here: https://whad.readthedocs.io/en/stable/ble/started.html

```
from whad.ble import Central
from whad.ble.profile import UUID
from whad.device import WhadDevice
from struct import unpack
import sys

def on_charac_update(characteristic, value, indication=False):
    global out_of_bounds
    if indication:
        print(f"[indication] {characteristic.uuid if characteristic.uuid else "Unknown"} sent value: {value}")
    else:
        print(f"[notification] {characteristic.uuid if characteristic.uuid else "Unknown"} sent value: {value}")

if len(sys.argv) >= 2:
    interface = sys.argv[1]
    # Create the Central connector & the WHAD device
    central = Central(WhadDevice.create(interface))

    # Connect to a specific device
    device = central.connect('ca:fe:13:37:00:01', random=True)

    # # Discover and display the profile
    print("[i] Discovering profile...")
    device.discover()
    print("[i] Discovered profile: ")
    for service in device.services():
        print('-- Service %s' % service.name)
        for charac in service.characteristics():
            print(' + Characteristic %s' % charac.name)
            for desc in charac.descriptors():
                print('   + Descriptor %s' % desc.name)

            if(charac.can_notify()):
                c = device.get_characteristic(UUID(service.uuid), UUID(charac.uuid))
                c.subscribe(
                    notification=True,
                    callback=on_charac_update
                )
                print("[i] Subscribed to notifications for %s" % charac.name)
            if(charac.can_indicate()):
                c = device.get_characteristic(UUID(service.uuid), UUID(charac.uuid))
                c.subscribe(
                    indication=True,
                    callback=on_charac_update
                )
                print("[i] Subscribed to indications for %s" % charac.name)

    # Disconnect
    device.disconnect()
    central.close()
else:
    print("Usage: ", sys.argv[0]+" <interface>")

```

However, I get the following error:
```
Exception has occurred: AttributeError
'int' object has no attribute 'uuid'
  File "/home/user/BT1501/whad-client/examples/ble/ble_central_notify_indicate.py", line 10, in on_charac_update
    print(f"[indication] {characteristic.uuid if characteristic.uuid else "Unknown"} sent value: {value}")
                                                 ^^^^^^^^^^^^^^^^^^^
AttributeError: 'int' object has no attribute 'uuid'
```

Which deems to indicate the characteristic variable being passed through to the callback isn't an object (whereas it is for the notification, which prints out the UUID fine, and seems to be a whad.ble.profile.device.PeripheralCharacteristic object which is more like what I'd expect.) Is this expected or a bug?

virtualabs, 2025-05-05:
It is clearly a bug, indication and notification callbacks  are supposed to be called with an instance of `PeripheralCharacteristic` as first argument, then the characteristic value and a third argument indicating if it is an indication or a notification. This third argument is provided in the case you may want to share a single callback function (or method) between characteristics.

Working on a fix.

virtualabs, 2025-05-05:
I've just pushed branch `bugfix/issue-181` that implements a candidate fix for this issue.

XenoKovah, 2025-05-05:
Doesn't seem to fix it for me (and I did pip reinstall it)

[image]

XenoKovah, 2025-05-05:
Doesn't seem to fix it for me (and I made sure I was in my venv and git pulled and then pip installed the branch)

[image]

It did seem to introduce a timing change though in that I had to add a sleep otherwise the code exited before I ever got any notification or indication (unlike previously)

XenoKovah, 2025-05-05:
OK, sorry if you saw the other messages that I deleted, but it looks like this did fix it for me. (I had apparently checked out a throwaway copy of WHAD in one tab and forgot to throw it away). The 1st parameter (characteristic) is now properly showing up as a whad.ble.profile.device.PeripheralCharacteristic object in the indication callback.

XenoKovah, 2025-05-05:
Misc question though (somewhat related to CLUES): If I try to change the print from .uuid to .name it doesn't work. Why does a whad.ble.profile.device.characteristic.Characteristic have a .name field and whad.ble.profile.device.PeripheralCharacteristic doesn't?

virtualabs, 2025-05-05:
TLDR; I forgot to include the `name` property in this `whad.ble.profile.device.PeripheralCharacteristic` ... 

The `whad.ble.profile.device.PeripheralCharacteristic` class is just a wrapper around the default `Characteristic` class used to trigger GATT procedures when needed. By default, the `Characteristic` class used by the `GenericProfile` class (storing the attribute database and some extra metadata) is a basic model for GATT charateristic but not tied to the BLE stack. The wrapper class allows to keep a reference of the BLE GATT stack layer to trigger specific procedures.

virtualabs, 2025-05-05:
Allright, I've included a fix in the `bugfix/issue-181` branch to add the `name` property to peripheral's characteristic and service classes.

XenoKovah, 2025-05-05:
Works now, thanks

virtualabs, 2025-05-06:
Merged branch `bugfix/issue-181`  into `main`.

closed 2025-05-06 by virtualabs