whad-team/whad-client #182: Bug: Missing Characteristic Descriptor 0x2901 from characteristic.descriptors()

XenoKovah, 2025-05-05:
Our latest version of the UVP has added a type 0x2901 "Characteristic User Description" Descriptor to Characteristic `444d0010-4865-6d6c-6f63-6b21f09fa4aa`. I can see that the discovery process is seeing the presence come back in the packets...

[image]

However, if I use the following code:

```
   for service in device.services():
        print('-- Service %s' % service.name)
        for charac in service.characteristics():
            print(' + Characteristic %s' % charac.name)
            for desc in charac.descriptors():
                print('   + Descriptor %s' % desc.name)
```

It only ever prints out the 0x2902 entry, never the 0x2901.

```
-- Service 444d0000-4865-6d6c-6f63-6b21f09fa4aa
 + Characteristic 444d0001-4865-6d6c-6f63-6b21f09fa4aa
 + Characteristic 444d0002-4865-6d6c-6f63-6b21f09fa4aa
 + Characteristic 444d0010-4865-6d6c-6f63-6b21f09fa4aa
   + Descriptor Client Characteristic Configuration (0x2902)
 + Characteristic 444d0012-4865-6d6c-6f63-6b21f09fa4aa
 + Characteristic 444d0013-4865-6d6c-6f63-6b21f09fa4aa
 + Characteristic 444d0011-4865-6d6c-6f63-6b21f09fa4aa
 + Characteristic 444d0020-4865-6d6c-6f63-6b21f09fa4aa
 + Characteristic 444d0021-4865-6d6c-6f63-6b21f09fa4aa
 + Characteristic 444d0022-4865-6d6c-6f63-6b21f09fa4aa
```

And indeed, manually inspecting the descriptors() in a debugger shows only a single entry:

[image]

XenoKovah, 2025-05-05:
Also I guess one thing I do see that looks like it's missing in the pcap right now is reading the value of the user description from the 0x2901. But perhaps that will be automatically fixed once it doesn't lose track of the 0x2901 itself.

virtualabs, 2025-05-07:
The current version of our BLE stack looks only for a *Client Characteristic Configuration* descriptor attached to each characteristic during discovery, and therefore does not detect any other descriptor type. This is quite inconsistent because it is also possible when defining a GATT profile to specify a *Characteristic User Description* descriptor...

I made some changes to the BLE stack but also to the `GenericProfile` class to implement a more generic descriptor discovery mechanism (including reading each descriptor's value). Descriptors are now included in any GATT profile JSON file and imported as well from a saved JSON profile. I've pushed a new branch called `bugfix/issue-182` that reflects these changes, you can try it and check if that fits your needs.

XenoKovah, 2025-05-07:
LGTM! 👍

```
    for service in device.services():
        print('-- Service %s' % service.name)
        for charac in service.characteristics():
            print(' + Characteristic %s' % charac.name)
            for desc in charac.descriptors():
                print('   + Descriptor %s' % desc.name)
                if(desc.value):
                    print('   + Descriptor value %s' % desc.value)
```
yields
```
-- Service Generic Attribute (0x1801)
 + Characteristic Service Changed (0x2A05)
   + Descriptor Client Characteristic Configuration (0x2902)
   + Descriptor value b'\x00\x00'
 + Characteristic Client Supported Features (0x2B29)
 + Characteristic Database Hash (0x2B2A)
-- Service Generic Access (0x1800)
 + Characteristic Device Name (0x2A00)
 + Characteristic Appearance (0x2A01)
 + Characteristic Peripheral Preferred Connection Parameters (0x2A04)
-- Service Battery (0x180F)
 + Characteristic Battery Level (0x2A19)
   + Descriptor Client Characteristic Configuration (0x2902)
   + Descriptor value b'\x00\x00'
   + Descriptor Characteristic Presentation Format (0x2904)
   + Descriptor value b"\x04\x00\xad'\x01\x06\x01"
-- Service Device Information (0x180A)
 + Characteristic Model Number String (0x2A24)
 + Characteristic Manufacturer Name String (0x2A29)
-- Service 444d0000-4865-6d6c-6f63-6b21f09fa4aa
 + Characteristic 444d0001-4865-6d6c-6f63-6b21f09fa4aa
 + Characteristic 444d0002-4865-6d6c-6f63-6b21f09fa4aa
 + Characteristic 444d0010-4865-6d6c-6f63-6b21f09fa4aa
   + Descriptor Characteristic User Description (0x2901), '1337ness'
   + Descriptor value b'1337ness'
   + Descriptor Client Characteristic Configuration (0x2902)
   + Descriptor value b'\x00\x00'
 + Characteristic 444d0012-4865-6d6c-6f63-6b21f09fa4aa
 + Characteristic 444d0013-4865-6d6c-6f63-6b21f09fa4aa
 + Characteristic 444d0011-4865-6d6c-6f63-6b21f09fa4aa
 + Characteristic 444d0020-4865-6d6c-6f63-6b21f09fa4aa
 + Characteristic 444d0021-4865-6d6c-6f63-6b21f09fa4aa
 + Characteristic 444d0022-4865-6d6c-6f63-6b21f09fa4aa
-- Service Object Transfer (0x1825)
 + Characteristic OTS Feature (0x2ABD)
 + Characteristic Object Name (0x2ABE)
 + Characteristic Object Type (0x2ABF)
 + Characteristic Object Size (0x2AC0)
 + Characteristic Object ID (0x2AC3)
 + Characteristic Object Properties (0x2AC4)
 + Characteristic Object Action Control Point (0x2AC5)
   + Descriptor Client Characteristic Configuration (0x2902)
   + Descriptor value b'\x00\x00'
 + Characteristic Object List Control Point (0x2AC6)
   + Descriptor Client Characteristic Configuration (0x2902)
   + Descriptor value b'\x00\x00'
```

virtualabs, 2025-05-07:
Fix merged into `main` branch through PR #189.

closed 2025-05-07 by virtualabs