whad-team/whad-client #183: Bug: "'Central' object has no attribute 'auto'" when testing example

XenoKovah, 2025-05-05:
I copied the code from https://whad.readthedocs.io/en/stable/ble/started.html#sending-and-receiving-pdu into the `ble_central_basic.py` example as follows:

```
from whad.ble import Central
from whad.ble.profile import UUID
from whad.device import WhadDevice
from scapy.layers.bluetooth4LE import *
import sys

if len(sys.argv) >= 2:
    interface = sys.argv[1]
    # Create the Central connector & the WHAD device
    central = Central(WhadDevice.create(interface))

    # Connect to a specific device
    device = central.connect('ca:fe:13:37:00:01', random=True)

    # Discover and display the profile
    print("[i] Discovering profile...")
    device.discover()
    print("[i] Discovered profile")
    for service in device.services():
        print('-- Service %s' % service.uuid)
        for charac in service.characteristics():
            print(' + Characteristic %s' % charac.uuid)

    # Get the device name
    device_name = device.get_characteristic(UUID(0x1800), UUID(0x2A00))
    print("[i] Device name: ", device_name.value)

    # Make sure connection has succeeded
    if device is not None:

        # Disable auto mode
        central.auto(False)

        # Send a LL_VERSION_PDU
        central.send_pdu(BTLE_DATA()/BTLE_CTRL()/LL_VERSION_IND(
            version = 0x0D,
            company = 0x1337,
            subversion = 0x1337
        ))

        # Wait for a PDU
        while central.is_connected():
            pdu = central.wait_pdu()
            if pdu.haslayer(LL_VERSION_IND):
                pdu[LL_VERSION_IND].show()
                break

    # Disconnect
    device.disconnect()
    central.close()
else:
    print("Usage: ", sys.argv[0]+" <interface>")

```

When it gets to `central.auto(False)` it gives the error:

```
Exception has occurred: AttributeError
'Central' object has no attribute 'auto'
  File "/home/user/BT1501/whad-client/examples/ble/my_LL_central.py", line 32, in <module>
    central.auto(False)
    ^^^^^^^^^^^^
AttributeError: 'Central' object has no attribute 'auto'
```

So I'm assuming this is just out of date documentation, but I skimmed through the definition of the Central class, and it wasn't clear to me what the replacement should be for `central.auto(False)` (if any)

XenoKovah, 2025-05-05:
Also, misc thing if this will just end up being a documentation change: LL_VERSION_IND might not be the best example to use, because apparently despite the [fix](https://github.com/secdev/scapy/pull/4662/files) for the LL_VERSION_IND sub-version field being defined wrong endian, being merged, that hasn't made its way into a release yet...

virtualabs, 2025-05-05:
You are right,  this piece of documentation is outdated. This old `auto()` method was added to force any connector to keep any received packet in a queue rather than forwarding it to a protocol stack. Disabling `auto` mode by calling `central.auto(False)` and then calling `central.wait_pdu()` was an attempt at providing the user a way to temporarily interrupt its automated processing of packets and take control. 

These methods have been respectively replaced by `enable_synchronous(enable: bool)` and `wait_packet(timeout: float = None)` that provide the same feature but in a more generic way.

XenoKovah, 2025-05-05:
This doesn't seem to yield any LL_VERSION_IND packet to be sent. I removed the device enumeration just in case that was affecting things, but it yielded the same result.

[LL_VERSION_IND_pcap.tar.gz](https://github.com/user-attachments/files/20043990/LL_VERSION_IND_pcap.tar.gz)

Exact code used (with argv[1] of "hci0")

```
from whad.ble import Central
from whad.ble.profile import UUID
from whad.device import WhadDevice
from scapy.layers.bluetooth4LE import *
import sys

if len(sys.argv) >= 2:
    interface = sys.argv[1]
    # Create the Central connector & the WHAD device
    central = Central(WhadDevice.create(interface))

    # Connect to a specific device
    device = central.connect('ca:fe:13:37:00:01', random=True)

    # # Discover and display the profile
    # print("[i] Discovering profile...")
    # device.discover()
    # print("[i] Discovered profile")
    # for service in device.services():
    #     print('-- Service %s' % service.uuid)
    #     for charac in service.characteristics():
    #         print(' + Characteristic %s' % charac.uuid)

    # # Get the device name
    # device_name = device.get_characteristic(UUID(0x1800), UUID(0x2A00))
    # print("[i] Device name: ", device_name.value)

    # Make sure connection has succeeded
    if device is not None:

        # Enable synchronous mode: we must process any incoming BLE packet.
        central.enable_synchronous(True)

        # Send a LL_VERSION_PDU
        result = central.send_pdu(BTLE_DATA()/BTLE_CTRL()/LL_VERSION_IND(
            version = 0x08,
            company = 0x0101,
            subversion = 0x0001
        ), conn_handle=device.conn_handle)

        # Wait for a packet
        while central.is_connected():

            # Wait for a packet to be returned, could be None if queue
            # is empty.
            packet = central.wait_packet()

            # If this packet is a LL_VERSION_IND PDU, show its content
            # and disconnet. Any other BLE PDU will be ignored.
            if packet is not None and packet.haslayer(LL_VERSION_IND):
                packet[LL_VERSION_IND].show()
                break

    # Disconnect
    device.disconnect()
    central.close()
else:
    print("Usage: ", sys.argv[0]+" <interface>")
```

virtualabs, 2025-05-05:
And that's normal because HCI interfaces cannot send control PDUs (but we should check that and raise an exception if the hardware is asked to send a control PDU while it definitely cannot due to missing capabilities).

virtualabs, 2025-05-05:
Is this something acceptable when a `Central` object is used to send a BLE control PDU with an interface (HCI for instance) that does not support it ? It basically reports an error through `logging` and raises an `UnsupportedCapability` exception.

```
Connecting to remote device ...
WHAD interface hci0 cannot send BLE control PDUs, please use another interface that supports sending raw PDUs.
Traceback (most recent call last):
  File "/foobar/test.py", line 17, in <module>
    result = central.send_pdu(BTLE_DATA()/BTLE_CTRL()/LL_VERSION_IND(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/foobar/whad-client/whad/ble/connector/central.py", line 370, in send_pdu
    return super().send_pdu(pdu, conn_handle, direction=direction,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/foobar/whad/ble/connector/base.py", line 830, in send_pdu
    raise UnsupportedCapability("RawInject")
whad.exceptions.UnsupportedCapability: UnsupportedCapability(RawInject)
```

XenoKovah, 2025-05-05:
Yeah, surfacing that it's not possible through HCI interfaces makes sense (especially in retrospect).

virtualabs, 2025-05-06:
Changes merged into `main` branch in #185.

closed 2025-05-06 by virtualabs