whad-team/whad-client #190: L2CAP Fragmentation error on read response with increased MTU

p2lu, 2025-05-07:
Refering to #186 and the following script

```py
from whad.device import WhadDevice
from whad.ble.connector.central import CentralServer
from whad.ble.profile import GenericProfile, PrimaryService, Characteristic, UUID
from whad.ble.stack.l2cap import logger
from whad.ble.stack.att import logger
import logging
import sys
import time
from string import ascii_uppercase

logger = logging.getLogger("whad.ble.stack.l2cap")
handler = logging.StreamHandler()
handler.setLevel(logging.DEBUG)
logger.addHandler(handler)
logger.setLevel(logging.DEBUG)

class MyCentral(GenericProfile):

    device = PrimaryService(
        uuid=UUID(0x1800),

        device_name = Characteristic(
            uuid=UUID(0x2A00),
            permissions = ['read', 'write'],
            notify=True,
            value=ascii_uppercase.encode()
        )
    )

# Create a central device
central = CentralServer(WhadDevice.create("uart3"), profile=MyCentral())
target = central.connect(sys.argv[1])
time.sleep(2)
central.set_mtu(517)
input("...")
```
This client is connecting to the peripheral, change the mtu, and then the peripheral is discovering the primary service and the correspond characteristics, searching for the 0x2A00 uuid Characteristic, then execute a read request.
This works well without the MTU increasing modification.

Here capture of an example when the client is increasing the MTU then respond to a read request (tested with NRF connect with my smartphone (with a custom > 22 bytes name and captured with btlejack)
 
[image]

Here is the same exchange with the whad framework (and the same script as mentioned before)

[image]

This is a single L2CAP fragment of the name[21:].

fatefully the peripheral is throwing a error: `[00:00:18.345,000] <err> bt_conn: Unexpected L2CAP continuation`

I don't think this is a expected behavior, i will try to dig more about this (maybe a PR if i succeed).

Just for letting you know about this, if you have more information about this behavior It's with pleasure.

Thanks.

p2lu, 2025-05-07:
The issue seems to be the `self.state.remote_mtu` which is not updated here
https://github.com/whad-team/whad-client/blob/main/whad/ble/stack/l2cap/__init__.py#L122

virtualabs, 2025-05-08:
I've made a fix related to this issue of class inheritance and pushed into the `feature/ble-central-server` branch. Python's MRO helped a bit to solve this problem, and the BLE stack now provides a `GattServerClient` layer for a GATT server with client procedures support and a `GattClientServer` layer for a GATT client that also accepts server-related procedures. These layers are used respectively in our `PeripheralClient` and `CentralServer` connectors.

Hoping this would solve this MTU issue...

virtualabs, 2025-05-12:
> I'd rather work with you than making a bad PR

That's totally fine ! I know the BLE stack and WHAD's stack layer model are quite difficult to apprehend, so it's normal to ask for help rather losing a lot of time to understand how things work to find the root cause of a bug 👍.

Your log is very helpful (glad you shared it !), indeed. I can see that WHAD sent to the hardware interface the first fragment of the answer to the received ReadRequest (a `SendRawBlePdu` message):

```
DEBUG:whad.device.device:Forward message <BleRawPduReceived access_address=597939335, channel=23, conn_handle=0, crc=2219836, crc_validity=True, decrypted=False, direction=2, pdu=b'\x06\x07\x03\x00\x04\x00\n\x03\x00', processed=False, relative_timestamp=221, rssi=0, timestamp=142823328> to connector <whad.ble.connector.central.CentralServer object at 0x73dbfd9a79e0>
INFO:whad.device.device:message concerns domain `ble`, forward to domain-specific handler
DEBUG:whad.ble.connector.base:[BLE connector] on_packet
DEBUG:whad.ble.stack:received a data PDU (9 bytes) for conn_handle 0
DEBUG:whad.ble.stack.l2cap:[l2cap] received start of fragmented or complete message
DEBUG:whad.ble.stack.l2cap:[l2cap] expected l2cap length: 7
DEBUG:whad.ble.stack.l2cap:[l2cap] actual l2cap length: 7
DEBUG:whad.ble.stack.l2cap:[l2cap] Received a complete L2CAP packet, process it
DEBUG:whad.ble.stack.gatt:Start handling PDU (tx/rx)
DEBUG:whad.ble.stack.llm:sending l2cap data PDU for conn_handle 0 (26 bytes)
DEBUG:whad.ble.stack:sending a data PDU (28 bytes) to conn_handle 0
INFO:whad.ble.connector.base:send data PDU to connection (handle:0)
INFO:whad.ble.connector.base:[ble connector] link-layer encryption: False
DEBUG:whad.hub:[hub] convert_packet(): packet is BLE
INFO:whad.device.connector:[connector] send packet command
INFO:whad.device.device:sending message <SendBleRawPdu access_address=2391391958, conn_handle=0, crc=4606362, direction=1, encrypt=False, pdu=b'\x02\x1a\x1b\x00\x04\x00\x0bABCDEFGHIJKLMNOPQRSTU'> to device <uart3>
DEBUG:whad.device.device:send_message:set_queue_filter
DEBUG:whad.device.device:set queue filter: <function message_filter.<locals>.<lambda> at 0x73dbfb97a520>
DEBUG:whad.device.device:sending b'\xac\xbe/\x00' to WHAD device uart3
DEBUG:whad.device.device:sending b'\x1a-r+\x08\x01\x18\xd6\xfd\xa6\xf4\x08"\x1c\x02\x1a\x1b\x00\x04\x00\x0bABCDEFGHIJKLMNOPQRSTU(\x9a\x93\x99\x02' to WHAD device uart3
DEBUG:whad.device.device:entering wait_for_message ...
DEBUG:whad.device.device:[WhadDevice] received raw data from device <uart3>: b'acbe04000a021200'
DEBUG:whad.device.device:[WhadDevice::on_message_received()][uart3] message received: <whad.hub.generic.cmdresult.Success object at 0x73dbfb98e540>
DEBUG:whad.device.device:[WhadDevice::on_message_received()][uart3] message queue filter: <function message_filter.<locals>.<lambda> at 0x73dbfb97a520>
INFO:whad.device.device:message does match current filter, save it for processing
INFO:whad.device.device:message added to message queue
DEBUG:whad.device.device:exiting wait_for_message ...
DEBUG:whad.device.device:Command result: <whad.hub.generic.cmdresult.Success object at 0x73dbfb98e540>
INFO:whad.device.connector:[connector] Command sent, result: <whad.hub.generic.cmdresult.Success object at 0x73dbfb98e540>
```

And it appears to effectively be the first fragment you are expecting to be sent:
```python
>>> BTLE_DATA(b'\x02\x1a\x1b\x00\x04\x00\x0bABCDEFGHIJKLMNOPQRSTU').show()
###[ BTLE data header ]### 
  RFU       = 0
  MD        = 0
  SN        = 0
  NESN      = 0
  LLID      = start
  len       = 26
###[ L2CAP header ]### 
     len       = 27
     cid       = attribute
###[ ATT header ]### 
        opcode    = 0xb
###[ Read Response ]### 
           value     = 'ABCDEFGHIJKLMNOPQRSTU'
```

But this PDU seems to be somehow lost in the process and ended up not sent by the connected hardware. The second fragment, however, made it through:

```
DEBUG:whad.ble.stack.llm:sending l2cap data PDU for conn_handle 0 (5 bytes)
DEBUG:whad.ble.stack:sending a data PDU (7 bytes) to conn_handle 0
INFO:whad.ble.connector.base:send data PDU to connection (handle:0)
INFO:whad.ble.connector.base:[ble connector] link-layer encryption: False
DEBUG:whad.hub:[hub] convert_packet(): packet is BLE
INFO:whad.device.connector:[connector] send packet command
INFO:whad.device.device:sending message <SendBleRawPdu access_address=2391391958, conn_handle=0, crc=15660481, direction=1, encrypt=False, pdu=b'\x01\x05VWXYZ'> to device <uart3>
```

This is the one we see in your capture.

Just to be sure before going hunting this bug, what hardware are you using with ButteRFly: an official nRF52 USB dongle or a MakerDiary dongle ?

My guess is that the hardware (running ButteRFly firmware) is told to send the first fragment and very shortly after to send the second one, but in the process the firmware is losing the first one (not sent because of some timing specificities due to BLE's connection events) but managed to send the second one.

I am quite sure we tested modifying a connection MTU only with HCI adapters and not ButteRFly, so this hypothesis looks promising. I'm going to review ButteRFly's firmware source code and see if I can find the root cause of this bug. I'll keep you posted.

p2lu, 2025-05-12:
You're absolutely right !
I'm using a NRF52840 with the [latest butterfly firmware](https://github.com/whad-team/butterfly/releases/tag/v1.0.1)
I managed a workaround for this issue by adding a time.sleep(0.1) in the sender located [here](https://github.com/whad-team/whad-client/blob/main/whad/ble/stack/l2cap/__init__.py#L140)

```py
import time
# ...
    @source('att')
    def on_att_packet_recv(self, data, channel='attribute'):
        """Process incoming packets from ATT that must be forwarded to link layer.
        """
        packets = self.get_fragments(data)
        # Send packets
        for i, pkt in enumerate(packets):
            # First packet is sent with no fragment flag
            time.sleep(0.1) # workaround here
            if i == 0:
                # Send packet to link layer
                self.send('ll', L2CAP_Hdr(cid=channel, len=len(data))/pkt, fragment=False)
            else:
                # Send packet to link layer
                self.send('ll', pkt, fragment=True)
```
And this works ! 

[image]

I don't think this valid for a long time support, so maybe searching in the message handler into the butterfly firmware will be necessary.

virtualabs, 2025-05-12:
I managed to reproduce this issue on my side, it will be easier to find the root cause of this bug 😄.

[image]

p2lu, 2025-05-12:
This is more of a comment than a continuation of the issue but it seems the length for a read response is limited to 64 bytes.

This image below shown a 66 'A' padded data chunk
[image]

And from my zephyr 3.7.0 device

```
[00:00:17.247,000] <err> bt_conn: Not enough buffer space for L2CAP data
[00:00:17.247,000] <inf> ble_handler: Receiving 43 length packet
[00:00:17.247,000] <inf> ble_handler: NAME: ABCDEFGHIJKLMNOPQRSTUVWXYZ_AAAAAAAAAAAAAAAA

[00:00:17.247,000] <dbg> bt_conn: bt_conn_unref: handle 0 ref 2 -> 1
[00:00:17.317,000] <dbg> bt_conn: bt_conn_ref: handle 0 ref 1 -> 2
[00:00:17.317,000] <dbg> bt_conn: wait_for_tx_work: conn 0x3ffca530
[00:00:17.317,000] <dbg> bt_conn: tx_complete_work: conn 0x3ffca530
[00:00:17.317,000] <dbg> bt_conn: tx_notify: conn 0x3ffca530
[00:00:17.317,000] <dbg> bt_conn: wait_for_tx_work: done
[00:00:17.317,000] <dbg> bt_conn: bt_conn_recv: handle 0 len 1 flags 01
[00:00:17.317,000] <err> bt_conn: Unexpected L2CAP continuation
```
(I receive successfully all the data if <=64, the 43 length correspond to the first and second combined L2CAP trames).

What is surprising is the data length for a request response from the NRF Connect app seems to be also limited to 64 bytes, even if  the data (as shown in the first message in this issue) is sent from a single ATT trame and not encapsulated in multiples L2CAP fragments like WHAD.

I searched about this and I found nothing really satisfying.

Is this a excepted behavior for the read response method to be limited as a 64 bytes maximum length data ?

virtualabs, 2025-05-12:
Oh jeez, it was in front of my eyes all along 🤦. I made a lot of additional captures with various sniffers, reviewed WHAD's Python code to make sure that both PDUs were correctly sent to the ButteRFly, and eventually added some debug messages into ButteRFly...

This is the piece of code responsible of handling `SendBleRawPdu` messages:

```cpp
       case whad::ble::SendRawPduMsg:
        {
            whad::ble::SendRawPdu query(bleMsg);
            uint32_t max_retry = 1 << 24;

            switch (query.getDirection())
            {
                /*** PDU injection stuff we don't care about, removed for clarity ***/

                case whad::ble::DirectionMasterToSlave:
                {
                    if (this->bleController->getState() == SIMULATING_MASTER || this->bleController->getState() == PERFORMING_MITM) {
                        this->bleController->setMasterPayload(query.getPdu().getBytes(), query.getPdu().getSize());
                        while (!max_retry && !this->bleController->isMasterPayloadTransmitted()) {--max_retry;}
                        if (max_retry != 0) {
                            response = new whad::generic::Success();
                        } else {
                            response = new whad::generic::Error();

                        }
                    }
                    else {
                        response = new whad::generic::WrongMode();
                    }
                }
                break;
```

The `max_retry` variable is set to `1 << 24` ( or `0x01000000`), so it's not zero. This while loop is just removed by the compiler as its condition will never be satisfied, causing the code to directly send a success response while a packet is still pending ! The first fragment of PDU is still waiting to be sent when the second one replaces it, and will be sent later.

The expected behavior of this type of messages (like `SendBlePdu` and others for other protocols) is to queue a packet for transmission rather than directly sending it, and this for a very good reason: we cannot lock the hardware in waiting mode, sending has to be asynchronous to keep the communication alive.
It adds another coin in our "We need to add new features to ButteRFly and improve its code quality" coinbox, and it is quite painful as I remember discussing this matter with some fellow researchers and contributors. But I guess we have no other option since most devices nowadays use BLE 5 for instance, which is still not supported by ButteRFly. We'll do some code cleaning as well. 

I'll issue a proper fix for this bug in the ButteRFly repository and plan a new release. In the meantime, here is a hotfix solving this issue (you'll need to flash your nRF52840 by following [this procedure](https://whad.readthedocs.io/en/latest/device/compat.html#installing-butterfly-firmware-on-a-nordic-nrf52840-dongle)).

Hope this helps !

[butterfly-hotfix.zip](https://github.com/user-attachments/files/20177794/butterfly-hotfix.zip)