whad-team/whad-client #191: Bug: WHAD MITM works, until reboot

XenoKovah, 2025-05-07:
I'm encountering a weird issue and I need some guidance on how best to debug it. tl;dr: I have WHAD MITM working in a VM snapshot, but after I reboot it stops working, and I can't figure out how to get it back into a working state.

To confirm, I manually issue the following after reboot (and confirm bluetoothd isn't running), so I don't think the issue is that bluetoothd has started across reboots.
```
sudo systemctl disable bluetooth.service
sudo systemctl stop bluetooth.service
```

In order to confirm that the code I'm running is the code in the folder, I also issue the following before rebooting just to ensure I'm consistent:

```
cd ~/class/whad-client
source ./venv/bin/activate
pip3 install .
```

Log files will be transferred out of band so as not to require full sanitization.

Any ideas on what else to collect to figure out what's going on?

virtualabs, 2025-05-09:
First, I may have some questions in order to have a clear idea of the context:
- are your HCI interfaces working fine with other tools like gatttool, gattacker, btlejack, mirage, anything except whad ?
- are you able to connect to a BLE device using the VM after reboot ?
- what is your test process to determine that the proxy is not working ?
- when the proxy is running, is it connected to the target device ?

From what I can tell from the logs you shared, it looks like WHAD is able to communicate with your HCI devices but the spoofed device spawn by the proxy did not receive any connection. 

Can you monitor both of your HCI devices with `bt-mon` and share the logs (when proxy is working and same after reboot), in order to detect any HCI-related issue that we may have missed ? I'll try to reproduce this with VMware on my side.

XenoKovah, 2025-05-13:
> are your HCI interfaces working fine with other tools like gatttool, gattacker, btlejack, mirage, anything except whad ?

Yes, confirmed with bluetoothctl

>are you able to connect to a BLE device using the VM after reboot ?

Yes, confirmed with bluetoothctl

>what is your test process to determine that the proxy is not working ?

Run `wble-proxy -i hci0 -p hci1 ca:fe:13:37:00:01` and see that while it *does* print out
```
Scanning for target device (timeout: 30 seconds)...
Found target device ca:fe:13:37:00:01 (random)
Proxy is ready, press a key to stop.
```
It never prints out any indications or data reads/writes, like it does when it's working.

Also I pcapped both connections, and while WHAD connects to the UVP and the UVC connects to WHAD, in the non-working case, the UVC-to-WHAD connection dies prematurely.

> when the proxy is running, is it connected to the target device ?

Yes, that side stays up fine. It's the WHAD fake peripheral slide which is terminating prematurely.

virtualabs, 2025-05-31:
In the end, I ended up rewriting WHAD's HCI virtual device code to ensure it follows the Bluetooth specification, while it appears this bug is caused by the USB BT dongle plugged into the virtual machine going in suspend mode. USB suspend mode is known to mess with VMWare and other virtualization software, disabling it in the guest OS is enough to solve this issue. 

Anyways, the rewritten HCI interface code has been kept as it is more complete than the previous implementation, and documentation has been updated to include a note about USB auto-suspend and how to disable it on Ubuntu (the recommended OS to use WHAD in VM).

virtualabs, 2025-06-02:
The improved HCI interface implementation has been merged into the `main` branch, along with additional fixes and improvements related to WHAD's BLE tools.