whad-team/whad-client #192: Using examples/ble/ble_existing_connection_sniffer.py

XenoKovah, 2025-05-08:
Since WHAD Is The Future™ I wanted to see if I could do btlejack stuff with it. So I tried out `ble_existing_connection_sniffer.py`. I disabled PHY update procedure (reasoning that it's a currently-unsupported BT 5.0+ thing) in the UVP by adding `CONFIG_BT_PHY_UPDATE=n` to `UltraVulnerablePeripheral/prj.conf`. I then connected the UVC to the UVP and sniffed the initial connection with Sniffle. I copied the AccessAddress and CRCInit values out of Wireshark and into `ble_existing_connection_sniffer.py` and also set the initial channel map to ffffffff1f which is what was used in the CONNECT_IND. (I also confirmed separately that I can see initial values via `ble/ble_new_connection_sniffer.py`). Then I ran `ble_existing_connection_sniffer.py` with a uart0 interface from a Nordic nRF53840 dongle running ButteRFly. However I never see anything printed out in `ble_existing_connection_sniffer.py`

Is there anything else in this pcap which would suggest why there are problems?

[UVC_UVP_Connection.pcapng.zip](https://github.com/user-attachments/files/20103230/UVC_UVP_Connection.pcapng.zip)

virtualabs, 2025-05-09:
It's been a while we have not tested injection using WHAD and ButteRFly, mostly because recent BLE controllers use (and abuse) of channel map updates making it very difficult to synchronize with an existing connection, or simply use BLE 5 which is unfortunately not yet supported in ButteRFly. I'm adding this to my todo list.

XenoKovah, 2025-05-11:
I disabled the CSA#2 in the UVP by adding `CONFIG_BT_CTLR_CHAN_SEL_2=n` to `UltraVulnerablePeripheral/prj.conf` and then tried again.

And I modified the printout slightly to print the full packet instead of just the summary of layers, like this:

```
            # Iterate over traffic
            sniffer.start()
            for i in sniffer.sniff():
                print(i)
                i.show()
```

However, while some stuff does get printed out by ``, it isn't stuff that should be printed out if it's targeting my specific AccessAddress 0x41f8e5fa, which was put into the code before starting, after extracting from the CONNECT_IND. E.g.:

```
BTLE / BTLE_ADV / BTLE_ADV_IND
###[ BT4LE ]### 
  access_addr= 0x8e89bed6
  crc       = 0x55c980
###[ BTLE advertising header ]### 
     RxAdd     = public
     TxAdd     = random
     ChSel     = #2
     RFU       = 0
     PDU_type  = ADV_IND
     Length    = 0x1b
###[ BTLE ADV_IND ]### 
        AdvA      = 63:82:42:5e:ce:2d
        \data      \
         |###[ EIR Header ]### 
         |  len       = 2
         |  type      = flags
         |###[ Flags ]### 
         |     flags     = general_disc_mode+simul_le_br_edr_ctrl+simul_le_br_edr_host
         |###[ EIR Header ]### 
         |  len       = 14
         |  type      = mfg_specific_data
         |###[ EIR Manufacturer Specific Data ]### 
         |     company_id= 0x4c
         |###[ Raw ]### 
         |        load      = '\x0f\x05\\x90\x00Y\\x8c\\xa7\x10\x02.\x04'
         |###[ EIR Header ]### 
         |  len       = 2
         |  type      = tx_power_level
         |###[ TX Power Level ]### 
         |     level     = 0
```

But I do agree that there are various channel map updates from the Android phone Central after 5+ seconds of idle traffic, which indicate to use non-all channels sometimes (and switching back to the all-channels some other times.)