whad-team/whad-client #196: Missing documentation: `can_indicate()` and `can_notify()`

XenoKovah, 2025-05-12:
In the example about notifications and indications, https://whad.readthedocs.io/en/latest/ble/started.html#subscribe-for-notification-indication, it just jumps in to registering the callback. However, when I'm teaching students about enabling notifications and indications, I want them to only attempt this on things which support those functionalities. I can see in the debugger the presence of `can_indicate()` and `can_notify()`, which are the kinds of things which would need to be checked. However those currently aren't found in the documentation when searched for. So this is requesting those be added to the documentation, so I can link to it for students.

virtualabs, 2025-05-13:
Pushed a `doc/issue-196` branch that includes a more complete description of WHAD's BLE `Central` connector and the related classes, including how to correctly handle notifications and indications. Will it be ok for you @XenoKovah ?

I'll do the same for the `Peripheral` class by adding more details and use cases to the main documentation.

XenoKovah, 2025-05-13:
Documentation update LGTM!