whad-team/whad-client #239: AttributeError: 'GattServer' object has no attribute 'set_client_model'

XenoKovah, 2025-07-10:
The following replicated in 4/4 tries on the latest commit of master aff240afd3c5a322d5f9206a9d272264827d84e9

So the good news is that the core MITM is still working reliably, it's just not terminating correctly.

Repro steps:
1) Set up proxy with `wble-proxy -i hci0 -p hci1 ca:fe:13:37:00:01`
2) Connect UVC to fake WHAD UVP by clicking connect slider in upper right corner of Android UI
3) Confirm connection works and wait for indications to complete.
4) Submit an answer for question 1-A. (It doesn't matter if it's correct or not), and press "done" to return back to the main screen.
5) Wait for 3x indications to complete (based on CLI proxy output).
6) Within UVC UI, click the disconnect slider in the upper right corner of Android UI
7) Observe below output

```
(venv) user@BTVM:~/class/whad-client$ wble-proxy -i hci0 -p hci1 ca:fe:13:37:00:01
Scanning for target device (timeout: 30 seconds)...
Found target device ca:fe:13:37:00:01 (random)
Proxy is ready, press a key to stop.
Remote device connected
[!] Subscribed to indication for charac. 444d0010-4865-6d6c-6f63-6b21f09fa4aa
<<< [!] Indication for charac. 444d0010-4865-6d6c-6f63-6b21f09fa4aa:
00000000: 01 61 02 01 62 00 01 63  00 01 64 00 01 65 00 01  .a..b..c..d..e..
00000010: 66 00 01 67 00 02 61 02  02 62 01 02 63 00 02 64  f..g..a..b..c..d
00000020: 02 02 65 02 02 66 02 02  67 02 03 61 00 03 62 00  ..e..f..g..a..b.
00000030: 03 63 00 03 64 00 03 65  00 03 66 00 03 67 00 04  .c..d..e..f..g..
00000040: 61 00 04 62 00 04 63 00  04 64 00 04 65 00 04 66  a..b..c..d..e..f
00000050: 00 04 67 00 05 61 02 05  62 00 05 63 00 05 64 00  ..g..a..b..c..d.
00000060: 05 65 00 05 66 00 05 67  00 06 61 02 06 62 00 06  .e..f..g..a..b..
00000070: 63 00 06 64 00 06 65 00  06 66 00 06 67 00 07 61  c..d..e..f..g..a
00000080: 02 07 62 00 07 63 00 07  64 00 07 65 00 07 66 00  ..b..c..d..e..f.
00000090: 07 67 00 08 61 00 08 62  00 08 63 00 08 64 00 08  .g..a..b..c..d..
000000A0: 65 00 08 66 00 08 67 00  09 61 00 09 62 00 09 63  e..f..g..a..b..c
000000B0: 00 09 64 00 09 65 00 09  66 00 09 67 00           ..d..e..f..g.
<<< [!] Indication for charac. 444d0010-4865-6d6c-6f63-6b21f09fa4aa:
00000000: 01 61 02 01 62 00 01 63  00 01 64 00 01 65 00 01  .a..b..c..d..e..
00000010: 66 00 01 67 00 02 61 02  02 62 01 02 63 00 02 64  f..g..a..b..c..d
00000020: 02 02 65 02 02 66 02 02  67 02 03 61 00 03 62 00  ..e..f..g..a..b.
00000030: 03 63 00 03 64 00 03 65  00 03 66 00 03 67 00 04  .c..d..e..f..g..
00000040: 61 00 04 62 00 04 63 00  04 64 00 04 65 00 04 66  a..b..c..d..e..f
00000050: 00 04 67 00 05 61 02 05  62 00 05 63 00 05 64 00  ..g..a..b..c..d.
00000060: 05 65 00 05 66 00 05 67  00 06 61 02 06 62 00 06  .e..f..g..a..b..
00000070: 63 00 06 64 00 06 65 00  06 66 00 06 67 00 07 61  c..d..e..f..g..a
00000080: 02 07 62 00 07 63 00 07  64 00 07 65 00 07 66 00  ..b..c..d..e..f.
00000090: 07 67 00 08 61 00 08 62  00 08 63 00 08 64 00 08  .g..a..b..c..d..
000000A0: 65 00 08 66 00 08 67 00  09 61 00 09 62 00 09 63  e..f..g..a..b..c
000000B0: 00 09 64 00 09 65 00 09  66 00 09 67 00           ..d..e..f..g.
<<< [!] Indication for charac. 444d0010-4865-6d6c-6f63-6b21f09fa4aa:
00000000: 01 61 02 01 62 00 01 63  00 01 64 00 01 65 00 01  .a..b..c..d..e..
00000010: 66 00 01 67 00 02 61 02  02 62 01 02 63 00 02 64  f..g..a..b..c..d
00000020: 02 02 65 02 02 66 02 02  67 02 03 61 00 03 62 00  ..e..f..g..a..b.
00000030: 03 63 00 03 64 00 03 65  00 03 66 00 03 67 00 04  .c..d..e..f..g..
00000040: 61 00 04 62 00 04 63 00  04 64 00 04 65 00 04 66  a..b..c..d..e..f
00000050: 00 04 67 00 05 61 02 05  62 00 05 63 00 05 64 00  ..g..a..b..c..d.
00000060: 05 65 00 05 66 00 05 67  00 06 61 02 06 62 00 06  .e..f..g..a..b..
00000070: 63 00 06 64 00 06 65 00  06 66 00 06 67 00 07 61  c..d..e..f..g..a
00000080: 02 07 62 00 07 63 00 07  64 00 07 65 00 07 66 00  ..b..c..d..e..f.
00000090: 07 67 00 08 61 00 08 62  00 08 63 00 08 64 00 08  .g..a..b..c..d..
000000A0: 65 00 08 66 00 08 67 00  09 61 00 09 62 00 09 63  e..f..g..a..b..c
000000B0: 00 09 64 00 09 65 00 09  66 00 09 67 00           ..d..e..f..g.
>>> Characteristic 444d0012-4865-6d6c-6f63-6b21f09fa4aa written
00000000: 01 61 01                                          .a.
>>> Characteristic 444d0011-4865-6d6c-6f63-6b21f09fa4aa read
00000000: 01 61 01                                          .a.
<<< [!] Indication for charac. 444d0010-4865-6d6c-6f63-6b21f09fa4aa:
00000000: 01 61 01 01 62 00 01 63  00 01 64 00 01 65 00 01  .a..b..c..d..e..
00000010: 66 00 01 67 00 02 61 02  02 62 01 02 63 00 02 64  f..g..a..b..c..d
00000020: 02 02 65 02 02 66 02 02  67 02 03 61 00 03 62 00  ..e..f..g..a..b.
00000030: 03 63 00 03 64 00 03 65  00 03 66 00 03 67 00 04  .c..d..e..f..g..
00000040: 61 00 04 62 00 04 63 00  04 64 00 04 65 00 04 66  a..b..c..d..e..f
00000050: 00 04 67 00 05 61 02 05  62 00 05 63 00 05 64 00  ..g..a..b..c..d.
[... 112 lines cut ...]
    self.send_message(command, keep)
  File "/home/user/class/whad-client/venv/lib/python3.12/site-packages/whad/device/device.py", line 1124, in send_message
    self._on_whad_message(message)
  File "/home/user/class/whad-client/venv/lib/python3.12/site-packages/whad/device/device.py", line 1134, in _on_whad_message
    getattr(self, callback_name)(message)
  File "/home/user/class/whad-client/venv/lib/python3.12/site-packages/whad/device/virtual/hci/__init__.py", line 1491, in _on_whad_ble_stop
    self.terminate_connection()
  File "/home/user/class/whad-client/venv/lib/python3.12/site-packages/whad/device/virtual/hci/__init__.py", line 1153, in terminate_connection
    if self._disconnect():
       ^^^^^^^^^^^^^^^^^^
  File "/home/user/class/whad-client/venv/lib/python3.12/site-packages/whad/device/virtual/hci/__init__.py", line 169, in _wrap
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: HCIDevice._disconnect() missing 1 required positional argument: 'handle'
(venv) user@BTVM:~/class/whad-client$ git log -1
commit aff240afd3c5a322d5f9206a9d272264827d84e9 (HEAD -> main, origin/main, origin/HEAD)
Merge: 8a756e29 02f4ff0f
Author: Romain Cayre <rcayre@laas.fr>
Date:   Mon Jul 7 10:50:38 2025 +0200

    Merge pull request #236 from erdnaxe/test-relative-import
    
    tests/protocol/ble: use relative imports

```

XenoKovah, 2025-07-12:
Initially it looked like the disconnect was being handled more gracefully when I was doing the disconnect from within the UVC UI, because it only printed `ERROR:whad.ble.tools.proxy:GATT connection lost while unsubscribing from characteristic` and then continued. But I then did a 2nd try and got the `AttributeError: 'GattServer' object has no attribute 'set_client_model'` error again. So I'm not sure if that's expected with the stated fix for 2/3 of the issues or not.

```
<<< [!] Indication for charac. 444d0010-4865-6d6c-6f63-6b21f09fa4aa:
00000000: 01 61 02 01 62 00 01 63  00 01 64 00 01 65 00 01  .a..b..c..d..e..
00000010: 66 00 01 67 00 02 61 02  02 62 01 02 63 00 02 64  f..g..a..b..c..d
00000020: 02 02 65 02 02 66 02 02  67 02 03 61 00 03 62 00  ..e..f..g..a..b.
00000030: 03 63 00 03 64 00 03 65  00 03 66 00 03 67 00 04  .c..d..e..f..g..
00000040: 61 00 04 62 00 04 63 00  04 64 00 04 65 00 04 66  a..b..c..d..e..f
00000050: 00 04 67 00 05 61 02 05  62 00 05 63 00 05 64 00  ..g..a..b..c..d.
00000060: 05 65 00 05 66 00 05 67  00 06 61 02 06 62 00 06  .e..f..g..a..b..
00000070: 63 00 06 64 00 06 65 00  06 66 00 06 67 00 07 61  c..d..e..f..g..a
00000080: 02 07 62 00 07 63 00 07  64 00 07 65 00 07 66 00  ..b..c..d..e..f.
00000090: 07 67 00 08 61 00 08 62  00 08 63 00 08 64 00 08  .g..a..b..c..d..
000000A0: 65 00 08 66 00 08 67 00  09 61 00 09 62 00 09 63  e..f..g..a..b..c
000000B0: 00 09 64 00 09 65 00 09  66 00 09 67 00           ..d..e..f..g.
ERROR:whad.ble.tools.proxy:GATT connection lost while unsubscribing from characteristic
Remote device disconnected
Remote device connected
[!] Subscribed to indication for charac. 444d0010-4865-6d6c-6f63-6b21f09fa4aa
[!] Subscribed to indication for charac. 444d0010-4865-6d6c-6f63-6b21f09fa4aa
<<< [!] Indication for charac. 444d0010-4865-6d6c-6f63-6b21f09fa4aa:
00000000: 01 61 02 01 62 00 01 63  00 01 64 00 01 65 00 01  .a..b..c..d..e..
00000010: 66 00 01 67 00 02 61 02  02 62 01 02 63 00 02 64  f..g..a..b..c..d
00000020: 02 02 65 02 02 66 02 02  67 02 03 61 00 03 62 00  ..e..f..g..a..b.
00000030: 03 63 00 03 64 00 03 65  00 03 66 00 03 67 00 04  .c..d..e..f..g..
00000040: 61 00 04 62 00 04 63 00  04 64 00 04 65 00 04 66  a..b..c..d..e..f
00000050: 00 04 67 00 05 61 02 05  62 00 05 63 00 05 64 00  ..g..a..b..c..d.
00000060: 05 65 00 05 66 00 05 67  00 06 61 02 06 62 00 06  .e..f..g..a..b..
00000070: 63 00 06 64 00 06 65 00  06 66 00 06 67 00 07 61  c..d..e..f..g..a
00000080: 02 07 62 00 07 63 00 07  64 00 07 65 00 07 66 00  ..b..c..d..e..f.
00000090: 07 67 00 08 61 00 08 62  00 08 63 00 08 64 00 08  .g..a..b..c..d..
000000A0: 65 00 08 66 00 08 67 00  09 61 00 09 62 00 09 63  e..f..g..a..b..c
000000B0: 00 09 64 00 09 65 00 09  66 00 09 67 00           ..d..e..f..g.
<<< [!] Indication for charac. 444d0010-4865-6d6c-6f63-6b21f09fa4aa:
00000000: 01 61 02 01 62 00 01 63  00 01 64 00 01 65 00 01  .a..b..c..d..e..
00000010: 66 00 01 67 00 02 61 02  02 62 01 02 63 00 02 64  f..g..a..b..c..d
00000020: 02 02 65 02 02 66 02 02  67 02 03 61 00 03 62 00  ..e..f..g..a..b.
00000030: 03 63 00 03 64 00 03 65  00 03 66 00 03 67 00 04  .c..d..e..f..g..
00000040: 61 00 04 62 00 04 63 00  04 64 00 04 65 00 04 66  a..b..c..d..e..f
00000050: 00 04 67 00 05 61 02 05  62 00 05 63 00 05 64 00  ..g..a..b..c..d.
00000060: 05 65 00 05 66 00 05 67  00 06 61 02 06 62 00 06  .e..f..g..a..b..
00000070: 63 00 06 64 00 06 65 00  06 66 00 06 67 00 07 61  c..d..e..f..g..a
00000080: 02 07 62 00 07 63 00 07  64 00 07 65 00 07 66 00  ..b..c..d..e..f.
00000090: 07 67 00 08 61 00 08 62  00 08 63 00 08 64 00 08  .g..a..b..c..d..
000000A0: 65 00 08 66 00 08 67 00  09 61 00 09 62 00 09 63  e..f..g..a..b..c
000000B0: 00 09 64 00 09 65 00 09  66 00 09 67 00           ..d..e..f..g.
<<< [!] Indication for charac. 444d0010-4865-6d6c-6f63-6b21f09fa4aa:
00000000: 01 61 02 01 62 00 01 63  00 01 64 00 01 65 00 01  .a..b..c..d..e..
00000010: 66 00 01 67 00 02 61 02  02 62 01 02 63 00 02 64  f..g..a..b..c..d
00000020: 02 02 65 02 02 66 02 02  67 02 03 61 00 03 62 00  ..e..f..g..a..b.
00000030: 03 63 00 03 64 00 03 65  00 03 66 00 03 67 00 04  .c..d..e..f..g..
00000040: 61 00 04 62 00 04 63 00  04 64 00 04 65 00 04 66  a..b..c..d..e..f
00000050: 00 04 67 00 05 61 02 05  62 00 05 63 00 05 64 00  ..g..a..b..c..d.
00000060: 05 65 00 05 66 00 05 67  00 06 61 02 06 62 00 06  .e..f..g..a..b..
00000070: 63 00 06 64 00 06 65 00  06 66 00 06 67 00 07 61  c..d..e..f..g..a
00000080: 02 07 62 00 07 63 00 07  64 00 07 65 00 07 66 00  ..b..c..d..e..f.
00000090: 07 67 00 08 61 00 08 62  00 08 63 00 08 64 00 08  .g..a..b..c..d..
000000A0: 65 00 08 66 00 08 67 00  09 61 00 09 62 00 09 63  e..f..g..a..b..c
000000B0: 00 09 64 00 09 65 00 09  66 00 09 67 00           ..d..e..f..g.
>>> Characteristic 444d0012-4865-6d6c-6f63-6b21f09fa4aa written
00000000: 02 61 01                                          .a.
>>> Characteristic 444d0011-4865-6d6c-6f63-6b21f09fa4aa read
00000000: 02 61 01                                          .a.
<<< [!] Indication for charac. 444d0010-4865-6d6c-6f63-6b21f09fa4aa:
00000000: 01 61 02 01 62 00 01 63  00 01 64 00 01 65 00 01  .a..b..c..d..e..
00000010: 66 00 01 67 00 02 61 01  02 62 01 02 63 00 02 64  f..g..a..b..c..d
00000020: 02 02 65 02 02 66 02 02  67 02 03 61 00 03 62 00  ..e..f..g..a..b.
00000030: 03 63 00 03 64 00 03 65  00 03 66 00 03 67 00 04  .c..d..e..f..g..
00000040: 61 00 04 62 00 04 63 00  04 64 00 04 65 00 04 66  a..b..c..d..e..f
[... 37 lines cut ...]
[... cut ...]
  File "/home/user/class/whad-client/venv/lib/python3.12/site-packages/whad/ble/stack/__init__.py", line 53, in on_connection
    self.__connector.on_new_connection(connection)
  File "/home/user/class/whad-client/venv/lib/python3.12/site-packages/whad/ble/connector/central.py", line 489, in on_new_connection
    self.__gatt_client.set_client_model(self.__peripheral)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'GattServer' object has no attribute 'set_client_model'
^C/!\ wble-proxy stopped (CTL-C)
```

virtualabs, 2025-07-12:
That's exactly the same last issue but the other way around: this time the central connector instantiated a GATT server instead of a GATT client, due to the same race condition on the ATT class's sublayer definition.

I've found a clean way to solve this issue but I had to change WHAD's stack model a bit. I've added the possibility to define a new flavor for a given stack, in which any protocol layer implementation can be replaced by another but only for this specific flavor. Stack layer definition has evolved a bit and I still need to reflect theses changes in the documentation before submitting a PR, but IMHO that's a better way to manage different variants of a given stack without breaking its default behavior. Using this new feature, it's pretty easy to create two flavors for our BLE stack: one that will use a GATT client implementation as its GATT layer and another a GATT server. The selected role (or "flavor") is selected when the whole stack is instantiated, and it will stick with all the protocol layers associated with this "flavor". Another piece of software can then instantiate the same stack model but with another role, without dynamically modifying the stack model.

I'm not sure, as a non-native English speaker, that the term "flavor" is suitable to describe a specific variant of a protocol stack, any remark/comment/suggestion is more than welcome ! It could also be named "role" or "variant", but IMHO the first one does not fit as it better describes the role of a layer rather than a global role for a protocol stack,  while the other one seems almost fine but looks like it may describe some protocol stack that is significantly different from the standard one. But I'm maybe overthinking this and any term could be OK...

I'm working on it and it should be ready for testing pretty soon !